360Controller/360Controller #284: Xbox One Controller Analogue Sticks not working

wik007, 2016-05-28:
I am chasing some help with getting my Xbox One Controller working. Everything is working except for the analogue sticks (directional). I am currently using the latest 0.16 driver and am running MAC OS X 10.11.5 El Capitan.

FranticRain, 2016-05-28:
This is very unusual. Do you know if this is just a problem with this driver? Can you recreate it on a Windows machine or an Xbox One? Go into your system preferences through: Apple Menu -> System Preferences -> Xbox 360 Controllers. There you should be able to check if your sticks are working or not. Make sure that your deadzones are turned off (drag the slider all the way to the left). Let me know what you find.

wik007, 2016-05-29:
I believe it is a problem with either the driver or some other sort of hardware issue relating to my Mac as this particular controller I use quite often on my Xbox One, As for a windows machine, I do not have access to one to test it on.

The deadlines are all turned off and receive no input from the sticks. One of the other things that happens is that while it does not receive input from the analogue sticks, 50% of the time the controller is connected, it will see a 100% input to the right on both sticks, while still receiving no input.

FranticRain, 2016-05-29:
If it works with an Xbox One, then it is definitely the driver. Its unfortunate that you don't have a windows PC, because that would be helpful in debugging the problem. I'll try and put together a version of the driver that can tell me what is wrong with your controller, and then you can run it for me. Give me a bit.

FranticRain, 2016-05-29:
Okay. This version of the driver will print raw information that your controller is sending to the driver to Console.app. You can find Console.app in your Applications/Utilities folder. The output from the driver will look like this:
```
5/28/16 7:17:31.000 PM kernel[0]: Xbox One Controller - Packet Size: 14
5/28/16 7:17:31.000 PM kernel[0]: Xbox One Controller - Packet Data: 32 0 197 14 0 0 0 1890 -445 1092 458 
```
Simply install this version of the driver, reboot your computer, open up Console.app, and plug you controller in. Once it is connected, try moving the sticks. The last four numbers of that batch correspond the the values of the sticks.
4th from last: Left X. Left is negative, right is positive
3rd from last: Left Y. Down is negative, up is positive
2nd from last: Right X. Left is negative, right is positive
Last: Right Y. Down is negative, up is positive

Let me know what you see. https://drive.google.com/file/d/0B5PLCKoMd4zGblN6WjE4cWFieWs/view?usp=sharing

FranticRain, 2016-07-30:
Closing due to inactivity.

closed 2016-07-30 by FranticRain