360Controller/360Controller #285: No devices found

maxblaster5000, 2016-05-29:
Hello. I am trying to use an Afterglow Gamepad for Xbox 360 on my Macbook Pro (Retina, 15-inch, Late 2013) running OS X El Capitan (Version 10.11.5). The Product ID for the controller is 0xfafd and the Vendor ID is 0x24c6. I have spent hours trying to troubleshoot the problem on my own so it is possible that I may have screwed something up in the process. I am using version 0.16 of the drivers currently but have tried many different versions to get my controller to work. I have also attempted completely uninstalling it, restarting and then reinstalling to no avail. Your assistance would be greatly appreciated.

maxblaster5000, 2016-05-29:
I tried a build you provided to someone with the same Product ID and Vendor ID to no avail

FranticRain, 2016-05-29:
For some reason, the listing in the driver is for Vendor: 0x24c6 and Product 0xfafc. Here's a version that should actually contain this fix. Let me know how it works.

https://drive.google.com/file/d/0B5PLCKoMd4zGNmFnUDEwbFB5T2M/view?usp=sharing

maxblaster5000, 2016-05-30:
Success! Thank you so much!

closed 2016-05-30 by FranticRain