360Controller/360Controller #286: Controller not working

SirSteven2005, 2016-05-30:


SirSteven2005, 2016-05-30:
Afterglow Gamepad for Xbox 360:

  Product ID:	0x0213
  Vendor ID:	0x0e6f
  Version:	1.00
  Serial Number:	0693BCBB
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14400000 / 2
  Current Available (mA):	500
  Current Required (mA):	500
 some other stuff

SirSteven2005, 2016-05-30:
Please help

SirSteven2005, 2016-05-30:
It is glowing and everything but when I try to play games with it, it does not work!

SirSteven2005, 2016-05-30:
I am trying to play Castle crashers, Terraria ,and Battle Block Theater

SirSteven2005, 2016-05-30:
but it never works

FranticRain, 2016-05-30:
Check the controller in the preference pane. You can find this in the Apple Menu -> System Preferences -> Xbox 360 Controllers. Can you see movements and button presses in the preference pane?

SirSteven2005, 2016-05-30:
yeah
i can

SirSteven2005, 2016-05-30:
yes

FranticRain, 2016-05-30:
Well, that means that the controller is working. We cannot fix issues with games, you have to discuss that with the developer of the game. Terraria doesn't support controllers on Mac. For games made by "The Behemoth" such as Castle Crashers and Battleblock Theater, check this steam forum thread for support: https://steamcommunity.com/app/238460/discussions/1/523890046864680624/

FranticRain, 2016-05-30:
Duplicate of #187

closed 2016-05-30 by FranticRain

SirSteven2005, 2016-05-30:
but i don't see the preference pane area this is what i see [
[image]
](url)

SirSteven2005, 2016-05-30:
by the way it is not just three games it is everysingle game I own on steam

SirSteven2005, 2016-05-30:
and Terraria does support controllers with mac due to the 1.3.1 update

FranticRain, 2016-05-30:
If you see the movements in the preference pane, which is what that screenshot is, the driver is working. One easy way to test the driver outside of that is to open "Steam Big Picture." Once Big Picture is open, navigate to the settings and set up your controller. Then it should work to navigate Big Picture. This will prove that the driver is working properly. Once again, we CANNOT fix issues with games. You have to work that out yourself.