360Controller/360Controller #287: Thrustmaster Ferrari 458 Italia

kubafelix, 2016-05-31:
Hey, I noticed that you added the Thrustmaster Ferrari Vibration GT Cockpit 458 Italia Edition based on that issue: #173 . Could you please add also Thrustmaster Ferrari 458 Italia device? It's the older version of that model. 

Ferrari 458 Racing wheel:

Product ID: 0x5b00
Vendor ID: 0x24c6
Version: 31.22
Serial Number: 9364BDE
Speed: do 12 Mb/sek.
Manufacturer: Thrustmaster
Location ID: 0x14100000 / 3
Current Available (mA): 1000
Current Required (mA): 0

FranticRain, 2016-05-31:
Here's a build with that device added. Let me know how it works out. https://drive.google.com/file/d/0B5PLCKoMd4zGcG1aZ2xFbHhIZEU/view?usp=sharing

closed 2016-05-31 by FranticRain

kubafelix, 2016-05-31:
It helped, thanks!