360Controller/360Controller #289: Analogue LT and RT as Digital

de5perate, 2016-06-04:
Great driver by the way.

Would you be able to add an option in the driver to enable using the analogue LT and RT triggers as digital inputs?

Why - I am trying to use my Xbox One controller in GZDoom and currently I am unable to  assign my LT and RT triggers to any function other than movement (analogue inputs) - as you can imagine I would prefer to use these for running/shooting or opening doors.

Thanks

FranticRain, 2016-06-04:
I'd be interested in adding a feature like this, but unfortunately it is going to require a substantial amount of space in the preference pane. Space we don't currently have sitting around. I'm terrible at manipulating constraints in Objective-C, so I don't usually touch the preference pane. However, there are some ways you can get the experience you desire without changes to the driver. You can have your controller output keyboard commands using:  [Joystick Mapper](http://joystickmapper.com/), [Enjoyable](https://yukkurigames.com/enjoyable/) or [Enjoy2](https://github.com/fyhuang/enjoy2). This will make the triggers digital buttons, and should help solve your problem, for now.

tijames, 2016-06-14:
I was actually thinking about this for a while. Its an annoying issue wit certain games. An example would be Unepic. It took some experimentation to get the game to recognize the triggers (holding them half way down so they down overlap with other buttons). If they were digital inputs, this would solve the issue entirely with plenty of games, but my break other games the require it. So maybe just an on/off toggle in the preference pane if you can find a way to do it? Thanks so much.

FranticRain, 2016-06-14:
The trick about this is that it will require hot reloading of the HID report descriptor. It is something we tested before with the "Pretend 360" option for Xbox One controllers and it works, but it is far from clean. It can lead to some weird internal issues with OS X/macOS. Nothing a little reboot wouldn't fix, but its still not as clean as I would like. This option would also break some implementations of controller support. Not having it enabled should work normally, but as soon as it is activated, I assume any Unity game would absolutely have a fit.