360Controller/360Controller #290: [improvement] Adding an option to "square the circle" (outer deadzones)?

daniel-jozsef, 2016-06-13:
Hey, I'm wondering how hard it would be to add a "squaring the circle" option, ie. an outer deadzone setting. The Windows joystick calibration tool does this automatically by asking the user to touch the "corners" of the stick.

The analog sticks of the XB360 are circular, which means they cannot reach the corners (eg. [0,0] or [MAX,MAX]). This is okay in, say, third-person shooters, but can be an issue in simulation.

So by designating an outside deadzone where we allow the axes to max out, we could get independent full range on both axes on the analog sticks. I think this could be especially useful together with the "normalize" option.

FranticRain, 2016-06-13:
Enough requests have stacked up to get me to implement a new tab in the preference pane. I think I'm going to call it "Axes" and it will include fixes for this, #289, #153, and #140.

The way I imagine this working is "Normalize Maximums" button. This will require you to move the joystick into a corner position with one hand, and then press a "confirm" button with the other once the joystick is in a corner position that you feel comfortable with. You will have to do this for all four corners. Then, it will normalize all controller input to this range. So let's say you have a deadzone of 1,000, and a corner value of 30,000, then all inputs will be normalized using this range. Values less than 1,000 will be zero, values at or higher than 30,000 will be set to maximum (32,767 or -32,768). You'll lose a little bit of accuracy with this enabled, but it seems like the best way to implement this feature without fundamentally changing the feeling of the joysticks.

Alternatively, we can provide a slider like the current deadzones, and have it start at a maximum value. We could call it a "maxzone." Anything that exceeded the maxzone would automatically be set to the maximum value. Then, when normalize was activated, the input would be normalized as previously described.

Which of these implementations would you prefer?

FranticRain, 2016-06-15:
In addition, I'm having some issues understanding how the rest of controller operates after this squared circle implementation is completed. Do vertical or horizontal movements reach full before the stick has reached the maximum point? Or does this only accelerate movement in diagonal directions to counteract the circular shape of the joystick holes in the controller?

daniel-jozsef, 2016-06-16:
Hey :)

Okay so as far as I can tell, the Windows driver does the simplest thing imaginable. It just cuts off the outer part of the axes, and they max out at the corner position independent of each other. This should be the simplest to implement, and could be done in a similar way as the deadzone setting is implemented currently.

If you are interested in going further, it MIGHT be worth the trouble implementing a nonlinear transformation of the entire range of the analog sticks, but I'm not entirely sure. The Windows implementation feels good enough most of the time. Also, I'm afraid a nonlinear transform would mess up the intuitiveness of control.

Still if maximalism is the driving force, then probably the user should be asked to draw a full circle with the stick, and then wiggle it around in the deadzone, and the software taking this potato shape, and deciding what kind of transform would be the most useful.

FranticRain, 2016-06-16:
Here's a demo of how this works with the radial deadzones. The red circle is the inner deadzone, and the blue circle is the outer deadzone. Is that what you are looking for?
[image]

daniel-jozsef, 2016-06-17:
Lol no. Definitely no. That jump looks like an issue. It should be smooth. The available range from corner position to corner position should be stretched over the entire range.

That's why I was talking about a per-axis (square) outer deadzone, I don't know how this could be circularized effectively...

FranticRain, 2016-06-17:
So this is what you want:
[image]

daniel-jozsef, 2016-06-17:
Yep, that's pretty much it. Not hitting the corner dead on is not a major issue, as long as it gets close enough. :)

FranticRain, 2016-06-17:
Okay, it might take me a day or two to clean this code up and get everything put back together, but assuming everyone is on board with radial deadzones, this should be out pretty quick.

FranticRain, 2016-06-18:
Here's a build with this feature added. Let me know how it works for you. https://drive.google.com/file/d/0B5PLCKoMd4zGM2Riam1lYmF4WTA/view?usp=sharing

FranticRain, 2016-07-17:
Closing this due to inactivity.

closed 2016-07-17 by FranticRain

DarKcyde0, 2016-07-26:
I'm trying to accomplish this exact same thing for an unrelated joystick project.  I would love to see the code you used to do this.  It doesn't appear to be committed yet?

FranticRain, 2016-07-26:
It is all committed to my repository. I never pushed it upstream and it will never go upstream. You can find it here: https://github.com/FranticRain/360Controller/commit/23bfc5ac8bcf248c6fb3911f5e8e0380f176bf14

I don't have access to math.h in the kernel, so most of this code is useless garbage you shouldn't touch with a 10 foot pole. There are many different equations to get what you want, so check out this site: https://squircular.blogspot.com/2015/09/mapping-circle-to-square.html

Also note that the output of a controller isn't actually a circle. Its more of a rounded rectangle.

daniel-jozsef, 2016-07-27:
Lololol. ;) I wouldn't touch that code with a 10 mile pole tbh. For the sake of clarity though, implementing a nonlinear transformation never really seemed like a good idea to me either. As I mentioned above, the requirements would be met by simply cropping the useful area to a rectangular one.

Anyway if you really don't want to add this, I can live with that. We have userspace software for axis calibration.