360Controller/360Controller #291: XBone getting BT support

bennyp123, 2016-06-15:
heads up
https://www.rockpapershotgun.com/2016/06/14/xbox-one-controller-bluetooth-support-pc/#more-375021

FranticRain, 2016-06-15:
I would have preferred if we could have avoided this issue until September when the controller actually released, and people actually could work on supporting it. But I guess you got first dibs, so this is the issue. In the meantime, I'll be looking into ways to sniff Bluetooth packets, but knowing Microsoft, the protocol is going to be as archaic and obtuse as possible, just like with the wireless dongle.

For the internal purposes of this driver, this controller will now be referred to as the "2016 Xbox One Controller." As for distinguishing features, we will have to now adjust how we ask about controllers.
2013: Devoid of a 3.5mm jack.
2015: Has 3.5mm jack, Xbox button surrounded by glossy plastic (glossy plastic is separate piece)
2016: Has 3.5mm jack, Xbox button surrounded by matte plastic (body is a single piece)
There are no other obvious physical differences between the 2015 and 2016 controllers. Hopefully this won't prove to be too much of a hassle.

First and foremost with this new controller we will need a VID and PID. Hopefully Microsoft doesn't change the wired packet again with this release, but wired support should be trivial given how many times I've had to do this already.

Bluetooth is completely outside of the spectrum of this driver. Its been a USB driver for its entire existence. I'll see what I can come up with by designing a portion of the driver to connect to a Bluetooth Low Energy compatible Arduino device to save time before the controller launch. I'd definitely like to know for certain what protocol this controller ends up using, whether it is Bluetooth Low Energy or Bluetooth Classic. Maybe its even leveraging Bluetooth 5.0? That'd certainly be something. But given the fact that these controller will have to work with Scorpio, pushing 5.0 with the S might be a smart call. It'd be nice if anyone seeing this that has connections at E3 could ask some Microsoft representatives what protocol it is using, but I'm not sure they'd want to tell you even if they knew.

Something I've been seeing a lot of lately is that Bluetooth on Xbox One controllers would mean being able to connect your iOS device to a controller. Let me just say with the utmost certainty: You people have no idea what you are talking about. iOS only works with MFi certified controllers. Microsoft will not MFi certify their controllers. Because in order to be MFi certified, the controllers would have to follow a standardized HID protocol. If they did this, they would lose Windows 10 hooks. Although, Windows didn't support Bluetooth Low Energy until Windows 8, so the controllers can pretty much only work on 10 for actual technical reasons, not an arbitrary limitation. If Microsoft did actually standardize their protocol, that would be awesome. It would mean that this driver would no longer be needed, and everything would be beautiful and happy. Of course, if you jailbreak your device, you can use non-MFi certified devices like the DS4. I don't know much about jailbreaking and whether or not it will let you load kexts or not, but either way I have absolutely 0 interest in working on that.

As for Android, it sure doesn't seem like it will work on stock either. The controller would have to follow one of the protocols that Android supports, which of course is super duper unlikely. So there are the technical reasons why that pipe dream probably isn't going to happen. Please note that I know very little about the technical backends of iOS and Android, so I could be wrong, but on the surface level, this all looks improbable.

Once again, its highly likely that the Bluetooth wireless protocol packets will be identical to that of the wireless dongle. We still haven't cracked that nightmare of packets yet, so please temper your expectations. Don't think that because you hear "Bluetooth" and your computer has a Bluetooth radio in it that it will "just work." Your controller doesn't "just work" when you plug it in via USB, so it won't do that here. Please always have patience with us because this stuff isn't easy, we don't get paid to do it, and its honestly boring work, at least for the wireless protocol. Wired is way more fun.

If anyone has any additional concerns or questions, I'd love to hear them and I'll do my best to respond.

daniel-jozsef, 2016-06-17:
Even with a standardized protocol, the built-in drivers are just too "low budget". Having a full-featured stuff like this driver is a great thing.

Eg. the Dualshock 4 "works out of the box", but I kinda wish it didn't, because then people might build a driver, with deadzone adjustment, support for the touchpad, and similar crap. ;)

MaddTheSane, 2016-06-17:
It is possible to write drivers for bluetooth controllers for OS X. There's even an [Ouya driver](http://edmund.birotanker.com) (it's just to fix a problem with older firmwares where the cursor jumps to the top left corner).

Doing this for a bluetooth Xbox One controller would be a bit harder, although I don't think it'll be impossible.

RodrigoCard, 2016-06-18:
@FranticRain  AFAIK, iOS can connect to arbitrary BT low energy devices (and *only* BTLE devices) via public APIs. If this controller is BT 4.0 or greater it could be the first non MFI certified device to work on iOS without jailbreaking.

FranticRain, 2016-06-18:
@RodrigoCard You're absolutely right. I don't know how I forgot this. Heck, I've written that code. I keep forgetting that they just need to hack a connection together if all they want to do is game stream. It won't be a viable controller for iOS and Android games in general though.

FranticRain, 2016-07-30:
Looks like some support pages have gone up on the Microsoft Xbox support site regarding Bluetooth capable Xbox One controllers. Most important notes: They claim that Bluetooth connected controllers will not support ANY accessories and they claim only one controller can be connected via Bluetooth at a time to a Windows PC. Doesn't shed any light on Bluetooth classic vs low energy, but it is still an interesting read. https://support.xbox.com/en-US/xbox-on-windows/accessories/connect-and-troubleshoot-xbox-one-bluetooth-issues-windows-10

RodrigoCard, 2016-07-30:
Looks like they still need to sell wireless adapters :P

sakuranbokun, 2016-08-03:
I've just got one of these new pads. It connects to the Mac just fine (they're bluetooth devices after all), and it works a treat in OpenEmu. With Steam games though, it's kind of a disaster. Sleeping Dogs works partly, but it reads the R trigger incorrectly. Every other game I tried didn't see the pad at all.

The Xbox Controller Settings panel can't see the pad.

RodrigoCard, 2016-08-03:
So it means it works like regular gamepads. Which is partially good news.

FranticRain, 2016-08-03:
@sakuranbokun Make sure to go into Steam Big Picture -> Settings -> Controller and bind the controller properly to make sure its all set up. That might not fix Sleeping Dogs, but I'd like to know if it is correctly set up by default. Also if you could do me a solid and plug it in via USB to get the vendor and product IDs, that would be great.

So if they work by default, I guess we close this down?

MaddTheSane, 2016-08-03:
Does it have rumble support?

MaddTheSane, 2016-08-03:
What about Force Feedback? Does it have that out of the box?

If not, a simple wrapper to point to the feedback plug-in might be all that’s needed to support it.

RodrigoCard, 2016-08-03:
I also would like to know if it works wired through usb

sakuranbokun, 2016-08-03:
Some more info - Steam big picture worked with the pad without any configuration, but the buttons were messed up, so A was B, and B was Y. I went into Big Picture controller configuration and I was able to set everything up as it should be, except for the Guide button which Steam couldn't read for some reason. I fired up a few games which hadn't worked before and they were still broken. 

I tried plugging it in as a USB controller (I disconnected it from Bluetooth first) and this is what the Mac sees - 

 Product ID: 0x02ea
          Vendor ID: 0x045e  (Microsoft Corporation)
          Version: 3.01
          Serial Number: xxx
          Speed: Up to 12 Mb/sec
          Manufacturer: Microsoft
          Location ID: 0x14200000 / 10
          Current Available (mA): 1000
          Extra Operating Current (mA): 0

Although the controller vibrated when I plugged it in through USB the Guide button didn't light up and nothing could see it as a controller. It was 100% dead in everything. I re-enabled Bluetooth, and it worked again.

As for force feedback, I only played Sleeping Dogs for a few minutes and I didn't notice any rumble going on. I'll play more of it later. Also I tried a few more games - Borderlands The PreSequel, Tomb Raider, and Hitman, and none of them could see the pad. Curiously Costume Quest 2 worked fine though.

Perhaps some games explicitly want to see an Xbox pad, whilst others will accept any controller?

FranticRain, 2016-08-03:
Your assumption at the end there is correct. Some games only explicitly look for this driver, not normal controllers. That is a failing of the games though, not ours.

I'll get that VID and PID added to the driver and then it should work wired through the driver.

Its interesting that the Xbox button doesn't work in Bluetooth mode. Its probably because they are doing the same thing they do in wired mode, which is to send out a different packet for guide button presses than they do for normal button packets.

I wish we could find out if this was Bluetooth Classic or Low Energy. I don't know of any easy way to tell without having to download some third party tool.

sakuranbokun, 2016-08-03:
I don't mind helping out, if there's some tool you need me to grab that's not an issue. 

Do you think it would be possible to get it working as a normal Xbox pad over Bluetooth? I bought it to be free of wires to be honest!

FranticRain, 2016-08-03:
Here's a build that will work wired. New device is called MicrosoftXboxOneController2016. Hopefully they never manage to release two controllers in the same year. https://drive.google.com/file/d/0B5PLCKoMd4zGLURZQ0JGU0ZJZmc/view?usp=sharing

I just checked, and you can totally tell without some other tool. When the controller is connected via bluetooth, go to that same "System Report..." area you were in before is. Under the Bluetooth section, there should be a bunch of information about paired devices. I'd be interested to see all of it, but the "Class of Device" entry is the most important one.

Getting it working through the driver over Bluetooth is not a simple task. We may look into it at some point, but thats not a small undertaking.

sakuranbokun, 2016-08-03:
Quick update - I tried the new driver, thanks for that! Sadly though it didn't work. It looked promising at first, as the Xbox Guide button lit up and in the Xbox 360 Controllers preference panel it found the pad and reported it as an Xbox One Wired Controller. I then headed into Steam and tried Borderlands, Hitman, and Tomb Raider again, and none of them detected the pad.  

Back in bluetooth mode I grabbed the info you were after:

 Xbox Wireless Controller:
  Address:	C8-3F-26-13-79-B1
  Major Type:	Peripheral
  Minor Type:	Peripheral
  Services:	Hands-Free unit, Xbox Bluetooth Gamepad
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Unknown (0x8, 0x00)
  Firmware Version:	0x0903
  Vendor ID:	0x045E
  Product ID:	0x02E0
  Class of Device:	0x05 0x02 0x0508
  AFH:	On
  AFH Map:	000FCFFFFBFFFFF7F
  RSSI:	-47
  Role:	Master
  Connection Mode:	Sniff Mode
  Interval:	7.5 ms
  Host Connectable:	Yes
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes

FranticRain, 2016-08-03:
Well. I'm guessing that means that it is Bluetooth Classic. Because mine explicitly said "Low Energy."

So when you say the guide button lit up, do you mean it went solid? No flashing? And did the preference pane correctly read inputs from the controller?

sakuranbokun, 2016-08-04:
Yep, the guide button light was solid, and the preference pane was reading it correctly. I jiggled the sticks to test it and they moved on screen as they should. 

Any idea why games weren't picking it up?

MichaelMKenny, 2016-08-04:
I just purchased a Xbox One S Wireless Controller and can confirm your custom build did correctly read the inputs (and rumble, although the "Rumble Options" was missing some the options the regular Xbox One controller has). 

But steam wasn't recognising it as an a xinput controller.

FranticRain, 2016-08-04:
If the preference pane understands the controller, its probably the games that are the issue. I'll check out those games with a regular Xbox One controller and 360 controller once I get back from school in a couple hours to see if they are the issue or not. I don't have your exact titles so I'll be testing: Hitman: Absolution, Borderlands 2, and Tomb Raider (2013).

I'm not sure why those rumble options weren't in the preference pane. I'll make sure to check why that is happening.

XInput is a Windows concept. It doesn't exist in OS X. The best way to get Steam to treat this new controller like a 360 controller is to use the "Pretend to be an Xbox 360 Controller" option in the preference pane. Valve programs in all of their own layouts for controllers, so they just haven't pushed out an update for a custom layout of this controller.

FranticRain, 2016-08-04:
Here are my findings. "Xbox One" is a 2015 Xbox One controller. "Pretend 360" is a 2015 Xbox One controller with the "Pretend to be an Xbox 360 Controller" checkbox checked. "360" is a wireless Xbox 360 controller connected using a wireless adapter.

| | Xbox One | Pretend 360 | 360 | Rumble? |
| ----- | ----- | ----- | ----- | ----- |
| Hitman: Absolution | No | Yes | Yes | Yes |
| Borderlands 2 | Yes* | Yes* | Yes* | No |
| Tomb Raider | No | Yes | Yes | Yes |
*Had a drift to the right axis that is caused by the game.

All of these are Feral Interactive ports (including Sleeping Dogs). You might want to look at their page regarding using controllers, see as the Bluetooth version of this controller will be treated as a driverless controller. For Tomb Raider and Hitman, there are menus before the game starts that will help you figure out if the game recognizes the controller in the first place.

Unfortunately, I own the original version of Sleeping Dogs, not the Definitive Edition, so I can't test that game on OS X.

@MichaelMKenny There is nothing that jumps out at me as to why those options aren't there. This should be picked up as an Xbox One controller and should be handled properly. I'm not sure what the problem is here.

@MaddTheSane @RodrigoCard Is there even such a thing as a Bluetooth device-specific kernel extension? I can't find any documentation on how that would work.

MaddTheSane, 2016-08-04:
[This repo](https://github.com/edmundmk/ouyacontrollerosx) is a driver for the Ouya controller.

FranticRain, 2016-08-04:
Oh yeah, you mentioned that before. Sorry for forgetting. Since I don't have one of these, I'm going to try to get the driver working with a Dualshock 4 first. I don't know if we'll actually want to keep any of that code, but it should let me test if games will see it as a fake 360 controller.

MichaelMKenny, 2016-08-05:
Using "Pretend to be an Xbox 360 Controller" works. And I sort of knew that Xinput was a windows thing, but I was really talking about the label Steam Big Picture uses when it sees an Xbox controller.

Here is Xbox One S controller using the pretend thing in big picture:
https://dl.dropboxusercontent.com/u/8225228/Xinput%20controller.png

FranticRain, 2016-08-05:
Yeah, that is just Steam's way of saying "This is a whitelisted Xbox 360 controller vendor and product ID combination."

sakuranbokun, 2016-08-05:
Where is the pretend to be an Xbox 360 option? Don't think I saw that one when I was configuring the controller in big picture.

RodrigoCard, 2016-08-05:
Look at the bottom of the advanced tab.

-- 
Enviado do Gmail para celular

FranticRain, 2016-08-05:
That isn't an option in the Steam Big Picture preferences. Its an option of the driver. You can find it in the "Advanced" tab of the Preference Pane. Apple Menu -> System Preferences -> Xbox 360 Controllers -> Advanced. The option is in the bottom right hand corner.

FranticRain, 2016-08-05:
Okay. Here is an update on the bluetooth driver progress. I've managed to get the Dualshock 4 to pair with the driver. There are some quirks to this that I don't understand fully, but I think it is fine. It is now spitting out the Xbox 360 wired controller name as well as VID and PID. I have no idea if this will actually do any good though, since I assume most games will be looking for a USB device, not a bluetooth one. Just spoofing the VID and PID might not be good enough. The main problem now is that the controller bypasses the `handleReport` function. It looks like the `IOBluetoothHIDDriver` doesn't have a `handleReport` function. This is a bit of a surprise, since I expected this to work very similarly to `IOUSBHIDDriver`, the thing Xone-OSX uses. `IOBluetoothHIDDriver` has `getReport` and `getReportWL` but I can't get those to give me anything either. Anyone have any ideas?

Also, I'm going to need to find a game to test against at some point that is very specific about only working with Xbox controllers. A sort of trial by fire, if you will. I don't know of any games that fit the bill off of the top of my head, so I'd love some suggestions there, as well. If you want to check it against my Steam library, [you can find that here.](https://steamcommunity.com/id/FranticRain/games/?tab=all)

sakuranbokun, 2016-08-06:
I've just tried "Pretend to be an Xbox 360 Controller" and it works in Steam games just fine. Good to have the controller working fully, but I really hope you can get it working over bluetooth too. That really would make it the go to controller for OS X.

FranticRain, 2016-08-06:
I can view controller packets in `processInterruptData` but it doesn't look like it will let me mangle them at all.

Using a GOG copy of Don't Starve, I was able to make the DS4 work with the game using the driver, when it did not work before. So that is promising, at least. There is a problem with the trigger axes making it impossible to properly bind the controls, so I'd definitely like to fix that, at least. That may just end up being a custom HID report descriptor to match the button and axis layouts of the Xbox 360 controller. Beyond that, I'm not sure there is much more I can do with this. I'll put together a report descriptor for the DS4 later and give it a shot. Unfortunately, I haven't seen any documentation on the 2016 controller's bluetooth packets yet, so I can't do the same for the 2016 controller at this point.

FranticRain, 2016-08-07:
Turns out I'm actually an idiot. You can totally edit the packets in `processInterruptData`. Just as it turns out, using the USB HID format for your report descriptors for Bluetooth HID report descriptors is just as dumb as it sounds.

FranticRain, 2016-08-09:
Okay. Here is a SUPER experimental build that *should* pair up to the controllers via Bluetooth. The preference pane WILL NOT recognize the controller as connected. The main idea here is to check the controllers with games with work with the controller wired, but not via Bluetooth. I want to know if this does anything to help with that. If it doesn't: I scrap this, we add a force feedback update for it and it works with what it works with. If it does: I'll keep looking into this.

https://drive.google.com/file/d/0B5PLCKoMd4zGTlpZSWtKbTBkR3c/view?usp=sharing

sakuranbokun, 2016-08-09:
Just taken your experimental driver for a spin. The results are interesting.  Here are the games I tried, all of which worked with the previous driver release when wired, but not at all with either no driver or the previous official driver release.

Hitman Absolution (Feral) - does not see pad
Tomb Raider (Feral) - does not see pad
Borderlands Pre-Sequel - Works fine
Guacamelee - Works fine
Outland - Works fine

I also tried Flame and the Flood, which I hadn't tried with this pad before. It partially worked, but the "back" button acted erratically, rapidly firing when held. Oh and I took Sleeping Dogs for a spin again, and it was unchanged from before, so it all works fine except that the R trigger is broken.

FranticRain, 2016-08-09:
Thanks for the feedback. I've got one more crazy idea I want to try. These currently pretend to be Xbox 360 Controllers, but what if they pretended to be a Dualshock 4? With those three options, we can probably get this thing into the best shape possible to resolve issues. I'll get a build out that does that in a little bit.

As for the Flame and the Flood, make sure you check that one wired to see if that back button thing is just the game. It could be some weird way Microsoft sends packets when "View" is pressed on an Xbox One controller. That would differ from how it reacts wired though, so I think that is highly unlikely. The current driver you have installed should work with the controller wired as well, so you don't need to reinstall. Just make sure it doesn't just suck power from USB to run a Bluetooth connection. I doubt it will, but I just want to make sure.

Sleeping Dogs may be a similar issue. I've made the controller look like a 360 controller, but haven't actually bothered to change anything about how it describes itself or how it orders the packets. For this driver, we say that the left stick is axes X and Y. Right stick is Rx and Ry. Triggers are Z and Rz. But most standardized HID pads treat it as: Left Stick - X, Y. Right Stick - Z, Rx, Triggers: Ry, Rz. So its likely something is getting mixed up in there. Or it could be that the trigger is treated as a single axis by the controller. WHO KNOWS? I sure don't! Because I can't see the packets!

Speaking of which, I'll put out a special version of the driver that attempts to "sniff" packets. It'll horribly clog up your Console.app when you use it, so don't keep it installed.

FranticRain, 2016-08-09:
Okay. Here is the one that looks like a DS4: https://drive.google.com/file/d/0B5PLCKoMd4zGcWJaMFFWN2U3bTg/view?usp=sharing
It does not re-order the packet, so the Feral games may see it in the launcher and then not work at all in game. If that happens, that is good. If it works perfectly, thats better.

Here is the sniffing one. When you install this one, before you connect, open up "Console.app" from "Applications/Utilities". When you connect it, press the A button a couple of times and then turn off the controller. Your Console.app will now be filled with a bunch of garbage that says: "Packet - Blah blah blah". I'm going to need all of that. Then probably reinstall a different version of the driver over the sniffing driver because I don't want your log file to be huge. https://drive.google.com/file/d/0B5PLCKoMd4zGOEp5NUJzMzFxQ2c/view?usp=sharing

sakuranbokun, 2016-08-09:
Another update.

I tried Flame and the Flood wired and it worked perfectly. I realised when playing it that the game doesn't even use the back button, it's supposed to be the L trigger. With the pad in bluetooth mode the game doesn't recognise the L trigger and has someone mapped that function, in a broken form onto the back button. The start button, which should be menu/pause, it also broken when in bluetooth mode, but works fine wired.

Now onto the new drivers. The DS4 one was a failure I'm afraid. I tried a couple of Feral games - Tomb Raider and Hitman. Neither could see it in the launcher or in game.

Then I tried the packet sniffing driver, and this too was a failure. Rather than fill up the console when I connected and mashed a few buttons, all I got was a few lines back - 

09/08/2016 20:43:33.353 blued[93]: Device Connected with address: c8-3f-26-13-79-b1 and classOfDevice 0x000508 and connectionHandle 0x000c
09/08/2016 20:43:33.363 blued[93]: HID Device connected isIncoming = 1, deviceClassMinor = 2, isPaired = 1, isConfigured = 1
09/08/2016 20:43:33.499 blued[93]: Trigger core audio to connect to bluetoothaudiod
09/08/2016 20:43:45.464 ApplicationManager[408]: [CopyAvailableUpdate] Software is already up to date
09/08/2016 20:43:45.464 ApplicationManager[408]: No updates available
09/08/2016 20:43:52.000 kernel[0]: [HID] [ATC] AppleDeviceManagementHIDEventService::handleInterruptReport Ignoring interrupt report id = 0x000000c6, length = 4

FranticRain, 2016-08-09:
Hmmm. That is extremely confusing. I just tested the packet sniffing version with my DS4 and it absolutely spammed me with reports. I guess I still don't get to know what the packets look like.

That Flame and Flood issue makes perfect sense. If I reordered the packet, that *might* be solved.  I still need to know what those packets look like. I can't possibly understand why that didn't work.

DS4 spoofing being a failure is disappointing, but it makes things simpler for me at least.

Ultimately, I need to know what those packets look like to continue. Its likely that its not worth anyone's time but ours to actually sniff that thing, since it works fine with Linux by default, no one is going to be chomping at the bit to document it. Not to mention, this is just a game support hack. This is actually a problem that the game developers should be fixing. We shouldn't have to have a driver at all. At this point, I'm glad we can trick some games into seeing the thing, but I think that responsibility may need to rest with an external program, outside of 360Controller.

If everyone just used DDHIDLib we wouldn't have this problem. /s

FranticRain, 2016-08-10:
@sakuranbokun Did you get any force feedback playing any of those games over Bluetooth with or without the driver?

sakuranbokun, 2016-08-10:
No, no force feedback at all over bluetooth.

FranticRain, 2016-08-10:
Okay, so I guess we need to add a force feedback wrapper and call this done. Getting this controller recognized by games sounds an awful lot like a job for game developers, Joystick Mapper or Enjoyable. We can't make a Bluetooth device look like a USB one within the kernel, and even if would could, that sounds like it is out of scope.

@MaddTheSane Will you take care of that force feedback thing? I'm not too familiar with that stuff.

MaddTheSane, 2016-08-10:
@FranticRain ~~This is my initial attempt at making the bluetooth controller work with the Force Feedback framework: https://github.com/MaddTheSane/360Controller/tree/bluetoothCodeless~~

I just remembered that the feedback driver is quite exclusive and won't load on anything else other than those implemented by the 360 driver.

A simple hack should be coming soon on that branch, though.

FranticRain, 2016-08-10:
@MaddTheSane Thanks for working on this.

FranticRain, 2016-08-15:
I went ahead and built/signed @MaddTheSane's work. Give it a shot and let us know if you get rumble through the gamepad over Bluetooth with this installed. https://drive.google.com/file/d/0B5PLCKoMd4zGallWc1NkLTVNTDA/view?usp=sharing

sakuranbokun, 2016-08-16:
I've just tried to give this a go but the results were rather alarming. I installed the driver and rebooted, then I tried to reconnect my pad. It wouldn't connect. So I rebooted and tried again, and again, and again. It wouldn't connect at all. I removed it from the bluetooth control panel and attempted to re-pair several times, but with no success. So I uninstalled the driver (by right clicking on the control panel and selecting remove) then rebooted, and the pad still won't connect.

Could be a coincidence, who knows, but it was working fine before the latest driver. Any thoughts on how I can fix it?

FranticRain, 2016-08-16:
That build introduces a new file that probably wasn't deleted by the uninstaller. Look in "/Library/Extensions" for "XboxOneBluetooth.kext" and delete it. The best way to get to this folder is to open Finder, press Ctrl+Shift+G and then type in "/Library/Extensions" into the window that appears.

sakuranbokun, 2016-08-16:
No sign of that file in the Extensions folder unfortunately.

FranticRain, 2016-08-16:
I had an issue like this with the Dualshock 4. The fix is a little convoluted. Lets see if I can recount it correctly.

Turn off Bluetooth. Open the terminal and type in: ```sudo rm -rf /Library/Preferences/com.apple.Bluetooth.plist```. Then restart your computer. Once you have restarted, navigate to "Library/Preferences" using that Ctrl+Shift+G trick and open the newly formed com.apple.Bluetooth.plist in your text editor of choice. Delete all of the contents of this file and save it. This will unforunately delete all of your pairing history for Bluetooth, but it fixed my similar issues.

sakuranbokun, 2016-08-16:
Terminal kicks me out with a "permission denied". I have admin rights.

FranticRain, 2016-08-16:
Navigate to it in finder and delete it there then. Make sure nothing is using Bluetooth and that it is turned off.

sakuranbokun, 2016-08-16:
No joy unfortunately, it's completely dead. I also tried using the debug options in the bluetooth menu to forget all devices.

sakuranbokun, 2016-08-16:
The Mac connects to the pad for a fraction a second, then immediately disconnects, then connects again, the disconnects and so on. Looking at the Bluetooth menu I see I have two "Connect" options.

Very odd.

FranticRain, 2016-08-16:
If you've reset Bluetooth using and method and you've deleted that file, then I don't know what the problem could be. Maybe if you reset the controller by plugging it in via USB to a version of the driver that will work with USB connections. I don't know if that will work, but thats the best I can think of.

FranticRain, 2016-08-16:
Here is a version where the uninstaller will uninstall the new Bluetooth force feedback kext. https://drive.google.com/file/d/0B5PLCKoMd4zGOUJvc0RlVWdYeE0/view?usp=sharing

sakuranbokun, 2016-08-17:
Thanks for that. I gave it a quick shot this morning before heading off for work but it sadly didn't make any difference. Perhaps it was a freak event or random coincidence that the Mac decided it no longer wanted to pair with the controller, who knows. It'll be headed back to Amazon shortly.

Luke-Filewalker, 2016-08-20:
Ok i tried it with Absolut Drift. Wired rumble works but via Bluetooth no rumble.
Also there is a problem with connecting the controller. It connects then loses the connection, connects again and repeats that way. After few tries i got it to connect and then it worked like always.

paulyd33, 2016-09-03:
@FranticRain I'm highly appreciative of the work that all persons input into this project. Personally when it comes to the programming side of things I lack the knowledge to be able to implement my ideas. In regards to the 2016 XBone Bluetooth compatible controller, is it possible to link this into an upcoming release of the driver? The only reason I ask is there is little support for the controller within MacOS games at this moment in time. And when I do play games, I like to be able to see the button prompts etc. that are used with the 360 controllers. Or if I could be given some knowledge as to how controllers and what not are added to the driver i'd be more than happy to have a crack at it myself and let you know how I go?

FranticRain, 2016-09-03:
@paulyd33 This controller already works via wired connections to the driver. If you want good compatibility with games, use it that way, and activate the "Pretend to be a 360 Controller" option. The Bluetooth version of the controller works with OS X natively. We will not be adding any additional features to the driver to support it in any way beyond a force feedback plugin. That is the only reason this issue is still open. Game support for the controller via Bluetooth is going to be spotty, but that's entirely on the game developers. We can't do anything about that. And as far as I know, we can't make a Bluetooth device act like a USB device, so we can't do anything about it. We've got bigger problems with stuff like macOS bugs causing wired controllers to not initialize properly, and supporting a controller that already works is pretty much the lowest possible thing on my list of "want to do."

[19 more replies]

PR #526 (2017-09-29): Bluetooth Force Feedback
This adds initial support for Force Feedback for the BlueTooth Xbox One controller.

Partially fixes #291
closed 2017-09-29 by MaddTheSane