360Controller/360Controller #292: Switch to Radial Deadzones - Discussion

FranticRain, 2016-06-15:
I'd like the trash the deadzone code and fully shift over to "scaled radial deadzones." Essentially, what this means is that the driver would start using a circular deadzone instead of the axial deadzone it uses by default. This would eliminate the need for the "Linked" option. Additionally, input would be automatically normalized, so there would be no need for a "Normalize" option either.

The objective is to eliminate all of the branches in the current deadzone code and remove confusing options in order to provide a better deadzone experience for the end user.

I'd like to hear opinions from others, so I'll leave this open for a few days before I make any commits.

slime73, 2016-06-15:
This is a (long, but) good read: http://www.gamasutra.com/blogs/RyanJuckett/20160329/268937/Interpreting_Analog_Sticks_in_INVERSUS.php

electric-monk, 2016-06-15:
It was a long time ago now, but I vaguely recall I made the deadzone stuff square and "linked" optional as half the games I tried it on only needed one axis deadzoned, apparently because they weren't really conditioning the input data in any form themselves (or they were, but somehow badly).

I vaguely recall an FPS where forwards/backwards were fine (pretty much "perfect" in terms of responsiveness), but the character would lazily rotate without the horizontal deadzone. I think I added the "linked" option later, since there were obviously games that had a problem with both, too, and it certainly seems reasonable that the "linked" mode might want radial instead.

FranticRain, 2016-06-16:
@slime73 Thanks a lot for that. I didn't know about the need for an outer deadzone. That's a great piece of information.

@electric-monk The way I view the deadzone options in the driver is: To provide basic and general conditioning for stick input when a game doesn't. Additionally, it could also be used if you had a really bad controller with an exceptionally high amount of drift. The idea behind making these circular deadzones is to provide a higher standard of basic input tuning. The issues you mentioned should be solved by a correctly configured radial deadzone.

All of that being said, it seems clear that "Normalize" should continue to be an option, not automatically implemented. This way, users can implement a deadzone if they have a drifting stick without ruining the careful tuning a developer might make for their game.

FranticRain, 2016-06-16:
After hacking through an implementation, its probably better to avoid radial deadzones, however great they are, because they require quite a bit of trig. It wouldn't be a big deal, but including math.h breaks the driver, so I had to write a custom implementation of square root. I got radial deadzones working great, and even got "Normalize" done really well. The only problem is that the outer deadzone requires arctangent, sine, and cosine. Because if a value exceeds the outer deadzone, it needs to be clamped to the unit circle. That would mean finding the arctangent of the current x and y values to get an angle, and then push those back through sine and cosine to get the new x and y values. I could probably do all of that, no big deal, but I don't know how much all of that trig would really matter. Ideally, I'd just want to hook into Intel's SVML and do the math properly, but I'm not sure OS X would let me hook into intrinsics in a kext like that.

daniel-jozsef, 2016-06-16:
I doubt anyone does radial deadzones... Handling each axis independently should be good enough. Maybe what would be an improvement is separate deadzone settings on the x and y axes of a stick, often you have a larger deadzone in one direction than the other.

FranticRain, 2016-06-16:
Radial deadzones are quite common and well regarded as the best implementation of deadzones among game developers. Every article I've read about deadzones declares radial deadzones as the gold standard. Usually, its really trivial to program too. But when you don't have any access to trig functions, it becomes exponentially harder. I have no interest in further complicating the deadzone system by introducing separate x and y deadzones. My original goal was to simplify the system so that users wouldn't feel overwhelmed by all of the crazy checkboxes and sliders that can suddenly alter when traditionally console gamers never see any of this kind of stuff.

daniel-jozsef, 2016-06-16:
Well PC gamers tend to get off on them though, haha... :D And console games just assume a pretty large deadzone, and rarely need (or have) the precision PC games do, especially in simulation and in competitive gameplay.

Anyway I think the most "user friendly" would be to have the user wiggle around the stick in what he considers the deadzone, and then the software taking this shape and doing its best.

FranticRain, 2016-06-16:
Here are some preview gifs of what the radial deadzones look like. I figured out how to do it using only square roots. I need to find my linear algebra professor and personally apologize for taking so long to remember how to normalize a vector.
Inner Radial Deadzone:
[image]
Outer Radial Deadzone:
[image]
Normalization of Radial Deadzones: 
[image]
(I haven't re-implemented the stick "true position" marker on normalized yet, sorry.)

FranticRain, 2016-06-18:
I pushed this changes to my local repo so you can check them out. I've built a version of the radial deadzone driver that people can try out here: https://drive.google.com/file/d/0B5PLCKoMd4zGM2Riam1lYmF4WTA/view?usp=sharing
Here are some gifs of the driver with radial deadzones.
[image]
[image]
[image]

FranticRain, 2016-07-17:
Closing this because it seems like there was 0 enthusiasm, and the people who supposedly wanted this stuff never got back to me.

closed 2016-07-17 by FranticRain

daniel-jozsef, 2016-07-17:
Erm... I totally thought this was going into the main repo, which is why I didn't pursue the question further. The experimental build works perfectly for me at least..

FranticRain, 2016-07-17:
Well, I do have a fundamental issue in that: 
1. Its a hack to fix a problem with a bad controller implementation.
2. It's pretty hacky because changes in deadzone size change how well the joystick tracks to the corner.
3. Its a fundamental change to how deadzones operate, which has reportedly been an issue before.

There is one issue that was similar to yours in #302, you are correct. However the user never responded. I have no affirmation that this fix was what he or she wanted. For my part, if a user never responds after an original issue post, I don't consider their issue valid unless it is simply an issue to add a device.

The more time I spend working on this driver, the more I realize that it shouldn't provide solutions to these game specific problems. I get increasingly frustrated that I  was responsible for adding the remapping feature to the driver. I think this is an issue that should be taken up by user space programs - ones that have real square roots they can use, not my Quake III inverse square root hacks.

First and foremost, this is a "game specific" issue. It might appear across multiple games, sure, but its indicative of a poor controller implementation. A driver should hand off information from the controller without mangling it, and a user-space application should be the one to apply changes to that input. Unfortunately, due to the fact that the OS doesn't support controllers particularly well, and game developers do a generally terrible job supporting controllers, this driver gets caught in the lurch. The real answer here is that you should complain to the developers of the game you are trying to play and let them know that their controller implementation is wrong. Its terrible and idiotic to pull raw number data off of a controller like that. And quite frankly, I don't want to support hundreds of lines of bad code in this driver to fix that fact that some idiot didn't understand how to properly write controller code.

RodrigoCard, 2016-07-18:
+1 for Quake 3 reference!

daniel-jozsef, 2016-07-27:
Okay I can understand that standpoint. If it's an ugly hack, then don't put it in. Controllermate is a viable solution, even if it can get complicated.

The main problem is that games (or rather, game frameworks and engines - most game devs these days don't bother with low level stuff) do a horrid job at supporting controllers, and that OSX also does a shit ass job. At least in Windows you can set the priority for controllers, so if you remap a controller in userspace, you can prioritize the virtual controller. Not in OSX.