360Controller/360Controller #293: NO DEVICES FOUND!!!

Planeguy102, 2016-06-19:
This crap doesn't even work! don't download it becuase it's just a waste of space. 3 hours of work and THE PROGRAM CAN"T SEE THE STUPID CONTROLLER! Xbox 360 conttroller is nowhere to be seen. horrible code, people!

FranticRain, 2016-06-19:
Are you attempting to use a wireless Xbox 360 controller connected to your PC via a Play & Charge kit? Or are you using some sort of third party Xbox 360 controller? We'd love to help you solve your problem, but we can't do that if you don't give us any information.

Planeguy102, 2016-06-19:
sorry i got so mad, but i can't seem to be able to move from the place where the PC detects it and doesn't know what to do with it. nothing except control panel can see it.

FranticRain, 2016-06-19:
If the pref pane can see it, and a game can't, open up Steam and go into Big Picture mode. There is a way to configure controllers in there. That system is super reliable in seeing controllers, so that should allow you to see that the controller is working with other programs. After that, if you still have trouble with a specific program, it is likely that the game does not support controllers.

Planeguy102, 2016-06-19:
nope, it can't see it. also, i can't get your program to install, and that may be my problem. how exactly do you do that on windows 8.1?

FranticRain, 2016-06-19:
Windows 8.1 automatically downloads a driver for the Xbox One controller. Simply plug it in. This driver is only for OS X.

closed 2016-06-19 by FranticRain

Planeguy102, 2016-06-19:
but i have an xbox 360 controller

FranticRain, 2016-06-19:
Sorry, yes. Windows automatically downloads the driver for Xbox 360 controllers as well.

Planeguy102, 2016-06-19:
then why won't anything work with it?

FranticRain, 2016-06-19:
I don't know, but I'm sorry. We do not support Windows Xbox controllers.

Planeguy102, 2016-06-19:
Ok. thanks for the help, anyway.

FranticRain, 2016-06-19:
If it is a wireless controller that you are connecting to the PC via a Play & Charge kit, then it won't work. The play and charge kit only provides power to a controller, it does not make a wireless controller wired.