360Controller/360Controller #294: Help with MacOS 10.11.5

pxlronin, 2016-06-19:
Hi, 

Has anyone been able to get  this driver working with MacOS 10.11.5?

I bought an XBOX One Controller/Wireless Receiver combo from Microsoft, thinking t'd be no big deal to get it working on my Mac, and well....I've had not luck so far. I've tried all drivers I've found and tried all suggestions recommended and no luck yet.

Thanks advance for your help.

Cheers!

FranticRain, 2016-06-19:
No one has cracked the wireless protocol yet. Microsoft purposefully encrypted the data it sends and receives to prevent knockoffs of the wireless receiver. If you plug the Xbox One controller in via Micro USB it should work perfectly though!

closed 2016-06-19 by FranticRain

pxlronin, 2016-06-19:
HI Drew,

Thanks for the reply. OK I’ll try that.

Cheers!

Joe