360Controller/360Controller #295: Driver works but games do not

krp312, 2016-06-20:
Hello,

Using 360 wireless controllers with the USB dongle. Controllers connect successfully and I can even see the buttons light up when I press them when viewing the PrefPane. But I can't get it to work with games like Samurai Gunn, or SNES emulators.

Thoughts?

Thanks,
Kris

FranticRain, 2016-06-20:
Unfortunately, not all games on OS X support controllers like they do on Windows. It looks like Samuri Gunn does not support controllers properly on OS X. You will need to use a tool to map controller inputs to keyboard buttons, such as Joystick Mapper or Enjoyable. As for the SNES emulator, I can't say for certain, but it is likely to have the same problem. If you are looking for a good set of emulators with impeccable controller support, I would highly recommend OpenEMU. It has a fantastic front end and works very well.

closed 2016-06-20 by FranticRain

krp312, 2016-06-21:
Thanks very much!

krp312, 2016-06-21:
So after thinking about it, it seems that I'd favor running everything on Windows. I was wondering if you had any recommendations on virtualization software, and an all-in-one emulator for Windows, because OpenEMU is rad and I had never heard of it.

FranticRain, 2016-06-21:
Unfortunately, there isn't anything as cool as OpenEMU for Windows. Its a real shame. You'll just have to download individual emulators on Windows. Other than that, I don't virtualize anything. I've got a good PC that I do most Windows work on, and if I need OS X, my Macbook is good enough for the job.