360Controller/360Controller #296: No devices found - Xbox One wired controller

somnolentsurfer, 2016-06-20:
As the title say, basically. I'm using a wired Xbox One controller, but the System Preferences panel is showing no devices found. Controller is connected fine, Xbox button is illuminated, and it works with my Boot Camp partition. Driver version 0.16.

FranticRain, 2016-06-20:
That is definitely odd. Before you used this driver, did you use any other drivers in order to get this controller to work? They could be interfering with this driver if they were not uninstalled correctly. This isn't a very common issue so I don't have any perfect solution for you. I want to make absolutely certain though that the light on the controller is a stead illumination, not a blinking illumination?

somnolentsurfer, 2016-06-21:
Yep, it's steady. I might have had an older version of this driver installed before... I did have one that worked on Mavericks. Since upgrading to El Capitan I'd not had one that worked at all, though some of the old fils may not have deleted properly, I guess.

I also discovered after posting this my wireless 360 controllers caused a complete system freeze and crash to a black screen saying my Mac was rebooting because of an error. After a couple of reboots though, they did seem to start working, though they don't charge.

FranticRain, 2016-06-21:
Use the uninstall feature built into the driver to do a clean install. This driver definitely shouldn't be causing kernel panics. If this continues to be an issue, it almost seems like your OS is messed up.

somnolentsurfer, 2016-06-23:
Finally found the time to reinstall the driver and it's picking up the Xbone controller now. Weird. Thanks!

closed 2016-06-23 by somnolentsurfer

FranticRain, 2016-06-23:
Glad to hear it worked out for you! Have fun.