360Controller/360Controller #297: Question about the d-pad :)

daniel-jozsef, 2016-06-21:
Hey... One thing I don't quite understand.

So in Windows, the D-pad directions are reported as "D-pad up", "D-pad down", etc., or alternatively "hat switch up", "hat switch down", etc. Which makes me think there's actually a USB HID descriptor specific for this kinda thing.

Is there a reason this driver reports the D-pad as regular buttons, and not as a D-pad / hat switch?

electric-monk, 2016-06-21:
From what I recall (this was 10+ years ago now, mind) I tried making them actual hat switches, and it caused a lot of games to recognise them as axis. This was fairly useless, as you couldn't use them for anything useful, and the actual physical buttons used on the D-pad are low quality, so they're not even good if you want to use them as axis (glitchy D-pad inputs were bad enough to make some games unplayable).

Thus, so they'd at least work "well enough", and most games that want D-pad style input (like emulators) will accept buttons for input anyway, I made them buttons. There's also plenty of 360 games that use the D-pad as actual buttons, so that fits onto this paradigm anyway.

FranticRain, 2016-06-21:
If you really want to look at how joysticks can be classified using HID descriptors, you'll want to look at [page 26 of this document.](http://www.usb.org/developers/hidpage/Hut1_12v2.pdf) Additionally, an example implementation of a joystick HID descriptor can be found in [Appendix D of this document.](http://www.usb.org/developers/hidpage/HID1_11.pdf)

FranticRain, 2016-07-08:
Closing this due to the question being answered and inactivity.

closed 2016-07-08 by FranticRain