360Controller/360Controller #299: no controller detection

gilbe17568, 2016-06-24:
on mac late 2013,
  Product ID:	0x0006
  Vendor ID:	0x0079
  Version:	1.07
  Speed:	Up to 1.5 Mb/sec
  Manufacturer:	Microntek
  Location ID:	0x14200000 / 10
  Current Available (mA):	500
  Current Required (mA):	500
 
was able to get it working on a game that automatically detected it, trying to play half life 2 on steam

FranticRain, 2016-06-24:
Can you tell me a bit more about this controller? I can't find any reference to a Microntek controller online. Can you confirm that this controller would work with an Xbox 360 console? The way you describe it, this controller doesn't seem like an Xbox 360 controller. This driver only works with controllers that very specifically follow the Xbox 360 controller protocol. We do not support other controllers.

gilbe17568, 2016-06-24:
it is straight usb so it doesn't have a normal normal port for an 360 controller. the package says it's model number 360 and made in china. it is certified USB 2.0. says it works for CPU P-166MHZ or above and RAM 32MB or above. it says it's compatible with WIN 98/2000/XP/VISTA/7. the controller itself only says made in china but i have the product info above. Would there be any kind driver i can download? its a generic one from amazon https://www.amazon.com/gp/product/B007LQG9A6/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1
it also came with a disc but i don't have a drive and it's the size of a gamecube game so i have no way to use it. the disc say "25minute 225megabytes". thats about all the info i can get except that my controller recognizes it as a USB joystick. thanks

FranticRain, 2016-06-24:
I've added your vendor and product IDs to an experimental build of the driver. If it works, then you should be good to go, if it doesn't, then there isn't much I can do to help you. https://drive.google.com/file/d/0B5PLCKoMd4zGUmd0SE5EWEVRYlE/view?usp=sharing

gilbe17568, 2016-06-24:
yea it's still not being detected by the driver, thanks for your help i'm just going to return it and try and get a cheap one that will definitely work. do you have any suggestions on one?

FranticRain, 2016-06-24:
I have no recommendations. I don't use third party controllers, or really controllers much at all. I just use the ones that come with the console.

closed 2016-06-24 by FranticRain

gilbe17568, 2016-06-24:
so i should just by an actual controller and just get a usb adapter?

FranticRain, 2016-06-24:
That would only work for an official Xbox One controller. Wireless Xbox 360 controllers will always be wireless and wired Xbox 360 controllers will always be wired. You can't use any adapter to turn a wireless Xbox 360 controller into a wired one. You should be able to get an official wired Xbox 360 controller for less than 30 USD on Amazon. Just make sure that you are buying from a reputable source, because some of the Amazon distributors ship knock-off controllers instead. (Basically just make sure that you are buying directly from Amazon, not a separate seller on Amazon.)

gilbe17568, 2016-06-24:
i was referring to a corded xbox 360 port to usb but i see what you are saying. if i got a wired 360 controller i would need an adapter to make it usb

FranticRain, 2016-06-24:
No. There is no such thing as an "Xbox 360 port." The Xbox 360 uses USB to connect wired controllers to the system.

gilbe17568, 2016-06-24:
alright will do, thanks!