360Controller/360Controller #300: Rock Candy Xbox one controller isn't detected

Okdakor, 2016-06-25:
I installed the latest release package, there's some activity in the "controller test" panel : when plugged a device is found : "Xbox One Wired Controller (Wired)", but no reaction when the buttons are pushed.

OSX El Capitan 10.11.14
My controller is "Rock Candy Wired Controller for Xbox One"
prod id : 0x0346
vendor id : 0x0e6f
version : 4.02

It's strange because I think I already saw the same issue (same prod id) and it was corrected

FranticRain, 2016-06-25:
Well you would think that with 3 existing listings for that controller we would have covered all of the outrageous PID and VID combinations they have come up with, but no! This is a new one! There is an existing listing for Product ID 0x0347 with the same VID, but this is a new PID! Here's a build for you:
https://drive.google.com/file/d/0B5PLCKoMd4zGY082aXdwOWdVeDg/view?usp=sharing

Okdakor, 2016-06-25:
Sorry, nothing changed with this build, uninstalled the last one and tried it. Same results.

FranticRain, 2016-06-25:
Hmmmm. That is definitely curious. Because that device is now definitely listed in the Info.plist. Additionally, you can also see the VID and PID in OS X, so you are using a data cable instead of a charge cable. It *could* be that this is an issue with the Xbox One packet initialization sequence sent to the controller, but in that case the Preference Pane would think that the controller was connected, but the controller wouldn't be functional. Do you see anything like that or does it appear that the controller is not responding at all?

Okdakor, 2016-06-25:
The cable is the one sold with the controller, didn't try with another one.

> the Preference Pane would think that the controller was connected, but the controller wouldn't be functional.

How can I check the first part, maybe I don't understand ? I can launch the "xbox 360 controllers" from the preference pane, but the controller doesn't react at all. But when I unplug / plug, it changes from "no device" to "Xbox One Wired Controller (Wired)". Tried a mapping, nothing. Tried Enjoyable, it has a "Xbox One Wired Controller" but no reaction here too.
Thanks for the quick answers

FranticRain, 2016-06-25:
Ah ha! So thats exactly what I wanted to know. It seems like an init packet issue! Its a real shame the Xbox logo doesn't turn on on Rock Candy controllers like the official one. That makes them a real pain in the butt to debug, but it shouldn't be too much of a problem. I believe there is a light directly above the Xbox button. I'm not very familiar with what exactly the purpose of that light is. Is that light on whenever it is plugged in as a sort of "charge" indicator, or does it only illuminate after the controller has paired? If you don't know, then just let me know if the light is on at all.

In order to solve a packet issue, you'll unfortunately need a Windows PC to sniff the packets for me. Do you have a PC running Windows around?

Okdakor, 2016-06-25:
I think that there should be a light, but nothing lights up, I don't even see a blink when I plug it.
I have a windows PC around, what do you want me to test ? Let me know and I'll try later.

FranticRain, 2016-06-25:
Here is the process I've documented for sniffing packets using a Windows PC. Let me know if you have any questions. https://franticrain.github.io/sniffing/index.html

Okdakor, 2016-06-25:
Here's the file : https://drive.google.com/file/d/0BzSAKrOZ66uHMlFrc05iRm9kTzQ/view?usp=sharing

FranticRain, 2016-06-25:
Oh yes, I'm sorry. I always forget. You need a slightly different version of these instructions. Please plug your controller in, find the port, disconnect the controller, start sniffing and then plug in the controller. Let it run for a couple seconds and then cancel out. I always forget that those instructions aren't for init packets, I really should change them.

Okdakor, 2016-06-25:
Hope this one is better ;)
https://drive.google.com/file/d/0BzSAKrOZ66uHRkdfYjhmWUVfSjA/view?usp=sharing

FranticRain, 2016-06-25:
Yup! That's exactly what I needed. Give me some time to look over it and see what the problem is, and then I'll get back to you.

FranticRain, 2016-06-25:
Give this a shot and let me know if it works. It looks like there may be some sort of specific call and response happening that I need to work on, but this version of the driver should be hard coded for your controller, at least. https://drive.google.com/file/d/0B5PLCKoMd4zGaGNxcGNwVTA0SGM/view?usp=sharing

Okdakor, 2016-06-25:
Still doing nothing... When I plugged it on the PC I could see where was the light... No light at all confirmed on osx. Thank you for your help, I don't know if this is can be fixed...

FranticRain, 2016-06-25:
I definitely don't want to give up. This sniff has a bunch of 0x01 and 0x06 packets being sent to your controller which I don't remember viewing in older sniffs. I'll give this one some more time to see if I can suss anything out. Because if this is a problem with your 3rd party controller, its probably going to be a problem with ALL 3rd party controllers.

Okdakor, 2016-06-25:
I'll keep an eye on this thread, thx for the work

FranticRain, 2016-06-25:
So my best guess is that 3rd party support has actually been broken since I fixed #257. I even noted in that issue that 3rd party support might have been broken. I've put together a build using the old init packets that *should* work for you. Once again, this isn't a permanent solution but it should get you working. Hopefully. Fingers crossed. https://drive.google.com/file/d/0B5PLCKoMd4zGNS1RS0hMcmhUVjg/view?usp=sharing

Okdakor, 2016-06-26:
Nope, still nothing... :(

FranticRain, 2016-06-26:
Well, I'm just going to have to start giving you builds that are going to give me information about what the driver is doing on your system. This new build should print some stuff out to Console.app. You can find it in `Applications/Utilities`. Specifically you should see `DEBUG - NO CHAT`, `DEBUG - SENDING XONE INIT`, and `DEBUG - SENT XONE INIT`. You may also see something that begins with `start - `. Please tell me what you see in the console after you install this driver. https://drive.google.com/file/d/0B5PLCKoMd4zGREg2TkFpVGdhYUk/view?usp=sharing

Okdakor, 2016-06-27:
OK, here's the console log (tested on 2 usb ports, same result on both) :
27/06/2016 12:18:16,000 kernel[0]: Unknown device release 402
27/06/2016 12:18:16,000 kernel[0]: start - unable to find chatpad interface
27/06/2016 12:18:16,000 kernel[0]: DEBUG - NO CHAT
27/06/2016 12:18:16,000 kernel[0]: DEBUG - SENDING XONE INIT
27/06/2016 12:18:16,000 kernel[0]: DEBUG - SENT XONE INIT
27/06/2016 12:18:16,000 kernel[0]: Unknown escape 128

FranticRain, 2016-07-07:
I've been working on this for a while now, but I think this may be the most promising set of init packets yet. Give this a shot: https://drive.google.com/file/d/0B5PLCKoMd4zGSnNRVUV5V3AwTVk/view?usp=sharing

FranticRain, 2016-07-07:
And here's a second one to try if that first one doesn't work. If it does work, then ignore this one! https://drive.google.com/file/d/0B5PLCKoMd4zGTlAzeWZQUlFWOTA/view?usp=sharing

FranticRain, 2016-08-17:
Here is another driver to test, but this has been inactive for over a month. https://drive.google.com/file/d/0B5PLCKoMd4zGa1pGcF9PaElLcXc/view?usp=sharing

lucmasta, 2016-08-26:
I'm having the exact same issue. I tried the last 3 drivers but none of them worked.

FranticRain, 2016-08-26:
Could you please confirm your product and vendor IDs for me and test this build as well? https://drive.google.com/file/d/0B5PLCKoMd4zGMXB6X0w1UzhuOUk/view?usp=sharing

lucmasta, 2016-08-26:
yeah the PID is 0x0346 and the VID is 0x0e6f

lucmasta, 2016-08-26:
Just rebooted and tried that build. Now the controller isn't even being recognized; "no devices found", tried in a couple USB ports and wiggled the cord around.

lucmasta, 2016-08-26:
oh. disabled the driver and reenabled it and it's recognizing the controller now, but still not working/initializing.

FranticRain, 2016-08-26:
One last different driver for good measure: https://drive.google.com/file/d/0B5PLCKoMd4zGbU5NeWRreXFXeW8/view?usp=sharing

lucmasta, 2016-08-26:
No dice. It's recognizing the controller but it's not working. I don't know if this makes a difference, but it has an extra button on the controller to control the audio.

FranticRain, 2016-08-26:
Does the controller light up in any way when it is properly connected to an Xbox, Windows PC or Linux machine (running xpad)? Because I'm pretty sure this is just an initialization thing, but I have been working on this with a different PDP controller. The other PDP controller works in very specific cases, and I can't reproduce it with your controller. I've exhausted pretty much everything I can do on my end, and I still can't get these things to initialize. But if you look at something like xpad, it doesn't do anything special, it just queues up a write and these third party controllers work. I don't get why these don't.

lucmasta, 2016-08-26:
There are no lights that I can see other than an LED in the audio button, but that won't light up.

Unfortunately I don't even have an xbox one to test it on. I went out today and picked this one up because I'm planning on getting a One in the near future and didn't want to spend $60 on a legit XOne controller since I'll get it when I get the console.

FranticRain, 2016-08-26:
I've exhausted pretty much every idea at this point. Here's one to make sure that it isn't just a weird timing issue. If your computer freezes up for about three second when you plug in the controller, that is intentional. https://drive.google.com/file/d/0B5PLCKoMd4zGZGkzc2huQWJoM2M/view?usp=sharing

lucmasta, 2016-08-26:
No worries man, thanks for trying. That build didn't work either

FranticRain, 2016-08-27:
I'm actually in the middle of talking to Apple about this issue right now, so I want to check some stuff with you since you are having a slightly different experience than what I'm seeing from #333. What version of OS X are you running right now?

lucmasta, 2016-08-27:
ok cool. I'm running 10.11.6 El Capitan on a mid-2009 macbook pro

FranticRain, 2016-08-27:
Okay, thats good. Is there any easy way for you to test with OS X 10.10 at all? If there isn't, don't worry about it. They'd just like to know if its specific to 10.11 because of some backend USB stuff that happened with that release.

lucmasta, 2016-08-27:
No sorry, I can't right now. I do have access to 10.10 on another computer but it's at my office

FranticRain, 2016-08-27:
It doesn't have to be at this very moment. If you get a chance at some point in the near future, I'd really appreciate it.

lucmasta, 2016-08-27:
yeah no problem, I'll give it a shot in the next couple days.

lucmasta, 2016-08-28:
Ok! So I just got to try it on a late 2011 macbook pro running OSX 10.10.5 and the original build works before you modified it for these issues. (the 0.16.3 one on the github release list)

FranticRain, 2016-08-28:
That is OUTSTANDING news. That means that I'm not actually insane, and this isn't my fault! I'll keep working with Apple to get this working on 10.11.

lucmasta, 2016-08-28:
Yeah I was really happy to see those buttons highlight in the settings! Excellent, thanks man

lucmasta, 2016-09-23:
Hey FranticRain, any chance you've been able to work something out with Apple to fix this issue? Thanks!

FranticRain, 2016-09-24:
Unfortunately, they refuse to do anything unless someone wants to send in their controller to Apple.

closed 2018-04-03 by FranticRain