360Controller/360Controller #302: Diagonal movement problem

oli180, 2016-07-01:
I use the 0.16 version of the Xbox controller pkg and everything works fine except the diagonal movement. With the default settings I couldn't run with "100% speed" diagonally and the only solution I found is this: 
[image] but in this case my character is always moving slowly even if I release the joystick. This is a screenshot of the controller preferences when I don't even touch the joystick:
[image] Do you have a solution for this (I also tried 0.15 but nothing happened)?

FranticRain, 2016-07-01:
Joysticks are circles by definition and should never hit their maximum values in the corners. If this is causing a problem in a piece of software, then the developer has done an exceptionally poor job at supporting the software. Controller direction should always be calculated using magnitude of direction. Could you tell me what program in particular is requiring the controller to act like this? We have gotten a request for this previously. Here is an experimental build that should provide the options you desire: https://drive.google.com/file/d/0B5PLCKoMd4zGM2Riam1lYmF4WTA/view?usp=sharing

FranticRain, 2016-07-17:
Closing due to inactivity.

closed 2016-07-17 by FranticRain