360Controller/360Controller #303: Unrecognized Controller

jukefox, 2016-07-01:
I've used previous iterations of this before on earlier editions of MacOS at which point I reverted to only using the controllers in games in a bootcamp Win10. Now I am trying to update and neither of my wired 360 controllers are recognized (only blink twice and shut off when plugged in). 
They are both manufactured by Microsoft and have the following Product / Vendor ID: 

Ox028e / Ox1bad 

and the following serial numbers: "363F1057" and "363F0928". 

What can I do? Thanks!

FranticRain, 2016-07-01:
Okay, here's what you need to do and it should solve your problem.
1. Uninstall the driver. You will need to do this one of two ways. If your current version of the driver includes an uninstall button in the preference pane, use it. If it does not, download and install the latest version on top of your current installation, and then run the installer.
2. Reinstall the latest version of the driver.

This will most likely solve your problem. Let me know how it works out.

jukefox, 2016-07-01:
Sorry, I should have included that I did that already.. Uninstalled 360ControllerInstall_0.11.dmg and downloaded the 360ControllerInstall_0.16.dmg - this did not correct the reported issue.

FranticRain, 2016-07-01:
What exactly was the process you went through to "uninstall" the driver? Unless you used the uninstall button in the preference pane, which I'm pretty certain wasn't in 0.11, then you didn't uninstall the driver.

jukefox, 2016-07-01:
Fair enough. I'll retry and reboot.

jukefox, 2016-07-02:
Uninstalled and reinstalled.. "No devices found" on preference pane.

I really appreciate the prompt responses!

FranticRain, 2016-07-02:
Since you said that your controller worked with previous versions of the driver, I just assumed you had first party, normal controllers. But I can assure you, these are most definitely not standard controllers. I should be able to add these to the Info.plist and get a build that will work for you just fine, but first I need to confirm something.

Product Id: 0x028e (Product ID of the Microsoft Xbox 360 controller)
Vendor Id: 0x1bad (Not the Vendor ID of the Microsoft Xbox 360 controller)

The real controllers have a vendor ID of 0x045e. So, do you know if this controller has some sort of specific third party name so we can list it properly, or should I just list it as a counterfeit?

Here is a build with that controller added: https://drive.google.com/file/d/0B5PLCKoMd4zGeFBlQWZrSmxRSTA/view?usp=sharing

jukefox, 2016-07-02:
This build works great! I think I picked these up secondhand in Chicago so they are likely counterfeits. 

Many thanks for taking the time to resolve this for me and make a new build!

FranticRain, 2016-07-05:
Glad to hear it worked out for you!

closed 2016-07-05 by FranticRain