360Controller/360Controller #306: With latest update controller keeps connecting/will vibrate, then turn off, and then connect again.

rowanfeely, 2016-07-07:
el capitan 10.11.5 and 0.16.1

FranticRain, 2016-07-07:
What controller are you using? Can you recreate this issue with 0.16 or is it exclusive to 0.16.1?

I have tested using a Microsoft Xbox One controller in both 2013 and 2015 variants, in addition to a wireless Xbox 360 controller on 10.11.5 and I cannot recreate this issue.

dtascarella, 2016-07-07:
I am having the same issue. Running OS X 10.11.5 using the latest x360 driver you released.

If you want I can record a video, but here is the issue:
After reboot, you plug in the controller via USB into the mac. The mac recognizes it and the controller vibrates, right after the vibrates the controller loses contact with the mac and the mac reattempts to connect to the controller. Its stuck in this cycle.

FranticRain, 2016-07-07:
Once again, what controller are you using? Can you recreate with 0.16? There have been almost no changes in the driver implementation between 0.16 and 0.16.1 looking at the commit history. My best guess is that you are using Xbox One controllers, likely third party, that don't like the altered init packets, but I actually need to know what controllers you are using instead of just saying "it doesn't work and here's how it doesn't work."

dtascarella, 2016-07-07:
Where can I get 0.16? I'm using the official Xbox One Controller that came with my Xbox One when I bought it last year, nothing third party or anything.

FranticRain, 2016-07-07:
The releases page contains every single release of the driver. https://github.com/360Controller/360Controller/releases

I'd also really like to know how you downloaded the driver without viewing the releases page like that.

trukhinyuri, 2016-07-07:
Bug reproduced only in 0.16.1, 0.16 works correctly. XBOX One original controller.

dtascarella, 2016-07-07:
Yup I got it to work in 0.16 it doesn't disconnect or anything, works perfectly fine. Plug it back into 0.16.1 it starts the whole cycle again.

FranticRain, 2016-07-07:
Okay. I've got a new set of init packets that I've been working on that seem like they might cover all three Xbox One controller firmware areas (old firmware, new firmware, third party). This being the group that has old firmware, #257 being new firmware, and #300 being third party. So I'm going to need all three groups to test these new init packets and fingers crossed, it will work for everyone.

Download and install this version of the driver: https://drive.google.com/file/d/0B5PLCKoMd4zGSnNRVUV5V3AwTVk/view?usp=sharing

rowanfeely, 2016-07-07:
I have a legit microsoft xbox one controller and all versions of the driver worked perfectly up until the update, brought it back to 0.16 and worked fine again until reinstalled the latest release.

FranticRain, 2016-07-07:
@rowanfeely Please try the version listed in the comment directly above yours. It should solve your problem. If I can get confirmation that it works with other problematic controllers then I will push it out to a 0.16.2.

steven807, 2016-07-08:
I had the same problem, and tried the version mentioned in the comment above, which fixed things nicely.  Thanks!

RodrigoCard, 2016-07-09:
I just experienced the same issue in 0.16.1
BUT, I noticed that, if you press the guide button once, as you normally do to connect it wirelessly, it works. It also works if the controller is blinking.

RodrigoCard, 2016-07-09:
Please check my answer in #257

FranticRain, 2016-07-09:
I'm going to go ahead and push out a 0.16.2 that should work for everyone, because getting fresh faces in to complain about issues is a whole lot easier than getting previous issue holders in actually respond to questions.

Something something better to ask for forgiveness than to ask permission.

FranticRain, 2016-07-09:
Marking this as resolved with the release of 0.16.2.

closed 2016-07-09 by FranticRain