360Controller/360Controller #307: Razer Wildcat Xbox One Wired Controller

alexkilty, 2016-07-07:
After installing the software the xbox controller lights up now, but controller input isn't going through.
Product ID: 0x0a03
Vendor ID:  0x1532
Any chance on getting this to work? Thanks!

FranticRain, 2016-07-07:
This is a duplicate of #306. Please try the suggested solution found [here](https://github.com/360Controller/360Controller/issues/306#issuecomment-231089095) and let me know if it works or not.

alexkilty, 2016-07-07:
You my friend are amazing! I can see the button and joystick work in OSX Settings!
When I start rocket league though (steam streaming) it doesn't seem to work.
Do I have to do anything else to set it up with Steam?

FranticRain, 2016-07-07:
I don't remember how to do it outside of Big Picture, so you'll have to go into Big Picture, go to the settings gear in the top right, and then there will be a "Controller" section. You should be able to find everything you need right there.

closed 2016-07-07 by FranticRain

alexkilty, 2016-07-07:
You are correct! Found it and it worked!
Hurray! :)