360Controller/360Controller #309: Special Edition Covert Forces Wireless Controller not Detected

Primalus, 2016-07-09:
I've got it plugged in with a data (and charging) cable, but in the Settings Pane, it doesn't seem to recognize it.

FranticRain, 2016-07-09:
Please follow these instructions from the README: https://github.com/360Controller/360Controller#wired-xbox-one-controllers

Additionally: What version of the driver are you using? Have you used previous versions of the driver?Has this controller ever worked with this driver before?

Primalus, 2016-07-10:
I am currently using the latest version of the driver.  I haven't bothered checking the other versions.  My computer does recognize that the controller is plugged in, but I am unable to use it.

FranticRain, 2016-07-10:
Try 0.16.1 real quick and let me know if it changes anything. That VID/PID combination should work just fine. Its the same as the normal controllers.

Does the preference pane seem to think a controller is connected but you can't get it to see any button presses? Or does it say that no devices are connected?

Primalus, 2016-07-10:
The preference pane said that no devices were connected.  The older version works for my device.  Thanks!

FranticRain, 2016-07-10:
Glad to hear it works for you.

closed 2016-07-10 by FranticRain

FranticRain, 2016-07-10:
If you could do me a huge favor and try out this version of the driver, that would help me out a ton. I'm still trying to nail down exactly what the Xbox One controller initialization packet needs to be since there appear to be two separate ones. If this one doesn't work, just let me know and revert to 0.16.1.

https://drive.google.com/file/d/0B5PLCKoMd4zGMHkwZ2c0bkxNMHc/view?usp=sharing

Primalus, 2016-07-10:
The driver that you posted above seems to work just fine.  I'll have to do some more testing with it, but so far, it recognizes the controller, joystick movements, and button presses.

FranticRain, 2016-07-10:
Okay, thank you very much.

justinxreese, 2016-07-17:
Having this issue on macOS Sierra beta 2. Tried 0.16.1, 0.16.2, 0.16.3 - none of them connect to the controller.

FranticRain, 2016-07-17:
Could you please verify the vendor and product ID of your controller by accessing the Apple menu, selecting "About this Mac", and then selecting "System Report..." on the "Overview" tab. On the left hand side of the new window, select "Hardware". If the controller is plugged in, there should be an entry in this window called "Controller".

Did the controller work with the driver in OS X 10.9?

justinxreese, 2016-07-17:
The controller did work on OSX 10.9 but I did see similar issues at times when first setting up that I would have to plug it in at a certain time in the process.

Right now I don't see anything in the hardware tab (this menu is cool, didn't know about it). The light on the controller actually just blinks for a few seconds like it's trying to find an xbox and then never goes solid before turning off.

This is a plain black controller with a third party batter pack connected to a 2016 macbook by a USB-C adapter.

FranticRain, 2016-07-17:
Either its the adapter or the cable, but one of those isn't a "data" cable. Likely, the cable you are using is a "charge" cable. It is wired differently than a data cable and therefore the OS can't detect it, and therefore the driver can't pair with it. Try some other micro-usb cables and see if that fixes the issue.

justinxreese, 2016-07-17:
@FranticRain Thanks. Will report back. I probably wasted a lot of time last night.

justinxreese, 2016-07-17:
It was the cable 😭 

Thanks for being patient and helpful, @FranticRain

justinxreese, 2016-07-17:
Easy to assume something is wrong when using a beta. Good news is that this works with sierra!

FranticRain, 2016-07-17:
No problem! Glad to hear that it works for you.