360Controller/360Controller #310: Controller flashing on and off...

eyeshark, 2016-07-10:
Product ID:	0x02d1
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	 1.01

The controller is recognized in the System Information. My mac is using 10.9.5. I have tried the latest driver as well as the previous one and the same issue occurs. When I plug it in and look at the Controller Test in the settings window, it shows me pressing buttons and moving the sticks, but every few seconds it acts like it disconnects and reconnects. Have tried several USB cables so I am fairly certain that is not the problem. It will switch back and forth between "no devices found" and "Xbox One Controller (Wired)." Any ideas? 

Thanks!

FranticRain, 2016-07-10:
Is this issue exclusive to OS X or can you recreate it on Windows? This seems an awful lot like a problem with the controller's micro USB port. I can't think of any way I could fix this with the driver, sorry.

eyeshark, 2016-07-10:
I use the same controller and cable on my Windows 10 pic without any problems.

FranticRain, 2016-07-10:
I'd personally like to leave this issue open for now. I'll keep looking into it.

eyeshark, 2016-07-10:
No prob. I'll try another controller when I can. I have an older Mac mini so it may just be flipping out. Thanks for the replies though!

pekiringan, 2016-07-10:
Product : Xbox one controller
Driver : 0.16.1
Mac OS : ver 10.11.5
I have the same problem, my controller has flashing on off on off. keep looking for the solving.
thanks

FranticRain, 2016-07-10:
Is that just with this latest release of the driver or does it happen with previous versions of the driver?

pekiringan, 2016-07-10:
let me try with new release. I just downloaded the driver yesterday, and never use this before. thanks

pekiringan, 2016-07-10:
still flashing on and off in controller with the new driver 0.16.2 or the old 0.16.1 :( thank you

NeoStargate, 2016-07-10:
Exact same problem. It only disappeared when downgrading to version 0.16.

pekiringan, 2016-07-10:
yes, it solved by using ver 0.16 
thanks

FranticRain, 2016-07-10:
In order for me to actually solve this problem in future releases, I'm going to need you to try out some experimental releases so I can figure out exactly what the problem was. Install these versions of the driver and tell me how they work:

A: https://drive.google.com/file/d/0B5PLCKoMd4zGSzNoOXZzVGpyeHc/view?usp=sharing
B: https://drive.google.com/file/d/0B5PLCKoMd4zGNXh0N2R2OTJaOG8/view?usp=sharing

Some notes for my personal reference: A is xpad init packets. B is 0.16 init packets.

larianstudios, 2016-07-11:
Remember I had exactly same issue..

pekiringan, 2016-07-11:
hi, just tried our 2 versions to trial, both working good in xbox one controller. now i am using the 0.16.2 ver (the second driver)
thanks

FranticRain, 2016-07-11:
That's excellent news. I'll put out a 0.16.3 using the xpad init packets that should solve this issue with firmware between Microsoft controllers. I'm still searching for whatever the heck is the problem with third party controllers though. I'd like to get that ironed out before pumping out more releases.

vtsatskin, 2016-07-13:
OS: Max OS 10.11.5 (15F34)
Version: 0.16.2, 0.16.1

I had a similar issues with an Xbox One controller using the aforementioned versions. Trying experimental build B fixed the problem.

FranticRain, 2016-07-13:
Did A not work? Or did you only test B?

vtsatskin, 2016-07-13:
Tested A and it worked as well.

JanMo, 2016-07-14:
installing A solved my problem on OS 10.11.5 (15F34).

FranticRain, 2016-07-14:
Going to go ahead and close this since 0.16.3 should be identical to version A and should work with all first party Xbox One controllers.

closed 2016-07-14 by FranticRain