360Controller/360Controller #312: Xbox One Controller appears in dropdown box, but does not react.

githubuser099, 2016-07-10:
My Xbox One controller appears in the dropdown box, however when any button is pressed nothing reacts...
Any help would be So helpful ! Thank You
Product ID:	0x542a
Vendor ID:	0x24c6
However my third party 360 controller works Perfectly, Thank You!!

FranticRain, 2016-07-10:
What version of the driver are you using?

Try these alternate versions of the driver and let me know if they work for you.
A: https://drive.google.com/file/d/0B5PLCKoMd4zGSzNoOXZzVGpyeHc/view?usp=sharing
B: https://drive.google.com/file/d/0B5PLCKoMd4zGNXh0N2R2OTJaOG8/view?usp=sharing

githubuser099, 2016-07-10:
Downloading and testing right now, will report back in a few minutes...

githubuser099, 2016-07-11:
Aw man, both did not fix my problem ! By the way I didn't happen to mention that this controller works perfectly on Windows

FranticRain, 2016-07-11:
Third party Xbox One controllers are such a pain. Okay, I'm going to need you to do some packet sniffing on Windows. You can find the process to sniff packets on Windows here: https://franticrain.github.io/sniffing/index.html

After that, I'll be able to duplicate the exact init packets being sent from Windows to your controller. However, based on previous experience, it probably won't solve the problem. It may help me out in the long run though. I need as many third party sniffs from different controllers as possible to get this crud working.

githubuser099, 2016-07-11:
Yes Sir, here comes the info in a couple minutes

githubuser099, 2016-07-11:
http://s000.tinyupload.com/index.php?file_id=56414519356195150607
Hope I did it properly !!! Haha

FranticRain, 2016-07-11:
Yup! You got it. Give me some time to figure some stuff out and then I'll get you another build to try out.

githubuser099, 2016-07-11:
Thank You So Much, trust me I'll be waiting haha

FranticRain, 2016-07-11:
Unfortunately, the init packets in that sniff are identical to the packets I sent in option A I had you install earlier. I focused on the first 6 packets sent from the computer to the controller, believing those to be the init packets that started the controller, but that doesn't seem to be the case. For comparison, your sniff shows the computer sending 29 packets to the controller, while my sniffs on first party controllers contain only 4 packets sent to the controller. I'll try making a driver that sends all 29 of those packets, but its going to take a little bit. Lots of numbers and letters to copy and paste.

There's also something tricky in here where the data is being sent to a different pipe than usual. It usually all happens on "1.X.1", even in other third party sniffs. But in your case the ins happen on "1.X.1", and the outs happen on "1.X.3". So we'll see how that impacts the outcome of this.

githubuser099, 2016-07-11:
Thank You so much, I appreciate all Your efforts. Ill be patiently waiting as hard as I can hahahahaha

FranticRain, 2016-07-11:
Okay, give this a shot: https://drive.google.com/file/d/0B5PLCKoMd4zGbUd6R1laa2N6blU/view?usp=sharing

githubuser099, 2016-07-11:
I also got the same result with that one as well :......(

FranticRain, 2016-07-11:
Okay. Lets try some more packets then!

C: https://drive.google.com/file/d/0B5PLCKoMd4zGTjhIczlpNHFGeTQ/view?usp=sharing
D: https://drive.google.com/file/d/0B5PLCKoMd4zGeFVna0tBeHJqNG8/view?usp=sharing

githubuser099, 2016-07-11:
Downloading and testing haha

githubuser099, 2016-07-11:
Neither of those worked either man :...............(

FranticRain, 2016-07-11:
Ugh. How does this work for xpad and not for me?  (╯°□°）╯︵ ┻━┻

I've sent the exact packets the sniff has, I've send the exact packets SteamOS would send to this controller to get it work, but it doesn't work here. I assume its that pipe thing that I noted before, but it'll take me a while to figure that part out. That's all for a while. I'll probably need a couple of days before I have another one to try again.

githubuser099, 2016-07-11:
Yes Sir, I'll keep my eyes peeled for You !

FranticRain, 2016-07-14:
I don't expect this build to fix anything, but it should give me a little bit of information that I need. First, install this version of the driver: https://drive.google.com/file/d/0B5PLCKoMd4zGUTk4RFlVX0R4Wlk/view?usp=sharing

Then, open up Console.app found in Applications/Utilities. Then plug in your controller. This should cause some stuff to show up in Console.app. Look for anything that starts with `start - ` or `3RD PARTY DEBUG - `.

You should see stuff like:
`3RD PARTY DEBUG - IN:1, OUT:1`
and
`start - unable to find chatpad interface`

Let me know everything you see that starts with one of those two phrases.

adamgajzlerowicz, 2016-07-14:
I have the same problem - I have too tried several releases of the driver.
With the latest one in the console.app I am getting stuff like this:

```
kernel: Unknown device release 101
kernel: start - unable to find chatpad interface
kernel: Unknown escape 128
```

I hope this helps

FranticRain, 2016-07-14:
What is the VID and PID of your controller? You can find it in: Apple menu, selecting "About this Mac", and then selecting "System Report..." on the "Overview" tab. On the left hand side of the new window, select "Hardware". If the controller is plugged in, there should be an entry in this window called "Controller".

adamgajzlerowicz, 2016-07-14:
They are
```
Product ID:	0x543a
Vendor ID:	0x24c6
```

FranticRain, 2016-07-14:
I just want to make sure since its so close to the OP, that is 0x543a not 0x542a? Do you know that name this controller is sold under? Like Spectra A Illuminated Xbox One or something like that?

adamgajzlerowicz, 2016-07-14:
It is a CONTROLER PAD XBOX ONE PC-MODEL PREMIUM

```
Xbox ONE liquid metal controller:

  Product ID:	0x543a
  Vendor ID:	0x24c6
  Version:	1.01
  Serial Number:	0000000000000000
  Speed:	Up to 12 Mb/sec
  Manufacturer:	PowerA
  Location ID:	0x14100000 / 10
  Current Available (mA):	1000
  Current Required (mA):	500
  Extra Operating Current (mA):	0
```

adamgajzlerowicz, 2016-07-14:
I will try this?
https://github.com/360Controller/360Controller/issues/213

FranticRain, 2016-07-14:
Give it a shot.

adamgajzlerowicz, 2016-07-14:
That didn't work :(

FranticRain, 2016-07-14:
Well, its not exactly surprising now that I know companies are using the same VID/PID combinations for multiple controllers. I'll keep looking into this, but I can't promise anything.

adamgajzlerowicz, 2016-07-15:
I just realised - You are the only person in the entire world that can make this controller work on my laptop. How interesting!

I'd really appreciate your help on this one.

FranticRain, 2016-07-15:
I wouldn't quite put it like that. There are plenty of people that work on the Linux end of this that seem to know more than I do. Like I didn't figure out a key part of Xbox One controller rumble until someone fixed it on the SteamOS kernel. I've also heard conflicting reports that they have solved 3rd party support.

I'd be super curious to know if either of these controllers work on SteamOS or if they work on other linux distros with the latest version of xpad.

githubuser099, 2016-07-16:
Going to try it right now!

githubuser099, 2016-07-16:
It did not work however I got 
7/15/16 7:40:19.000 PM kernel[0]: Unknown device release 101
7/15/16 7:40:19.000 PM kernel[0]: 3RD PARTY DEBUG - IN:1, OUT:3
7/15/16 7:40:19.000 PM kernel[0]: start - unable to find chatpad interface
7/15/16 7:40:19.000 PM kernel[0]: Unknown escape 128
in the console after doing the steps

FranticRain, 2016-07-17:
Well the good news is that it isn't the pipes, like I thought it might be. The driver automatically took care of that. The bad news is that I have no more ideas.

FranticRain, 2016-08-17:
Try out this version of the driver. Make sure to check your "Console.app" found in "Application/Utilities" when you connect the controller for any sort of debug information. https://drive.google.com/file/d/0B5PLCKoMd4zGa1pqRmJMYU82OE0/view?usp=sharing