360Controller/360Controller #313: Rock Candy Xbox 360 Wired Controller inputs differ from regular Xbox 360 wired controller

sdermott, 2016-07-10:
Trying to use the Rock Candy Xbox 360 controller while playing Left 4 Dead 2 and I'm noticing that the DPad UP, LEFT, RIGHT, DOWN, and Start / Select inputs are not recognized as they are if I use a regular xbox 360 controller.

All works fine on the regular xbox 360 controller.
If I choose to remap Flashlight to Up on DPad within the game
Rock Candy input is recognized as Joy 13
regular Xbox 360 controller is recognized as UP

Also oddly enough for the Rock Candy controller the Start and Select inputs appear to be triggering in game when pressing the Left Thumbstick in (triggers Select) and Pressing the Right Thumbstick in (triggers Start).

On a regular Xbox 360 controller in mapping keys of Left 4 Dead 2:
Pressing start registers as Stick1
Pressing Back registers as Stick2
Pressing LeftStick in registers as Back
Pressing RightStick in registers as Start

So maybe something is reversed here (Strangely the default controller configuration still recognizes start and back buttons of regular controller)?

Any help debugging / patching the issue if solution is not too hacky would be greatly appreciated thanks!

I'm also trying to use the Rock Candy controller for Battle Block Theatre which doesn't recognize any of its inputs. Possibly could be related if for some reason the Rock Candy device inputs are not the same as an Xbox 360 controller ie. system is treating it as a joystick rather than xbox 360 controller.

FranticRain, 2016-07-10:
We can not and will not fix issues that are specific to certain games or certain controllers. The driver treats every series of controller exactly the same. Every 360 controller is the same, no matter first or third party. This is an issue with Left 4 Dead 2 and Battle Block. With regards to Left 4 Dead 2, make sure your controllers are set up properly in "Big Picture" mode. With regards to Battle Block, The Behemoth's games have pretty terrible controller support on Mac, so it's their problem, not ours.

sdermott, 2016-07-10:
Thanks for the speedy reply!
I understand fixing issues that are game specific would be considered out of scope.
It just seemed a little odd to me that while both controllers' inputs are recognized fine by the driver the outputs are not the same when those inputs are routed beyond the System Preference Xbox 360 Controllers app.

Looking into the Big Picture Mode the regular Xbox controller is recognized as an XInput controller while the third party Rock Candy controller is recognized as "cherry". The type of the controller I suppose is the issue in this case as the XInput controller works fine in both the above games.
And the goal of the driver is not to convert one device's inputs into the equivalent output of a standard controller but rather just allow the device to be recognized by Mac OS? 
Nothing you guys could do to change the type of device right? :P

Thanks 😄

FranticRain, 2016-07-10:
We provide what is called a HID Report Descriptor. It tells the computer what each bit in the packet means. For example, bit 3 is the Button 3 state, byte 4 is the left stick x axis position, etc. Every Xbox 360 compliant controller sends the exact same kind of packet to the driver. The only things that are different to a program after it gets funneled through the driver (as far as I know) are the vendor and product IDs of the controller. Those are maintained explicitly for each controller. So every controller is putting out the same data, in the same formats, in the same way.

However, that rule breaks down for Xbox One controllers and the "Pretend" option. That replaces the product and vendor ID of the Xbox One controller (be it third or first party) with that of the official wired Xbox 360 controller. This is purely a hack to help provide backwards compatibility for Xbox One controllers because devs are, in general, exceedingly awful at programming for controllers on OS X.

sdermott, 2016-07-11:
Ahh yes I see! The HID report descriptor certainly works fine for both controllers.

Looking a bit into what Window's recognizes the Rock Candy controller as in Big Picture Mode to see if it would be consistent with what I'm seeing in Mac and it actually is recognized as an XInput Controller on Windows rather than "cherry" (the value seen on Mac) which is the Rock Candy controller's model.

I feel this issue is diverging from the original issue that was reported should I close this as game specific and open a different one regarding the Rock Candy controller not being recognized as an XInput Controller? It looks like it may be picking up this info in the same way as it picks up the VendorID / ProductID if this property retrieved by GetControllerType in DeviceItem.m

FranticRain, 2016-07-11:
Even if you create a new issue, we can't fix this problem. I have been calling this "game specific" but the real correct term is "implementation specific." I'll do my best to explain this.

XInput is a Windows API. It is a framework that Windows applications hook into to access controllers on the system. XInput does not exist on OS X. So no controller is ever "recognized as an XInput Controller" on OS X. This driver takes controllers and translates a specific stream of bytes to HID objects. Games access these in the same way. Lets take buttons for example.
From `Pref360ControlPref.m`
```
- (void)buttonChanged:(int)index newValue:(int)value

...

            case 0:
                [_wholeController setAPressed:b];
                break;

            case 1:
                [_wholeController setBPressed:b];
                break;

...
```

Software queries the OS for devices, and then registers for updates from the system about their status. When a button is pressed, the OS informs the program that a specific button number has been pressed. These button numberings are identical for EVERY controller in this driver. It is up to the game to decide what button number corresponds to what. We don't say "Hey the A button was pressed!" We say "Hey button 0 was pressed!"

So as you can see the driver tells the operating system identical information for every controller in the system. Whats different is the implementation in the software asking the operating system for information. In Valve's case, they have explicitly chosen to alter how their software reacts to first and third party Xbox 360 controllers. That is their choice. We will not provide some sort of patch to prevent that. That is their problem, not ours. We have done our job to the best of our ability, and if they want to treat third party controllers as second class citizens, that is their problem, not ours.

sdermott, 2016-07-11:
Once again thank you for the speedy reply and for the detailed description :)

Closing issue as per your comments.

If I am able to hack together something I'll comment back here in the case others ever come across similar issue regarding a 3rd party controller.

Cheers :)

closed 2016-07-11 by sdermott