360Controller/360Controller #315: Xbox One Elite Controller owners, I need your responses!

FranticRain, 2016-07-14:
I've gotten a couple of reports that Xbox One Elite controllers no longer work with current versions of the driver. However, I've been unable to get anyone to stick around long enough to actually confirm that this is an issue. Please report whether or not your controller works with 0.16.3! If it doesn't, please try out this experimental version of the driver with an altered init packet specifically for Elite controllers. If you do test the experimental version, see if it also starts up your normal Xbox One controllers.

Experimental: https://drive.google.com/file/d/0B5PLCKoMd4zGSkFmdFc2M1d4LTQ/view?usp=sharing

Some notes so I don't forget later:
```
UInt8 xoneInit0[] = { 0x05, 0x20, 0x00, 0x09, 0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x55, 0x53 };
UInt8 xoneInit1[] = { 0x05, 0x20, 0x01, 0x01, 0x00 };
UInt8 xoneInit2[] = { 0x0a, 0x20, 0x02, 0x03, 0x00, 0x01, 0x14 };
```

grenyer, 2016-07-19:
What was the reported issue? My elite seems to work fine with both the 0.16.3 and the experimental driver. I don't have a normal controller to test, sadly.

FranticRain, 2016-07-19:
Off the top of my head, I had a report that this was an issue [here.] (https://github.com/360Controller/360Controller/issues/257#issuecomment-220051791) More importantly, I noticed that the SteamOS kernel uses a different init packet for Elite controllers than xpad does. Since we use the same init packets as xpad does now, I just wanted to check it.

If your elite controller works with 0.16.3, then there is no need to test it with normal drivers. I'd like to get a few more confirmations before I close this down though.

Gab-Nihill, 2016-07-27:
Nope, not working anymore. 

I just installed the driver two days ago, played Oxenfree for the first time and it worked well. Today i'm trying again, but the controller don't seem be recognized anymore. The light is still not blinking (i assume that this happends when you're connected to something that detect the controller) but when i launch the game, nothing work. I reinstalled it multiple times, even tried your spécial Elite package found in this issue, still not working. 

I just don't get how this can be possible, since the past days all i was doing on my mac was watching netflix and coding a mobile app...

FranticRain, 2016-07-27:
Can you confirm your vendor and product IDs for me following the instructions listed here: https://github.com/360Controller/360Controller#wired-xbox-one-controllers

Additionally, if you open up the preference pane, does it seem to think the controller is connected, or does it say that no devices were found?

dezza69, 2016-08-07:
My Elite Controller is working fine with 16.3

celynwalters, 2016-08-14:
I haven't got my elite controller to work with anything else, but the Controller Test in the prefpane is working perfectly in 0.16.3. What should the vendor and product IDs be?

`  Product ID:	0x02e3`
`  Vendor ID:	0x045e  (Microsoft Corporation)`
`  Version:	2.03`

FranticRain, 2016-08-14:
Those are the right values. Now that I've gotten enough reports that this is working, I'm going to close this up.

closed 2016-08-14 by FranticRain