360Controller/360Controller #316: Multiple Controllers Issue

SoBrightie, 2016-07-17:
I tried to connect three xbox one controllers to the Mac using a 4 port usb hub. Two controllers works fine, but when the third one gets plugged in, it kicks out the second controller and replaces it. Is there a fix to this issue?

FranticRain, 2016-07-17:
Connect all three controllers, and then check to see if OS X sees all of them by accessing the Apple menu, selecting "About this Mac", and then selecting "System Report..." on the "Overview" tab. On the left hand side of the new window, select "Hardware". If the controller is plugged in, there should be an entry in this window called "Controller".

In your case, there should be at least three entries called "Controller." I need to make sure that OS X is seeing all of your controllers before I start trying to figure out what is wrong with the driver. So just check that area and let me know if OS X can see all of the controllers.

Additionally, I assume these are all first party xbox one controllers, correct?

SoBrightie, 2016-07-17:
Thanks for the quick reply.

I restarted my Mac and reconnected all the controllers to the hub and everything ran fine. I'm pretty sure it's just a problem on my end. Thanks for the help.

FranticRain, 2016-07-17:
Glad to hear it worked out for you.

closed 2016-07-17 by FranticRain