360Controller/360Controller #317: Progressive trigger

valala, 2016-07-18:
When progressive trigger will be avaible for Xbox One controller?

FranticRain, 2016-07-18:
I am unsure what you mean by "progressive trigger." If you mean the triggers as analog devices, they have done that since day one. They report a value between 0 and 255 corresponding to how far depressed the trigger is. You can see this in the "Xbox 360 Controllers" preference pane.

If you are referring to the trigger rumble, also known as "impulse triggers," then that is a bit more complicated answer. At the current time, impulse triggers only work with UWP games on Windows 10. Therefore, even if we did add support for the impulse triggers in our driver, it is highly unlikely that they would be supported within games. Additionally there are a few technical limitations (OS X only supports 3 axes of force feedback instead of 6) that prevent us from doing that. If you'd like, there is a way you can route all rumble to the triggers in the preference pane. Simply open the preference pane, and in the bottom right hand corner there is a dropdown box that will let you select your force feedback style. You can turn it off, default, triggers only, or both triggers and default rumble options.

Does this answer your question?

FranticRain, 2016-08-01:
Closing due to inactivity. Assumed answered.

closed 2016-08-01 by FranticRain