360Controller/360Controller #319: Original Xbox Controller with soldered cable

authorfunction, 2016-07-20:
I have an original Xbox controller with a modded cable (converted into usb with some simple soldering). The driver won't recognize it though, even though it works with the official drivers under Windows (and under Linux as well). Perhaps this can easily be fixed? I suspect it simply shows up with different information (class, id etc) compared to when connecting through an xbox to usb-adapter. (I've noted that it is identified as a usb-hub by the OS). This is the relevant info from system profiler:

------------
 Hub:
          Product ID: 0x0288
          Vendor ID: 0x045e  (Microsoft Corporation)
          Version: 1.21
          Speed: Up to 12 Mb/sec
          Location ID: 0x14200000 / 8
          Current Available (mA): 1000
          Current Required (mA): 100
          Extra Operating Current (mA): 0

w1der, 2016-07-20:
OK ... thank you for clearing that out swiftly for me!

authorfunction, 2016-07-20:
Well it’s great if I was of any help. For what it’s worth, after my initial post I also found something which seems to indicate that this could be a problem with El Capitan.

Look halfway down this thread: https://github.com/360Controller/360Controller/issues/127

User tmecklem reports this: 

--------------
”I have two xbox (original) controllers with a USB adapter, and both worked fine before an update to El Capitan. After the update, however, one controller stopped being recognized. The console output is: 

kernel[0]: Unknown device release 121
kernel[0]: start - unable to find the interface
System Preferences[671]: No devices found? :( device count==0, i==0

I thought this was strange, so I tried swapping the adapters. It didn't matter which adapter or USB ports I chose, the same controller kept giving that message while the other worked fine.

I finally tried it on a different mac I have with the same results. Pre-upgrade, both controllers worked. After the El Capitan, the same controller refused to work."
--------------

I checked, and I seem to be getting the same message (”Unknown device release 121”) when plugging in my original (modded) Xbox controller.

/Matts

FranticRain, 2016-07-20:
If this is the same issue as you just referenced, please get your endpoint identifiers as @kasbert recommends. If you have a machine running linux, or maybe even Windows 10 now, you should be able to find them using `lsusb`. I don't know how to find it in OS X.

authorfunction, 2016-07-20:
Ok so below is what lsusb outputs regarding the controller:

Bus 003 Device 003: ID 045e:0289 Microsoft Corp. Xbox Controller S
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x045e Microsoft Corp.
  idProduct          0x0289 Xbox Controller S
  bcdDevice            1.21
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        88 Xbox
      bInterfaceSubClass     66 Controller
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               4

Bus 003 Device 002: ID 045e:0288 Microsoft Corp. Xbox Controller S Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         0 Full speed (or root) hub
  bMaxPacketSize0         8
  idVendor           0x045e Microsoft Corp.
  idProduct          0x0288 Xbox Controller S Hub
  bcdDevice            1.21
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           25
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval             255

FranticRain, 2016-07-20:
I don't know anything about original Xbox controllers, so you'll unfortunately have to wait for @kasbert to show up to resolve this one, sorry. But thanks for getting that lsusb output, that should be very helpful!

authorfunction, 2016-07-20:
Allright, thanks a lot!

kasbert, 2016-07-21:
The code matching the controller looks like:

`
        // Find correct interface, Xbox original
        intf.bInterfaceClass=kIOUSBFindInterfaceDontCare;
        intf.bInterfaceSubClass=66;
        intf.bInterfaceProtocol=0;
        intf.bAlternateSetting=kIOUSBFindInterfaceDontCare;
interface=device->FindNextInterface(NULL,&intf);
`

But as the subclass and protocol is as expected, it should have worked.

authorfunction, 2016-07-21:
Ok. It seems that only the hub part (interfaceClass=9, subClass=0)
registers in El Capitan. So I guess there's technically no controller there
to be found by the driver. Does this mean it can't be fixed until Apple
does something? Or is there some way to work around this?

kasbert, 2016-07-23:
Hmm, the hub product id is listed in Info.plist. Is the driver trying to attach to the hub instead of the controller. Just wondering, why the hub is there..

FranticRain, 2016-07-24:
Here's a version of the driver without the hub in the plist, just in case that helps. https://drive.google.com/file/d/0B5PLCKoMd4zGcVVLUEVXWDhkMm8/view?usp=sharing

authorfunction, 2016-07-24:
I can test the driver later this week, don’t have the controller with me right now.

authorfunction, 2016-07-29:
Ok, so I tried the version with the modified plist, but it didn’t make any difference. My guess is that the hub isn’t the problem – if I remember correctly the old Xbox controller actually is designed as a hub as well as a controller (I guess the hub was used for the expansion ports). So it seems like El Capitan still recognizes the hub, but won’t accept the controller – which of course would explain why the driver can't communicate with it anymore. Am I correct?

authorfunction, 2016-07-29:
> 29 juli 2016 kl. 12:23 skrev Matts Lindström <cody.custard@gmail.com>:

Small update, some output from ioreg, like @kasbert suggested:

+-o IOUSBHostDevice@14200000  <class AppleUSBDevice, id 0x1000009eb, registered, matched, active, busy 0 (14 ms), retain 14>
      |       +-o AppleUSBInterface@0  <class AppleUSBInterface, id 0x1000009f2, registered, matched, active, busy 0 (8 ms), retain 5>
      |       +-o Android File Tra  <class IOUSBDeviceUserClientV2, id 0x1000009f5, !registered, !matched, active, busy 0, retain 5>

Don’t know if it helps, but this seems to be everything that gets registered when I connect the controller (the hub as far as I can tell, and as expected)

kasbert, 2016-08-15:
Could it be that the controller is under USB hub, but the hub is not recognized by El Capitan. So the controller is not enumerated either. If that is the case, one should first add a driver (or use existing) for the hub.

RoboSteve, 2016-08-31:
I am having the same issue, System Profiler calls the controller "Hub" and the driver is not finding it.

edrex, 2016-09-19:
https://macman860.wordpress.com/2013/05/03/xbox-driver-for-mac-os-x-lion/

This driver for original xbox controllers works for me under El Capitan with [CSR disabled](https://developer.apple.com/library/content/documentation/Security/Conceptual/System_Integrity_Protection_Guide/ConfiguringSystemIntegrityProtection/ConfiguringSystemIntegrityProtection.html) (not recommended 😱). Maybe someone on this project with a kext signing key could produce a signed build? Or it might be helpful as a reference for making the changes to this driver to support it.

FranticRain, 2016-09-19:
Here's a signed build of that kext. I'm too lazy to put it into an installer. I'm sure you can figure out how to install it yourself. https://drive.google.com/file/d/0B5PLCKoMd4zGYW1HOGtqQnNNczg/view?usp=sharing

edrex, 2016-09-19:
Thank you so much @FranticRain!

It works for me.

After running the installer linked in my previous comment, and unzipping the signed kext, run the following:

```
#!/bin/sh
cd ~/Downloads/
sudo chown -R root:wheel DWXBoxHIDDriver.kext
sudo xattr -d com.apple.quarantine DWXBoxHIDDriver.kext
sudo mv /System/Library/Extensions/DWXboxHIDDriver.kext/ DWXBoxHIDDriver.kext.bak
sudo mv DWXBoxHIDDriver.kext /System/Library/Extensions/
```

edrex, 2016-09-20:
Back to the original issue report, the 360Controller driver seems to work for me with an original XBox Controller S (X08-69873). @authorfunction can you verify that the gamepad isn't working, for example with https://yukkurigames.com/enjoyable/? I'm not sure if there's a command to enumerate input devices on OSX like `xinput` on Linux.