360Controller/360Controller #321: Farming Simulator 15 ... combined button with axis

w1der, 2016-07-20:
I heard from someone who previously tried this with FS15, that it is not able to map a button and axis in combination. 

This is a must to be able to play this rather complex game. 
One axis is used to steer the vehicle/move around ... and the other one to control the camera. 
But one must also be able to use the axis in combination with LB and RB to control functions such as front loaders and cranes (using both sticks is required for this) without having the camera spin around while trying to do it and so on ...   

Would this be possible to accomplish in future updates of this app?

FranticRain, 2016-07-20:
I'm not sure I fully understand your issue, but it sure sounds like an issue that is specific to Farming Simulator 15, that is not existent in other games. We do not solve game specific issues. We can not make changes to the driver specifically for a game. We provide the OS with exactly the controller data it needs. It is up to game developers to use it properly.

w1der, 2016-07-20:
The game FS15 is also available for xbox360 and this feature works on that console so the controller itself has support for this.

I will try to explain it better ... 
The game uses both sticks basically the same way as you would in GTA ... your steer your character/vehicle with one stick and control the camera view with the other. 
However ... the sticks in FS15 needs to be used for other functions in scenarios where the camera view is active but needs to be disengaged to let you use this stick to operate the "tools" attached on your vehicle. 
To accomplish this on the xbox console, one needs to press either the RB or the LB to use the extra functions on the stick ... doing so disables the camera view to be engaged and you can use the sticks to control these other functions. 
You may even have to press LB+RB for vehicles with many functions.

This is a game that would appeal to many mac users as we are used to "double commands" ... ;) 

The game creator Giants is not doing anything wrong ... they are just using the "full potential" of the controller and getting the max out of it.
It is other game developers that have not understood how to use this (yet) ... 
 
This game is typically played with mouse and keyboard to be able to use the crazy amount of keys needed for everything you can do in the game. 
Saitek actually developed a steering wheel and pedals with an additional side panel customized for this game. 
http://saitek.com/uk/prod/farmsim.html

I agre that most games does not have the need for this and I understand if you will not make this your priority. 
However ... I can see how flight sims and other sims could benefit from the possibility to use the controller like this also, as it allows you to use less keys on your keyboard while playing ... 
In theory, using the LB and RB in combination with the sticks gives you the total of 8 sticks (4 setups) on your controller instead of only 2 sticks (1 setup). 
Basically all games where you can map your own buttons ... adding this function gives the gamer a lot of more options on how to setup the controls for the game. 

Please reconsider adding this in to your software in the future ... :)

FranticRain, 2016-07-20:
Lets take the example of moving one of the joysticks and having LB pressed. The driver reports to the operating system: "Here are the x and y coordinates of the stick. LB is pressed." The game queries the operating system. The operating system tells the game "Here are the x and y coordinates of the stick. LB is pressed." That is how this works in Windows AND that is how this works on Xbox 360 consoles. This is most definitely not a feature of the controller. That isn't how any of this works. That is a feature of the game and how it interprets controller input. Furthermore, even if I wanted to do this at a controller level, we can't. OS X lets us have 6 axes total. We use all six axes. X and Y on the left stick. Rx and Ry on the right stick. Z and Rz for the triggers. We cannot create more axes. OS X doesn't have more axes. We cannot create more axes for you to map things to. This is 100% on the game developers to have their controller bindings menu understand the concept of button + axis movement. There is NOTHING we can do.

w1der, 2016-07-21:
Oh I see ... It´s a limitation of the OS X ... BUMMER! 

Thanks for clearing this out ... 
That was the final "excuse" I needed to spend my money on a PC for gaming ... :(

FranticRain, 2016-08-09:
Closing out unfixable game specific issues.

closed 2016-08-09 by FranticRain