360Controller/360Controller #322: Mad Catz MLG Arcade FightStick TE is not working

julianmoreno, 2016-07-20:
I have a  third party controller the Major League Gaming Arcade FightStick: Tournament Edition from madcatz but doesn't work on Mac OSX El Capitan. 

MLG Arcade FightStick TE:

  Product ID:	0xf03e
  Vendor ID:	0x1bad
  Version:	4.90

Thank you in advance.

FranticRain, 2016-07-20:
Here you go: https://drive.google.com/file/d/0B5PLCKoMd4zGZXpMZHNDeFF6aGc/view?usp=sharing

closed 2016-07-20 by FranticRain

julianmoreno, 2016-07-20:
Thanks!!!!! It works perfectly...

julianmoreno, 2019-12-07:
Hi!

Mad Catz MLG Arcade FightStick TE is not working again with 360Controller 1.0.0-alpha.5 in macOS Catalina 10.15.1 

MLG Arcade FightStick TE:

Product ID: 0xf03e
Vendor ID: 0x1bad
Version: 4.90

Thank you in advance.