360Controller/360Controller #323: 3rd Party Controller Addition Request

HAZARDU5, 2016-07-21:
Hey there,

Would you be able to add support for my Afterglow AX.1 for Xbox 360 controller?. Here's the details:

Product ID: 0xf900
Vendor ID: 0x0e6f
Version: 1.02

Cheers!

FranticRain, 2016-07-21:
I honestly wonder what PDP is up to. They use two vendor IDs. 0x0e6f and 0x1bad. We have that same device with the same product ID and other vendor ID listed. These guys just love to waste product IDs. But if they are at 0xf900, they don't have many left!

Either way here's a version of the driver that will solve your problem: https://drive.google.com/file/d/0B5PLCKoMd4zGZUFKTFNTSGd1SHc/view?usp=sharing

closed 2016-07-21 by FranticRain

HAZARDU5, 2016-07-21:
Works great now - thanks!