360Controller/360Controller #324: Kernel panic running OS X Sierra Beta

briantun, 2016-07-22:
Please test on the OS X 10.12 betas -- for me as soon as I connect the Xbox one controller I get a kernel panic. MacBook Pro (Retina, 15-inch, Mid 2014)

FranticRain, 2016-07-22:
People have been using this driver successfully with macOS 10.12. What version of the driver are you using?

FranticRain, 2016-07-22:
Just tested this. There is no kernel panic. Update your driver.

closed 2016-07-22 by FranticRain

briantun, 2016-07-22:
I was running the latest released driver already.

I’m running an internal build of OS X, so perhaps this doesn’t repro with the public seed?

FranticRain, 2016-07-22:
That sure sounds like a beta bug in a beta OS, aka not our problem. If it makes it to a release candidate, I'll look into it. I'd rather not go around chasing a rabbit like we have with previous beta versions of OS X.

briantun, 2016-07-22:
Understandable - sorry for the noise.  I’ll chime in again if I see it repro in a newer build.

christ776, 2016-07-24:
I'm having the same issue reported by @briantun. Attached is the Kernal Panic file. This issue is happening with the Public Beta 1 and now with the Public Beta 2 on 0.16 and 0.16.3 as well.
[Kernel_2016-07-22-122855_Christians-MacBook-Pro.panic.zip](https://github.com/360Controller/360Controller/files/380511/Kernel_2016-07-22-122855_Christians-MacBook-Pro.panic.zip)

FranticRain, 2016-07-24:
I continue to be completely unable to reproduce this kernel panic. Are you running a clean install or an upgrade from 10.11?

christ776, 2016-07-24:
Good question! I'm running an upgrade from 10.11. Sorry, I think I've sent the wrong panic ! Here's the right one having "com.mice.driver.Wireless360Controller"

[Kernel_2016-07-21-145602_Christians-MacBook-Pro.panic.zip](https://github.com/360Controller/360Controller/files/380547/Kernel_2016-07-21-145602_Christians-MacBook-Pro.panic.zip)

christ776, 2016-07-24:
By the way, the panic is triggered after about a minute upon turning off the Wireless gamepad, and leaving the dongle connected.

FranticRain, 2016-07-24:
Try reinstalling the driver, and see if you still panic.

So your panic is with a wireless 360 controller after a having turned the controller off, as opposed to an immediate failure with an Xbox One controller. Sounds an awful lot like the driver got corrupted in an upgrade.

christ776, 2016-07-25:
Removing and reinstalling the driver did the trick! It works perfect now !

briantun, 2016-07-25:
fwiw that didnt help my case...  hoping itl start working in a later OS x build.

electric-monk, 2016-07-27:
I think the solution here is "file a radar" :P

briantun, 2016-07-27:
I updated to a newer build yesterday - if I still see the crash I’ll report back, and will file a radar if needed.  :)

briantun, 2016-07-28:
fyi - Tried this again on a newer build last night.

No crashes - but also, it no longer detected my Xbox one controller. :(

Tried uninstalling and reinstalling, but same result.

FranticRain, 2016-07-28:
Check and make sure the controller is still listed in the System Report and let us know what you find.

christ776, 2016-08-05:
Hey guys , I am now running macOS Sierra Public Beta 3 and just had a kernel panic ,after a few minutes having disconnected the XBOX360 wireless dongle.

Anonymous UUID:       5C49430D-212B-8D84-6799-C73E6F2F0EF3

Fri Aug  5 01:24:52 2016

*** Panic Report ***
panic(cpu 2 caller 0xffffff8015e09f6a): Kernel trap at 0xffffff801629f1e7, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x00000000000005a8, CR3: 0x000000001996d000, CR4: 0x00000000001627e0
RAX: 0x0000000000000000, RBX: 0xffffff804e3f6000, RCX: 0xffffff80465e6da0, RDX: 0xffffff80366affa0
RSP: 0xffffff9202373e10, RBP: 0xffffff9202373e10, RSI: 0x0000000000000000, RDI: 0xffffff804e3f6000
R8:  0x000000000000ffff, R9:  0x0000000000010001, R10: 0xffffff8016466f40, R11: 0xffffff8016339af4
R12: 0xffffff80465e6de8, R13: 0xffffff80465e6da0, R14: 0xffffff80366affa0, R15: 0xffffff80465e6da0
RFL: 0x0000000000010246, RIP: 0xffffff801629f1e7, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x00000000000005a8, Error code: 0x0000000000000000, Fault CPU: 0x2, PL: 0

Backtrace (CPU 2), Frame : Return Address
0xffffff9202373aa0 : 0xffffff8015cf84ac 
0xffffff9202373b20 : 0xffffff8015e09f6a 
0xffffff9202373d00 : 0xffffff8015caa743 
0xffffff9202373d20 : 0xffffff801629f1e7 
0xffffff9202373e10 : 0xffffff7f97152cdc 
0xffffff9202373e30 : 0xffffff801629832e 
0xffffff9202373e80 : 0xffffff801629981a 
0xffffff9202373f10 : 0xffffff8016299686 
0xffffff9202373f60 : 0xffffff801629b09e 
0xffffff9202373fb0 : 0xffffff8015ca7af7 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[C70A1F07-3542-3179-8FCD-ACE7247A5191]@0xffffff7f97150000->0xffffff7f971c7fff
            dependency: com.apple.driver.AppleFDEKeyStore(28.30)[B9C62BFE-2AF9-33FF-85AD-B2BCEEF35E57]@0xffffff7f97145000

BSD process name corresponding to current thread: kernel_task

Mac OS version:
16A270f

Kernel version:
Darwin Kernel Version 16.0.0: Tue Jul 26 18:32:56 PDT 2016; root:xnu-3789.1.4.1.1~1/RELEASE_X86_64
Kernel UUID: B9F1E341-610E-344F-A287-A1A079D87966
Kernel slide:     0x0000000015a00000
Kernel text base: 0xffffff8015c00000
__HIB  text base: 0xffffff8015b00000
System model name: MacBookPro11,3 (Mac-2BD1B31983FE1663)

System uptime in nanoseconds: 8373165721505
last loaded kext at 7360909495402: com.mice.driver.Xbox360Controller	1.0.0d15 (addr 0xffffff7f9917f000, size 77824)
last unloaded kext at 7522864540216: com.mice.driver.Xbox360Controller	1.0.0d15 (addr 0xffffff7f9917f000, size 40960)
loaded kexts:
com.mice.driver.Wireless360Controller	1.0.0d15
com.mice.driver.WirelessGamingReceiver	1.0.0d15
com.apple.driver.AppleHWSensor	1.9.5d0
com.apple.filesystems.autofs	3.0
com.apple.driver.AudioAUUC	1.70
com.apple.filesystems.ntfs	3.13
com.apple.driver.ApplePlatformEnabler	2.7.0d0
com.apple.driver.AGPM	110.23.7
com.apple.driver.X86PlatformShim	1.0.0
com.apple.driver.AppleOSXWatchdog	1
com.apple.driver.AppleGraphicsDevicePolicy	3.13.60
com.apple.driver.AppleHDA	276.16
com.apple.driver.AppleUpstreamUserClient	3.6.3
com.apple.driver.pmtelemetry	1
com.apple.iokit.IOUserEthernet	1.0.1
com.apple.GeForce	10.1.2
com.apple.iokit.IOBluetoothSerialManager	5.0.0d91
com.apple.driver.AppleIntelHD5000Graphics	10.1.8
com.apple.Dont_Steal_Mac_OS_X	7.0.0
com.apple.driver.AppleHV	1
com.apple.driver.AppleThunderboltIP	3.0.8
com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport	5.0.0d91
[... 88 lines cut ...]
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleACPIPlatform	5.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.Libm	1
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0
Model: MacBookPro11,3, BootROM MBP112.0138.B17, 4 processors, Intel Core i7, 2.5 GHz, 16 GB, SMC 2.19f12
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.21.171.5.1b9)
Bluetooth: Version 5.0.0d91, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500.28 GB
USB Device: USB 3.0 Bus
USB Device: Card Reader
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Vendor-Specific Device
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1

FranticRain, 2016-08-05:
Reinstall the driver again. You upgraded from Beta 2 to Beta 3, so it very well could be the same issue.

christ776, 2016-09-23:
Hi everyone, I'm still having this issue with the macOS Sierra build from the AppStore and 0.16.4.

```
Anonymous UUID:       5C49430D-212B-8D84-6799-C73E6F2F0EF3

Fri Sep 23 00:57:10 2016

*** Panic Report ***
panic(cpu 4 caller 0xffffff8010409f4a): Kernel trap at 0xffffff80108a0c37, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0x000000010a421000, CR3: 0x0000000013fb4000, CR4: 0x00000000001627e0
RAX: 0xc0ffee0b0607296a, RBX: 0xffffff8034de3400, RCX: 0xffffff803582ba40, RDX: 0xffffff8030a35e60
RSP: 0xffffff91fcfe3e10, RBP: 0xffffff91fcfe3e10, RSI: 0x0000000000000000, RDI: 0xffffff8034de3400
R8:  0x000000000000ffff, R9:  0x0000000000010001, R10: 0xffffff8010a66fb0, R11: 0xffffff801093bee4
R12: 0xffffff803582ba88, R13: 0xffffff803582ba40, R14: 0xffffff8030a35e60, R15: 0xffffff803582ba40
RFL: 0x0000000000010246, RIP: 0xffffff80108a0c37, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x000000010a421000, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0

Backtrace (CPU 4), Frame : Return Address
0xffffff81eeb55df0 : 0xffffff80102f748c 
0xffffff81eeb55e70 : 0xffffff8010409f4a 
0xffffff81eeb56050 : 0xffffff80102a9743 
0xffffff81eeb56070 : 0xffffff80108a0c37 
0xffffff91fcfe3e10 : 0xffffff7f9179cb4c 
0xffffff91fcfe3e30 : 0xffffff8010899d7e 
0xffffff91fcfe3e80 : 0xffffff801089b26a 
0xffffff91fcfe3f10 : 0xffffff801089b0d6 
0xffffff91fcfe3f60 : 0xffffff801089caee 
0xffffff91fcfe3fb0 : 0xffffff80102a6af7 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[EF97107C-5B48-3B48-A1CF-D9BAC7C2632D]@0xffffff7f9179a000->0xffffff7f91811fff
            dependency: com.apple.driver.AppleFDEKeyStore(28.30)[B9C62BFE-2AF9-33FF-85AD-B2BCEEF35E57]@0xffffff7f9178f000

BSD process name corresponding to current thread: kernel_task

Mac OS version:
16A323

Kernel version:
Darwin Kernel Version 16.0.0: Mon Aug 29 17:56:20 PDT 2016; root:xnu-3789.1.32~3/RELEASE_X86_64
Kernel UUID: 622D2470-C34D-31F9-A62B-6AA9A3C6A3CD
Kernel slide:     0x0000000010000000
Kernel text base: 0xffffff8010200000
__HIB  text base: 0xffffff8010100000
System model name: MacBookPro11,3 (Mac-2BD1B31983FE1663)

System uptime in nanoseconds: 35541954432
last loaded kext at 24730380498: com.mice.driver.Wireless360Controller	1.0.0d15 (addr 0xffffff7f937bd000, size 20480)
loaded kexts:
com.mice.driver.Wireless360Controller	1.0.0d15
com.mice.driver.WirelessGamingReceiver	1.0.0d15
com.apple.driver.AppleHWSensor	1.9.5d0
com.apple.driver.AudioAUUC	1.70
com.apple.filesystems.autofs	3.0
com.apple.driver.AGPM	110.23.7
com.apple.driver.ApplePlatformEnabler	2.7.0d0
com.apple.driver.X86PlatformShim	1.0.0
com.apple.filesystems.ntfs	3.13
com.apple.driver.AppleOSXWatchdog	1
com.apple.driver.AppleGraphicsDevicePolicy	3.13.60
com.apple.driver.AppleUpstreamUserClient	3.6.4
com.apple.driver.pmtelemetry	1
com.apple.iokit.IOUserEthernet	1.0.1
com.apple.driver.AppleHDAHardwareConfigDriver	276.26
com.apple.GeForce	10.1.2
com.apple.iokit.IOBluetoothSerialManager	5.0.0f18
com.apple.driver.AppleHDA	276.26
com.apple.driver.AppleIntelHD5000Graphics	10.1.8
com.apple.Dont_Steal_Mac_OS_X	7.0.0
com.apple.driver.AppleBacklight	170.9.10
com.apple.driver.ACPI_SMC_PlatformPlugin	1.0.0
[... 130 lines cut ...]
Serial ATA Device: APPLE SSD SM0512F, 500.28 GB
Model: MacBookPro11,3, BootROM MBP112.0138.B17, 4 processors, Intel Core i7, 2.5 GHz, 16 GB, SMC 2.19f12
Network Service: Wi-Fi, AirPort, en0
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Model: MacBookPro11,3, BootROM MBP112.0138.B17, 4 processors, Intel Core i7, 2.5 GHz, 16 GB, SMC 2.19f12
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.21.171.10.1a16)
Bluetooth: Version 5.0.0f18, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500.28 GB
USB Device: USB 3.0 Bus
USB Device: USB5534B
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: iPhone
USB Device: USB2134B
USB Device: Keyboard Hub
USB Device: Apple Keyboard
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1
```

meniku, 2016-12-11:
Same Problem here

```
Anonymous UUID:       FB1DA5D3-BC9A-141D-FE9C-B4FB884AA086

Sun Dec 11 23:44:58 2016

*** Panic Report ***
panic(cpu 6 caller 0xffffff80188065ea): Kernel trap at 0xffffff8018c9e077, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0x0000618000382000, CR3: 0x000000001be1b000, CR4: 0x00000000001627e0
RAX: 0x3f00114ce96b595a, RBX: 0xffffff8039d5e600, RCX: 0xffffff804ca29500, RDX: 0xffffff80383fa180
RSP: 0xffffff9228243e10, RBP: 0xffffff9228243e10, RSI: 0x0000000000000000, RDI: 0xffffff8039d5e600
R8:  0x000000000000ffff, R9:  0x0000000000010001, R10: 0xffffff8018e66fb0, R11: 0xffffff8018d3b584
R12: 0xffffff804ca29548, R13: 0xffffff804ca29500, R14: 0xffffff80383fa180, R15: 0xffffff804ca29500
RFL: 0x0000000000010246, RIP: 0xffffff8018c9e077, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000618000382000, Error code: 0x0000000000000000, Fault CPU: 0x6, PL: 0

Backtrace (CPU 6), Frame : Return Address
0xffffff81f631ddf0 : 0xffffff80186f368c 
0xffffff81f631de70 : 0xffffff80188065ea 
0xffffff81f631e050 : 0xffffff80186a5743 
0xffffff81f631e070 : 0xffffff8018c9e077 
0xffffff9228243e10 : 0xffffff7f98f092fc 
0xffffff9228243e30 : 0xffffff8018c971be 
0xffffff9228243e80 : 0xffffff8018c986aa 
0xffffff9228243f10 : 0xffffff8018c98516 
0xffffff9228243f60 : 0xffffff8018c99f2e 
0xffffff9228243fb0 : 0xffffff80186a2af7 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[1A48D920-280E-36FA-8D48-49B79A5656E6]@0xffffff7f98f06000->0xffffff7f98f7efff
            dependency: com.apple.driver.AppleFDEKeyStore(28.30)[EA5D0966-E8EA-337A-98EB-195806E8F723]@0xffffff7f98efb000

BSD process name corresponding to current thread: kernel_task

Mac OS version:
16B2555

Kernel version:
Darwin Kernel Version 16.1.0: Thu Oct 13 21:26:57 PDT 2016; root:xnu-3789.21.3~60/RELEASE_X86_64
Kernel UUID: 8941AC1C-B084-37DE-8A34-4CE638C5CFC9
Kernel slide:     0x0000000018400000
Kernel text base: 0xffffff8018600000
__HIB  text base: 0xffffff8018500000
System model name: MacBookPro11,4 (Mac-06F11FD93F0323C5)

System uptime in nanoseconds: 3548694572144
last loaded kext at 3016356121221: com.logitech.driver.LogiGamingMouseFilter	1 (addr 0xffffff7f9b651000, size 40960)
last unloaded kext at 3101991705969: com.apple.iokit.IOUSBHIDDriver	900.4.1 (addr 0xffffff7f9b642000, size 36864)
loaded kexts:
com.mice.driver.Wireless360Controller	1.0.0d15
org.pqrs.driver.Karabiner	10.22.0
com.driver.LogJoystick	2.0
com.mice.driver.WirelessGamingReceiver	1.0.0d15
com.apple.filesystems.smbfs	3.1
com.apple.driver.AudioAUUC	1.70
com.apple.filesystems.autofs	3.0
com.apple.driver.AGPM	110.23.11
com.apple.driver.X86PlatformShim	1.0.0
com.apple.driver.ApplePlatformEnabler	2.7.0d0
com.apple.driver.pmtelemetry	1
com.apple.iokit.IOUserEthernet	1.0.1
com.apple.iokit.IOBluetoothSerialManager	5.0.1f7
com.apple.Dont_Steal_Mac_OS_X	7.0.0
com.apple.driver.AppleHV	1
com.apple.driver.AppleOSXWatchdog	1
com.apple.driver.AppleGraphicsDevicePolicy	3.13.74
com.apple.driver.AppleUpstreamUserClient	3.6.4
com.apple.driver.AppleHDA	278.23
com.apple.driver.AppleIntelHD5000Graphics	10.2.0
com.apple.driver.AppleThunderboltIP	3.0.8
[... 87 lines cut ...]
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.DiskImages	444.20.3
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleACPIPlatform	5.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.Libm	1
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0
Model: MacBookPro11,4, BootROM MBP114.0172.B09, 4 processors, Intel Core i7, 2,2 GHz, 16 GB, SMC 2.29f24
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x802C, 0x31364B544631473634485A2D314736453120
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x802C, 0x31364B544631473634485A2D314736453120
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x152), Broadcom BCM43xx 1.0 (7.21.171.47.1a8)
Bluetooth: Version 5.0.1f7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0256G, 251 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1

zzz

james-m-henderson, 2017-07-15:
I am seeing a very similar issue happening with my non-beta OSX Sierra.

When I disconnect my HDE xbox360 wireless receiver, the macbook will crash after some period of time.

I have collected four kernel panic logs.  I am running 10.12.5 (16F73).
[kernel_panic_xbox_360_sierra.tar.gz](https://github.com/360Controller/360Controller/files/1150425/kernel_panic_xbox_360_sierra.tar.gz)