360Controller/360Controller #325: Third Party Controllers

Keezyy, 2016-07-22:
I have a PowerA Xbox One Wired controller, when i plug it in to my macbook the XBox Controller app detects the remote immediately. When i move the stick and the rest of the buttons NOTHING happens. Nothing on the controller shows up, but the remote works perfectly fine on my xbox. Please put it where my PowerA remote can work.

Deph0, 2016-07-22:
Please provide Product ID and Vendor ID of your device, read this if you don't know where to find that. ( https://github.com/360Controller/360Controller#wired-xbox-one-controllers )

FranticRain, 2016-07-22:
At this point, third party Xbox One controllers such as this PowerA controller do not work with this driver.

Duplicate of #312

FranticRain, 2016-08-17:
Try out this version of the driver. Make sure to check your "Console.app" found in "Application/Utilities" when you connect the controller for any sort of debug information. https://drive.google.com/file/d/0B5PLCKoMd4zGa1pqRmJMYU82OE0/view?usp=sharing

cvonsee, 2016-09-03:
I tried the driver suggested above with a PowerA controller, and it doesn't seem to work.  The System Preferences app appears to show that the controller is present (it says "Xbox 360 Wired Controller (Xbox 360) (Wired)") but I get no feedback when pressing buttons or moving the joystick on the controller. For grins I also toggled "Pretend to be an Xbox 360 controller" as well, but no joy.

The Mac (OS X El Capitan 10.11.6) sees this as the following:

Xbox ONE liquid metal controller:

  Product ID:	0x543a
  Vendor ID:	0x24c6
  Version:	1.01
  Serial Number:	0000000000000000
  Speed:	Up to 12 Mb/sec
  Manufacturer:	PowerA
  Location ID:	0x14200000 / 8
  Current Available (mA):	1000
  Current Required (mA):	500
  Extra Operating Current (mA):	0

Console.app provides the following feedback:

9/3/16 11:58:23.000 AM kernel[0]: IOHIDDevice::newUserClient called on an inactive device
9/3/16 11:58:23.000 AM kernel[0]: IOHIDDevice::newUserClient called on an inactive device
9/3/16 11:58:23.000 AM kernel[0]: IOHIDDevice::newUserClient called on an inactive device
9/3/16 11:58:23.000 AM kernel[0]: Unknown escape 128
9/3/16 11:58:23.000 AM kernel[0]: IOHIDDevice::newUserClient called on an inactive device
9/3/16 11:58:58.500 AM System Preferences[281]: Unable to simultaneously satisfy constraints:
(
    "<NSLayoutConstraint:0x7fd0ebb07000 H:[MyWhole360Controller:0x7fd0ebb03dc0(465)]>",
    "<NSLayoutConstraint:0x7fd0ebb07740 H:|-(NSSpace(20))-[MyWhole360Controller:0x7fd0ebb03dc0]   (Names: '|':NSView:0x7fd0e951f3e0 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fd0eb895ce0 h=-&- v=-&- H:|-(0)-[NSView:0x7fd0e951f3e0]   (Names: '|':NSTabView:0x7fd0e9520a50 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fd0eb895d30 h=-&- v=-&- H:[NSView:0x7fd0e951f3e0]-(0)-|   (Names: '|':NSTabView:0x7fd0e9520a50 )>",
    "<NSLayoutConstraint:0x7fd0eb8483c0 H:[NSTabView:0x7fd0e9520a50]-(0)-|   (Names: '|':NSView:0x7fd0e9520e30 )>",
    "<NSLayoutConstraint:0x7fd0eb848410 H:|-(0)-[NSTabView:0x7fd0e9520a50]   (Names: '|':NSView:0x7fd0e9520e30 )>",
    "<NSLayoutConstraint:0x7fd0ebb07350 H:[MyWhole360Controller:0x7fd0ebb03dc0]-(NSSpace(8))-[NSBox:0x7fd0e97f0550'Left Stick']>",
    "<NSLayoutConstraint:0x7fd0e97f4f10 H:[NSBox:0x7fd0e97f0550'Left Stick']-(NSSpace(20))-|   (Names: '|':NSView:0x7fd0e951f3e0 )>",
    "<NSLayoutConstraint:0x7fd0e97f3dc0 H:[NSBox:0x7fd0e97f0550'Left Stick'(145)]>",
    "<NSLayoutConstraint:0x7fd0eb83c2c0 H:[NSBox:0x7fd0eb83ba10'Box']-(0)-|   (Names: '|':NSView:0x7fd0e9520e30 )>",
    "<NSLayoutConstraint:0x7fd0eb83c360 H:|-(0)-[NSBox:0x7fd0eb83ba10'Box']   (Names: '|':NSView:0x7fd0e9520e30 )>",
    "<NSLayoutConstraint:0x7fd0eb848070 H:[NSBox:0x7fd0eb83ba10'Box'(668)]>"
)

Will attempt to recover by breaking constraint 
<NSLayoutConstraint:0x7fd0ebb07000 H:[MyWhole360Controller:0x7fd0ebb03dc0(465)]>

Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
9/3/16 11:58:58.805 AM System Preferences[281]: Unable to simultaneously satisfy constraints:
(
    "<NSLayoutConstraint:0x7fd0ebb07000 H:[MyWhole360Controller:0x7fd0ebb03dc0(465)]>",
    "<NSLayoutConstraint:0x7fd0ebb07740 H:|-(NSSpace(20))-[MyWhole360Controller:0x7fd0ebb03dc0]   (Names: '|':NSView:0x7fd0e951f3e0 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fd0eb895ce0 h=-&- v=-&- H:|-(0)-[NSView:0x7fd0e951f3e0]   (Names: '|':NSTabView:0x7fd0e9520a50 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fd0eb895d30 h=-&- v=-&- H:[NSView:0x7fd0e951f3e0]-(0)-|   (Names: '|':NSTabView:0x7fd0e9520a50 )>",
    "<NSLayoutConstraint:0x7fd0eb8483c0 H:[NSTabView:0x7fd0e9520a50]-(0)-|   (Names: '|':NSView:0x7fd0e9520e30 )>",
    "<NSLayoutConstraint:0x7fd0eb848410 H:|-(0)-[NSTabView:0x7fd0e9520a50]   (Names: '|':NSView:0x7fd0e9520e30 )>",
    "<NSLayoutConstraint:0x7fd0ebb07350 H:[MyWhole360Controller:0x7fd0ebb03dc0]-(NSSpace(8))-[NSBox:0x7fd0e97f0550'Left Stick']>",
    "<NSLayoutConstraint:0x7fd0e97f4f10 H:[NSBox:0x7fd0e97f0550'Left Stick']-(NSSpace(20))-|   (Names: '|':NSView:0x7fd0e951f3e0 )>",
    "<NSLayoutConstraint:0x7fd0e97f3dc0 H:[NSBox:0x7fd0e97f0550'Left Stick'(145)]>",
    "<NSLayoutConstraint:0x7fd0eb83c2c0 H:[NSBox:0x7fd0eb83ba10'Box']-(0)-|   (Names: '|':NSView:0x7fd0e9520e30 )>",
    "<NSLayoutConstraint:0x7fd0eb83c360 H:|-(0)-[NSBox:0x7fd0eb83ba10'Box']   (Names: '|':NSView:0x7fd0e9520e30 )>",
    "<NSLayoutConstraint:0x7fd0eb848070 H:[NSBox:0x7fd0eb83ba10'Box'(668)]>"
)

Will attempt to recover by breaking constraint 
<NSLayoutConstraint:0x7fd0ebb07000 H:[MyWhole360Controller:0x7fd0ebb03dc0(465)]>

Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
9/3/16 11:58:58.811 AM System Preferences[281]: ControllerType from prefs was 0, expected 3
9/3/16 11:58:58.885 AM System Preferences[281]: Result of kextstat = 0

FranticRain, 2016-09-03:
This is an Apple bug with macOS 10.11 and later. These controllers only work on macOS 10.10 and earlier. We are currently working with Apple to fix this issue, but it currently isn't resolved.

cvonsee, 2016-09-03:
Thanks for the quick response!  I appreciate all the work you guys are doing on this...

george2967, 2016-09-12:
Any progress here? I also have a PowerA wired controller that doesn't seem to work :(

FranticRain, 2016-09-15:
Unless you intend to mail your controller to Apple, it doesn't look like this one is going to go anywhere. Apple refuses to fix this issue, because that would mean admitting that they messed up the USB code. However, if they had one of these devices, they could figure out where they messed up the USB code and silently fix it in a "security" patch.

Only other fix is to revert to 10.10, before they screwed up the USB code.

closed 2018-04-03 by FranticRain