360Controller/360Controller #326: MS XBox Controller not found

lunaclara, 2016-07-25:
I'm trying to use a controller on my Mac for the first time. I'm not exactly certain how it all works. I have plugged the adapter into a USB port and I can see that the computer knows it's there. The LED light is not lit. I have tried to pair the controller and adapter with no luck. No light ever comes on the adapter, and the driver says no controller is found. The controller appears to be working. Is this a hardware problem or a software problem? My gut instinct was hardware, so I took the adapter back and got a new one. No luck. Currently, I have the 0.16.3 driver, but I did try others first.

FranticRain, 2016-07-25:
This sounds an awful lot like a "Play and Charge Kit." Its a fat adapter with an LED on it that connects to wireless controllers. It doesn't transfer any data. A wireless Xbox 360 controller is always wireless, no exceptions. [This is covered in the README, in addition to instructions on how to troubleshoot issues like this.](https://github.com/360Controller/360Controller#wired-xbox-360-controllers) They aren't called READMEs for nothing.

If you want to use a wireless Xbox 360 controller with this driver, you will need a wireless adapter. The "Play and Charge Kit" ONLY charges the controller. Does that clear this up?

lunaclara, 2016-07-25:
blockquote, div.yahoo_quoted { margin-left: 0 !important; border-left:1px #715FFA solid !important; padding-left:1ex !important; background-color:white !important; } What I have is a wireless adapter. It is not a Play and Charge.

FranticRain, 2016-07-25:
The best way to know if it is a hardware problem is to test on a Windows machine. However, if you don't have a Windows machine, you can give me your vendor ID and product ID. These can be found by accessing the Apple menu, selecting "About this Mac", and then selecting "System Report..." on the "Overview" tab. On the left hand side of the new window, select "Hardware". If the controller is plugged in, there should be an entry in this window called "Vendor-Specific Device".

FranticRain, 2016-08-29:
Closed due to inactivity.

closed 2016-08-29 by FranticRain