360Controller/360Controller #327: Rock Candy wired 360 (0x011f, 0x0e6f) interferes with USB operation, "A" button won't work

drscotthawley, 2016-07-26:
Hi, I saw this issue  about a Rock Candy Xbox One controller [https://github.com/360Controller/360Controller/issues/300](url) but I'm having a different problem, for a different controller.  

I have a wired Xbox 360 controller by Rock Candy.  I don't have Product ID or Vender ID info because "Controller" never appeared on the System Report. (UPDATE: See below.)  Prior to installing 360Controller, the controller wasn't recognized but at least my wired mouse & USB audio interface worked.

I installed the DMG for the main release, rebooted, and then found that *all* wired USB devices would go haywire -- "haywire" meaning they would rapidly come off & on repeatedly.  Audio would be in & out, mouse movement was unresponsive (even the red laser light on the bottom of the mouse would come on & off intermittently) and clicking on things became challenging.  This is *with* the game controller plugged in.  The controller itself won't function either, and the light around the big central start button comes on & off intermittently.    If you unplug the controller then other USB devices go back to normal.

I had this controller working fine in Windows with Boot Camp, and it played nicely with other devices.  

OS X El Capitan, Version 10.11.5
New 13" Macbook Pro ("early 2015" model).

UPDATE: Hub/No-HUB 

1. This was all done with devices running off a a USB hub (but no power-consumption issues were present in Windows, and also if I run only the mouse and controller off the hub with no other devices, the same problems persist).  If I instead plug the controller straight in to the mac, then the other devices work fine (although "Controller" still never appears on the System Report.).  So one could say that it interferes with other USB devices *on the same hub*.  ...But in Windows with Boot Camp, I had it on the same hub with zero trouble.

 
2. With the controller plugged straight in to the computer, the "A" button doesn't work. Or perhaps the "B" button is doing what the "A" button should.  In Steam "Big Picture Mode", pressing "A" is generally "Select" but pressing it has no effect right now.  
 
3. If you click on "USB" instead of waiting for "Controller" to appear in the System Report, then you get a full report which I paste portions of here:
-Rock Candy Gamepad for Xbox 360:
   Product ID:	0x011f
   Vendor ID:	0x0e6f
   Version:	1.00
-USB 3.0 Bus:
  Host Controller Driver:	AppleUSBXHCIWPT
  PCI Device ID:	0x9cb1 
  PCI Revision ID:	0x0003 
  PCI Vendor ID:	0x8086 
-Hub: (which is USB 2.0):
   Product ID:	0x0610
  Vendor ID:	0x05e3  (Genesys Logic, Inc.)
  Version:	92.10
  Speed:	Up to 480 Mb/sec
  Location ID:	0x14100000 / 9
  Current Available (mA):	1000
  Current Required (mA):	100
  Extra Operating Current (mA):	0

Got any tips?  I notice sometimes alternate builds/binaries are recommended.

FranticRain, 2016-07-26:
First, that stuff for #300 has ABSOLUTELY NOTHING to do with this. That is a 3rd party Xbox One specific issue. 

Second, the method of going Hardware->USB and finding the device named "Controller" is the right way to do that. That is a typo in the README that you are the first to point out. I'll get that fixed up later.

Third, make sure to go into big picture mode with no hub. Then go to Settings -> Controller and make sure your controller is configured.

Let me know if that configuration thing solves your problems in no hub configuration, and then we can continue from there.

drscotthawley, 2016-07-26:
Ok, thanks! ! No, hub:  The controller passes all the tests in System Preferences > Xbox 360 Controllers > Controller Test, with flying colors.  So the driver is doing it's job there!
Then I went into Steam Big Picture, configured the controller, and everything's working fine (including the "A" button).  Didn't realize that I needed to do that; on Windows I don't remember doing that.    Thanks for your patience.  

I do intend to upgrade my USB hub over the next several days anyway.  Can keep trying to debug the current issue, or can just wait & report back in a week or so, as to whether a better hub resolves things.

FranticRain, 2016-07-26:
Yeah, you only have to do that configuration on OS X with third party controllers. Its a weird decision by Valve to do that. I can't explain it. I'm pretty strongly inclined to think that this is an issue with the USB hub. I have no idea how to solve that since we don't really get access to that from the driver perspective. There might be a way to fix it but I have no idea how we would figure out how to solve it.

drscotthawley, 2016-07-27:
Confirmation: Better USB hub fixed the issue.   Old hub was an unpowered 4-port USB 3.0 hub by Sabrent, new hub is a powered 7-port 3.0 hub by Sentey.
With the new hub, 360 Controller now presents no interference with other devices, even when plugged in on the same hub.
Consider this problem closed.  Sorry for the false alarm.

closed 2016-07-27 by drscotthawley