360Controller/360Controller #328: multiplayer

thedroom7, 2016-07-27:
I am using a current (as of 2016) macbook pro running El Capitan, and have just installed this driver. It works like a charm with one controller (lights up as player one) on all my steam games that typically offer controller options. However, additional controllers don't work on all games. All controllers i'm testing work on each game by themselves, but players 2-4 controllers (2-4 lit up) only function on some of the games. These are all games that work with local multiplayer on PC (using multiple controllers), but for some reason a few games refuse to pick up on additional controllers (controllers still light up and work on a few games, but not other games).

FranticRain, 2016-07-28:
Unfortunately this sounds like an issue specific to certain games. We cannot fix issues with specific games. We can only do our best to provide an interface that developers can access. For example, at least one game works with multiple controllers, so the driver is clearly capable of it. You'll have to contact the developers of these individual games to get these problems resolved.

FranticRain, 2016-08-09:
Closing out unfixable game specific issues.

closed 2016-08-09 by FranticRain