360Controller/360Controller #332: Hori REAL ARCADE Pro.VX not found (V. 0.16.3)

michelporteneuve, 2016-08-05:
Good morning everyone,

I am using a *Hori REAL ARCADE Pro.VX* and it looks like it is not supported yet. It is an Xbox 360 certified product (This is branded as the Hori Ultimate Marvel vs. Capcom 3 arcade stick on a popular e-commerce website).

Please find below the information about the device:

* Product ID: 0x001b
* Vendor ID: 0x0f0d  (HORI CO., LTD.)
* Version: 5.33
* Serial Number: 090008F0

I am on El Capitan 10.11.6 and using 360Controller V. 0.16.3. I Hope it helps.

At your service,
M

FranticRain, 2016-08-05:
This build should solve your problem: https://drive.google.com/file/d/0B5PLCKoMd4zGaGQ2ZzJzbnhLUXM/view?usp=sharing

closed 2016-08-05 by FranticRain

michelporteneuve, 2016-08-05:
Yes it works perfectly. Thank you so much.