360Controller/360Controller #333: Performance Designed Products Controller

dezza69, 2016-08-07:
Hello,

Thank you for your fantastic controller driver.

It works great with an original Xbox One Controller, but I've hacked a Third-Party one to an "Arcade Stick" (I broke the PCB on an original Microsoft one during construction; 3rd party one is cheaper and easier to hack with soldering pads/single motherboard).

Anyway, it's not working on the Mac...

As suggested, from my "System Report":

 Product ID:	0x015c
 Vendor ID:	0x0e6f
 Version:	4.60
 Manufacturer:	Performance Designed Products

Thanks again, let me know if you need anything else.

Regards,

Dezza

FranticRain, 2016-08-07:
We've had some trouble with third party Xbox One devices, especially from PDP. This should work for you. https://drive.google.com/file/d/0B5PLCKoMd4zGcGtVS2hYUFV5eTA/view?usp=sharing

dezza69, 2016-08-07:
Hello,

Thank you very much for your fast response.

I've download the linked file and have installed it, unfortunately th PDP controller is not recognised.

Thank you for your efforts!

Dezza

FranticRain, 2016-08-07:
This is the same issue as #312 now. PDP does something weird with their controllers that somehow renders this driver incompatible. I have no idea how to fix this.

dezza69, 2016-08-07:
Ok, thank you again for your time and efforts.

Is there anything you'd like me to do which may help you and others, or are third-party controllers just a complete PIA and I should hack-up another genuine or Logitech or something??

:)

Dezza

FranticRain, 2016-08-07:
We already know all of the packets being sent back and forth to these controllers. We have tried sending all of the packets exactly like Windows does to these controllers. For some reason it doesn't work. I've exhausted pretty much every option I can find for getting them working. I'd like to know if PDP Xbox One controllers work with xpad for Linux, but not many people have any reason to run Linux and OS X.

dezza69, 2016-08-07:
Well, I'm a nerd and I'd like to give it a go.

I run multiple Linux "Server" VM's under both Parallels and VirtualBox - which distribution would be the easiest/best to test this?

Sorry for asking, but as this is new (not computers, this topic) to aide troubleshooting what specific things/output will you need (and if you know the applications, that'd be handy too).

Got to do some stuff right now, but I'll aim to set something up today at some stage.

Thank you.

Dezza

FranticRain, 2016-08-07:
Well, lets see. I'm not the most familiar with Linux myself, so I can't quite get you an exact description of what you will need.

First, you would need to clone xpad and edit it to add your controller. So following the README, https://github.com/paroj/xpad#installing
```sudo git clone https://github.com/paroj/xpad.git /usr/src/xpad-0.4```
That will clone the repository into /usr/src/xpad-0.4. You'll need to navigate to that directory and find `xpad.c`. Inside this file, at line 128-ish you'll find `xpad_device[]` which has a bunch of listed devices. The format for all of the entries is: ```{ vendor_id, product_id, name_string, options, controller_type },```
So you will need to add a new device to this list: ```{ 0x0e6f, 0x015c, "PDP Fight Stick", 0, XTYPE_XBOXONE },```. The order probably doesn't matter, but it would go in at line 183 if you want to do that. Then you just have to build and install it with ```sudo dkms install -m xpad -v 0.4```.

You'll need to be careful though. Since kernel version 4.4.0-20, unsigned kernel modules will not run with secure boot enabled. To get around that, you'll want to use this: http://gorka.eguileor.com/vbox-vmware-in-secureboot-linux-2016-update/. It has instructions for VirtualBox explicitly, so you may just want to stick to VirtualBox.

There isn't any real specific testing you need to do beyond seeing if the controller works. If it works, I can see all of xpad's source code and I can figure out what they do differently that we aren't. If it doesn't work for xpad, that means that no one has figured it out yet. Its a pretty binary test. If it doesn't work with xpad, then I don't think anything short of a USB protocol analyzer will be able to fix it.

dezza69, 2016-08-08:
Hi again,

I've just setup a Debian 8 VM (in Parallels 11) and install xpad, making the additions for that joystick as you suggested.

It works fine, with all buttons (which I've connected) giving responses on the command line.

Thanks,

Dezza

FranticRain, 2016-08-08:
That's excellent news. I'll continue to look into this, but it might take some time.

FranticRain, 2016-08-08:
Actually, can you get `lsusb -v` output from bash in through Parallels? The interface matching in 360Controller is really strict compared to xpad, and I was wondering if maybe I was matching to the wrong interface for PDP devices.

I'm specifically interested in the entries for:
```
bDeviceClass
bDeviceSubClass
bDeviceProtocol
```
If it is like a normal controller, they should be 255, 71, and 208 respectively. Because the problem has to be either: timing, pipes, or the interface. Because xpad and 360Controller send the exact same packets to the driver now.

dezza69, 2016-08-08:
Hi,

Here's the output for lsusb -v for that device -> I've just copied/pasted the whole device section as I thought it may have something else which helps...

:)

Dezza

Bus 001 Device 002: ID 0e6f:015c Logic3 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        71 
  bDeviceProtocol       208 
  bMaxPacketSize0        64
  idVendor           0x0e6f Logic3
  idProduct          0x015c 
  bcdDevice            4.60
  iManufacturer           1 Performance Designed Products
  iProduct                2 Wired Controller for Xbox One
  iSerial                 3 0000000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           64
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x00e0  1x 224 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0080  1x 128 bytes
        bInterval               1
Device Status:     0x0000

FranticRain, 2016-08-08:
Okay. Based on some very minor discrepancies in the lsusb output, I've got a version of the driver that should give me some debug information. It very likely will not solve your problem, but it will give me some information. Download it here: https://drive.google.com/file/d/0B5PLCKoMd4zGZU9HM2R2dU15SzQ/view?usp=sharing

You will need to check Console.app, which you can find in `Applications/Utilities`. You should see some debug output that should look like:
```
In Pipe Endpoint Address - 129
Out Pipe Endpoint Address - 1
```
when you plug in the controller. So open Console.app, connect the controller, and then check.

I've also changed the init packets from:
```
0x01, 0x20, 0x01, 0x09, 0x00, 0x04, 0x20, 0x3a, 0x00, 0x00, 0x00, 0x80, 0x00
0x05, 0x20, 0x03, 0x01, 0x00
```
to just
```
0x05, 0x20, 0x00, 0x01, 0x00
```
That is identical to the init packets of that build of xpad. The other set of init packets is from the SteamOS version of xpad.

dezza69, 2016-08-08:
Thanks for all this...

See attached screen shot for details from console...

Dezza

[image]

FranticRain, 2016-08-09:
Well. Thats exactly what I get, so that's probably a dead end. Boy I wish I knew what the deal was.

FranticRain, 2016-08-10:
Okay. I've got another total shot in the dark. Try this one out: https://drive.google.com/file/d/0B5PLCKoMd4zGNHNMQU1TeTVmczQ/view?usp=sharing

dezza69, 2016-08-10:
Thanks, I really appreciate your efforts.

Unfortunately there isn't much light in that there tunnel...

I've attached a screen shot for you.

Thank you again.

[image]

FranticRain, 2016-08-11:
Okay, I'm now 100% certain that we are talking on the right interface with the right pipes. So now I want to see what the controller is saying to me. This one is going to spam your Console.app. It is going to print all of the packets I get from the controller after I send my init. Even if the controller doesn't light up, if you could press a couple of the buttons to see if the device says anything, that'd be great. If it says something, anything, maybe I can figure out what is wrong.

https://drive.google.com/file/d/0B5PLCKoMd4zGTHJsYWlhY2RrNUE/view?usp=sharing

dezza69, 2016-08-11:
Wow!

You are seriously awesome! Thank you so much for all this coding you're doing to enable me (us) to use these devices! :)

Attached are two images, one of the console and one of the preference panel.

The last four lines in the console log are me pressing and releasing the "A" then "B" buttons.  You're probably aware that a button press generated 2 log lines.

The preference panel is there because I was stoked that the system now recognises the attached control and, all the buttons worked!!!

Yay!!!

[image]

[image]

FranticRain, 2016-08-11:
This is 110% not okay! Nothing I just did should have solved this! I'm going to need you to test a couple builds for me so I can figure out what it the world just happened here.

There are currently three blocks of code that differ from the first driver I sent you. These are: A. The init packet. B. The "outCounter" and C. The debug logs. What puzzles me is that the second driver I sent you (found here: https://github.com/360Controller/360Controller/issues/333#issuecomment-238273973) contains the changes to A and B and didn't work. That implies that the debug logs fixed this issue, which you may understand, makes me REALLY FRUSTRATED. So we are going to try all 8 combinations of these things and I'm going to need you to tell me which ones work and which ones don't!

1. (000) https://drive.google.com/file/d/0B5PLCKoMd4zGa284ZmFFTTQyb3c/view?usp=sharing (This should be the should be the same as the first one I gave you.)
2. (001) https://drive.google.com/file/d/0B5PLCKoMd4zGMi01NGJ0Mkl0elE/view?usp=sharing
3. (010) https://drive.google.com/file/d/0B5PLCKoMd4zGc1Bwd3ZNVm9kc1k/view?usp=sharing
4. (011) https://drive.google.com/file/d/0B5PLCKoMd4zGRFI2SzVnNlJjYU0/view?usp=sharing
5. (100) https://drive.google.com/file/d/0B5PLCKoMd4zGTlpwZWdWamd5OW8/view?usp=sharing
6. (101) https://drive.google.com/file/d/0B5PLCKoMd4zGSlRJTi1XTUVPWjA/view?usp=sharing
7. (110) https://drive.google.com/file/d/0B5PLCKoMd4zGRm5rQW5tMVlfQzQ/view?usp=sharing (This should be the same as the second one I gave you.)
8. (111) https://drive.google.com/file/d/0B5PLCKoMd4zGUjA3bVpTVzNGd0U/view?usp=sharing

(Notation isn't binary. It is referring to what parts are original and what parts are changed. From left to right: Init packet, outCounter, debug logs. 1 = Changed, 0 = Original (same as currently committed to master)

Sorry you have to test all of these, but I REALLY want to know.

dezza69, 2016-08-11:
> There are currently three blocks of code that differ from the first driver I sent you. These are: A. The init packet. B. The "outCounter" and C. The debug logs. What puzzles me is that the second driver I sent you (found here: #333 (comment)) contains the changes to A and B and didn't work. That implies that the debug logs fixed this issue, which you may understand, makes me REALLY FRUSTRATED. So we are going to try all 8 combinations of these things and I'm going to need you to tell me which ones work and which ones don't!

Sure - be happy to do so!!

I'll start testing them all after work today and compile a log list for you.

My testing methods:
- Always use the same USB port for devices
- Remove previous driver with an extra clean start between new install
- Install new driver
- Test on fresh start, no additional apps running

FYI, and it may help you(??) - I couldn't uninstall the driver from the one before last, it failed, so all consequent installs have been over the top.

I'll always try to uninstall previous versions first, as per above.

:)

Thank you very much for all this.

Dezza

FranticRain, 2016-08-11:
You don't have to worry about the uninstall too much. The driver should replace every existing file with a new one just fine. I don't know why the old ones didn't uninstall. That is super weird.

dezza69, 2016-08-12:
All done and tested.

6. (101) https://drive.google.com/file/d/0B5PLCKoMd4zGSlRJTi1XTUVPWjA/view?usp=sharing

From what I can see, this is the only one which works...

Again, thank you for all this!

:)

Dezza

[XBox Controller Testing.zip](https://github.com/360Controller/360Controller/files/414954/XBox.Controller.Testing.zip)

dezza69, 2016-08-12:
That was supposed to say 6. (101) - but it came through as a 1. ...

FranticRain, 2016-08-12:
Thanks for you notes and detailed descriptions of each build. Its led to me to believe that all of the previous situations where the controller has worked have been flukes of the build process. And in programming when you get some flukes like this, its probably a timing problem! So I've got two new builds with a delay introduced. The first one has even more debug output every time a button is pressed, because I just want to check some stuff. Its probably a total waste of time. The second one is the same as the first, but without the debug prints. This is just to make sure that the debug printouts aren't magically making it work.

1. https://drive.google.com/file/d/0B5PLCKoMd4zGTjFGZ0M2T255V2M/view?usp=sharing
2. https://drive.google.com/file/d/0B5PLCKoMd4zGU2RpWlVVcUFRRE0/view?usp=sharing

Something that really bothered me is how 2, 6, and 8 all put out button press output. Those button presses aren't sent out by the controller unless it is initialized. That means that for 2 and 8, the controller failed to show up as a valid device, but was initialized properly. I just want to make sure that you are plugging in your controller first before opening the Preference Pane. That thing can be really weird about what it recognizes after it opens.

dezza69, 2016-08-12:
Hey!

I'll get to testing these 2 drivers shortly - but I've always had the preference pane open before plugging the controller in.  I notched it didn't always recognise it which is why I did the unplug/re-plug thing!

Would you like me to re-test (any/all of) those previous 8 as well as these new 2 using your sequence??

😊

Dezza

FranticRain, 2016-08-12:
Don't re-test anything. Just the new ones. I got enough out of the Console logs to tell me what I needed to know.

dezza69, 2016-08-12:
Thank you very much for all your efforts.

:)

Please see attached file...

[XBox Controller Testing.zip](https://github.com/360Controller/360Controller/files/416275/XBox.Controller.Testing.zip)

FranticRain, 2016-08-12:
Oh man I'm so close I can smell it. I just have to figure out what the deal is with those unplug/replug issues. I'm going to bump the delay up in hopes that it helps. Not sure why it works the first time and then doesn't afterwards with the same delay. This one bumps the delay from 50ms to 100ms.

https://drive.google.com/file/d/0B5PLCKoMd4zGWUhnTzN0LWF4RmM/view?usp=sharing

We may just have to try to walk this delay in until we find the right value, so I hope you'll be patient with me. Also, none of these will contain button press debug packets. You'll notice a little message about the controller sleeping, though. I'm effectively delaying operation of the operating system by 1/10th of a second now, so I'd like to at least give users the courtesy of a log message.

dezza69, 2016-08-13:
Hello,

After unplugging/replugging there's not response from the System Preferences Joystick Test.

Here's more from the log.

Hope that helps - :) :)

[image]

[image]

FranticRain, 2016-08-13:
Hmm. That still didn't fix it, huh? I just want to make absolutely certain that this controller works perfectly on other operating systems, and that this is definitely an issue with the driver. The driver shouldn't treat a disconnect/reconnect any different than an initial connect.

dezza69, 2016-08-13:
Hey,

I've just tested this controller on a Windows 10 system and on the Linux VM (as per above).

It works with sequential unplugging/replugging.

I really have no idea, but as it crossed my mind - could the driver not be closing the pipe correctly upon removal?

Thank you for all your brain power, really appreciated.

FranticRain, 2016-08-13:
I've added some debug for the closing of the pipes, just in case. Unfortunately, that's just a function I call for OS X to handle. Not much I can do. It spits out a error that corresponds to "kIOReturnNotReady" but thats probably because closing a pipe is asynchronous, so you are just telling the operating system to do it when it is ready.

I've got two new builds to try out. First is one that uses delay for the first connect, but then removes the delay for subsequent connects. This is the way it seems to work on windows. First connect takes forever, then the later connects much faster. The second build has a shorter delay for the first connect and a longer delay for later connects. This operates under the assumption that loading the kext into memory is a time intensive process, so when the kext isn't loaded back into memory for subsequent connects, we need to delay more.

1. https://drive.google.com/file/d/0B5PLCKoMd4zGWXdBVTVfQkxCZTQ/view?usp=sharing
2. https://drive.google.com/file/d/0B5PLCKoMd4zGZXNzS3QzUlNoN3c/view?usp=sharing

I also want you to test another thing. You should be able to reset the driver to the same as a "first connect" without rebooting. Open up your terminal and type in `sudo kextunload -b com.mice.driver.Xbox360Controller`. Then if you type in `kextstat`, you won't see "com.mice.driver.Xbox360Controller" listed. It will always be the last item in the list, so you don't have to search for it.

dezza69, 2016-08-13:
Hello

Done a bit of testing with these two drivers, see the attached .zip for details when used through my OS X Mobile Account.  I also tried this as the main "Admin" user for this computer, details are not recorded as results appeared similar to the Network Admin User used in the .zip

"1" was tested twice (1 and 1 retest) as I thought I'd stuffed it up.

I've uninstalled and restarted computer between driver installs.

This particular computer has gone through successive OS X updates.  Next I'll start from scratch with a brand new, fully fresh El Capitan install and no installed apps and try again, see if that makes a difference.  Just need to do some external hard drive cage swapping to set this up.  I probably won't get to do this today.

Hope this helps.

[XBox Controller Testing.zip](https://github.com/360Controller/360Controller/files/416807/XBox.Controller.Testing.zip)

FranticRain, 2016-08-14:
This sleep stuff seems way too inconsistent. I'm going to try something a little bit different. Let me know if it works at all.

https://drive.google.com/file/d/0B5PLCKoMd4zGVGs2V3NRT1Nmamc/view?usp=sharing

dezza69, 2016-08-14:
Thank you for all this, very much appreciated.

It works - but not after unplugging, etc.

[XBox Controller Testing.zip](https://github.com/360Controller/360Controller/files/416885/XBox.Controller.Testing.zip)

FranticRain, 2016-08-14:
Using that same driver, connect once so it works, disconnect, then type:
```sudo kextunload -b com.mice.driver.Xbox360Controller```
```sudo kextutil /Library/Extensions/360Controller.kext```
That will forcibly unload the driver, and then forcibly reload the driver. That way we will know with absolute certainty that its something weird with the driver already being loaded.

dezza69, 2016-08-16:
Hi

Sorry for the delay, all this has had me chasing errors (unrelated to this, but highlighting some issues with El Capitan and Open Directory) found in console.

Thank for those commands, although after following your instructions there is no joy(stick)!

The joystick is recognised in System Preferences -> Xbox 360 Controllers, but it doesn't respond to any button input.

I've not tried a fresh re-load (haven't had time) - but I will soon.

Kind Regards,

Dezza

FranticRain, 2016-08-16:
Okay, here is one that is going to try repeatedly to send the init packets if something goes wrong with the pipes. Make sure to check your console for any logs. https://drive.google.com/file/d/0B5PLCKoMd4zGa1pqRmJMYU82OE0/view?usp=sharing

dezza69, 2016-08-17:
Thank you for all your brain power, appreciated.

After installing and plugging in the controller works, upon re-inserting it's a no go.

Screen shot from the log from that process and, loading up the System Preferences.

:)

[image]

FranticRain, 2016-08-17:
So my thought process is currently this: The controller has to treat every plug in the same because it is basically getting power cycled every time. It shouldn't matter to the controller if it was unplugged for 5 seconds or for 5 days because either way, it hasn't had power. So we have to ask ourselves what is changing. The only thing that changes is whether or not the kext is being loaded into memory. On reboot, com.mice.driver.Xbox360Controller is not loaded. However, after a first connect of the driver the driver is loaded into memory and then stays there. However, OS X will naturally unload the kext if it isn't required for long enough. This takes somewhere between 3 and 5 minutes on my system. I had hoped that by using `kextunload` I would be able to speed this process up and the kext would be reloaded when you plugged in the controller, but that didn't work. `kextunload` appears to flag the driver to prevent it from being reloaded. So I told you to reload it with `kextutil` after you unloaded it, but that loads the kext into a state that would be similar to a reconnect. This last build was an attempt to see if the driver could see that fact that it was failing in this manner and recover from it, but the driver can't tell. We can know for sure that this is happening by having an initial connect that works then waiting for the driver to unload and connecting again. If my hypothesis is correct, this reconnect will actually work.

Operating under the assumption that my hypothesis is correct, what is different about having the driver load for a match versus matching when already loaded? We actually don't get to see any of this since it is happening in the kernel, so its definitely not some piece of our code that is messing it up. The only thing I can think of is, once again, timing. Because the kext is loaded, it is sending the data too early. None of that makes sense though. The controller and operating system are sending each other requests querying data about USB before we get a chance to send any data. If the controller is powered up enough to respond to those USB requests, why can't we talk to it properly? Its possible that the chip that handles USB communications is separate from the main microcontroller on this controller. The USB chip handles all of the basic communications, but as soon as I send the init packet, the USB chip accepts the packet as valid and then sends it to the microcontroller whether it is ready for it or not. Therefore, I don't see any mention of a failure on my end.

But that doesn't make sense either. If the timing is based on JUST loading the kext, why didn't we get a first connect to work until I introduced a sleep into the driver? Because we have two cases where I can get an initial connect. 1. An `IOSleep` before sending the init packet. 2. Adding a timeout to the USB message. The current timeout is 3 seconds. This is an incredibly long time on these scales, but it doesn't seem crazy considering the fact that it takes 7 seconds to connect on first pairing with Windows with my official controller. Either way, it doesn't make sense because the driver NEVER reports a failure of writes. Not when the controller fails to start, not when it works, not on a reconnect. The kext has no idea these failures are happening. To OS X, these packets were sent without a single issue. It makes me wonder why a timeout helped at all. The packet should be sent as soon as it can be, otherwise we would see an error. So when there was a timeout of 0 seconds, the packet was sent successfully immediately. So why does adding a three second timeout work? It should send the packet immediately just like the zero timeout did, because everything should be ready. If it wasn't ready, it would have thrown an error, right? It drives me crazy. None of this adds up. None of this makes sense.

So here is what I need you to do. First, connect the controller normally, make sure it works, then unplug the controller. Let your computer stay awake for a while and check `kextstat` to make sure that the kext has been unloaded. Then connect the controller again and let me know if it works. That will let me know if my hypothesis is correct. Second, here is a driver with a 10 second timeout because I don't know, maybe that'll just work. Wouldn't be the first time I've fixed something in the kernel with such an arbitrary change. https://drive.google.com/file/d/0B5PLCKoMd4zGa1pGcF9PaElLcXc/view?usp=sharing

dezza69, 2016-08-17:
Thanks very much for all your efforts.

I'll do this after work. This weekend I'll also be doing a fresh load test, I have a feeling this particular OS installation is flawed after multiple version updates...

😊

dezza69, 2016-08-20:
Hello

After reloading my MacBook from scratch with the latest El Capitan 10.11.6 and extensively testing all the drivers you've provided above on a fully clean system - I can confirm:

a) All the test results above repeat (only after 'c').
b) After unplugging, regardless of command line commands or time, the controller fails to reconnect after replugging in to same or different USB port.
c) Before installing a driver, good habits remove any previous version and restart computer (the drivers didn't always work if this process wasn't adhered).
d) I'm currently using the last driver you've provided.  Thanks heaps.

Happy to do more testing when/if you need.

Hope any of this helps.

Kind Regards,

Dezza

FranticRain, 2016-08-22:
I've been working on some complicated backend stuff after emailing some guys at Apple. I've got a preliminary version up and running. It works with pretty much only wired Xbox One controllers at this point, so don't go expecting it to work properly with other controllers. This is just a test to see if it actually fixes the issue like the guys at Apple think, or not. Give it a shot: https://drive.google.com/file/d/0B5PLCKoMd4zGMXB6X0w1UzhuOUk/view?usp=sharing

dezza69, 2016-08-22:
Hey, thank you.

That one recognises the Controller, but none of the buttons respond even on a fresh start.  I uninstalled the driver, restarted x 2, installed the driver, restarted x 2 - recognised but not functioning.

I swapped between a genuine Microsoft Xbox One Controller (with 3.5" headphones) multiple times, only the genuine one is recognised and functions correctly.

Console shows below when the Performance Designs Controller is plugged in.

Thank you again.

:)

Dezza

[image]

FranticRain, 2016-08-22:
I have to submit an official Technical Support Incident with Apple over this issue. I only get two a year without having to pay extra so I want to make sure I have everything as well prepared as possible so I don't waste it. I'm going to need the specifics on this PCB you are using from PDP. Simply put, I need to know what the original product is here, just in case they want to try it against the real hardware to find the problem. Is it just a standard controller PCB that you have soldered on to, or is it some fightstick PCB you can just buy somewhere? Having that product name would be extremely useful. If you can think of anything else that might help in this process, let me know, but I think all I need from you is the product's name.

FranticRain, 2016-08-22:
First I want to test something silly that probably won't do anything, but I want to test it anyway. Once you've tested it out, do the next part. https://drive.google.com/file/d/0B5PLCKoMd4zGdllWM2k5dlpBQTg/view?usp=sharing

I've also simplified the driver's code by removing things like chatpad support so that the problem is more in focus, and not hidden by other non-important code. Just to make sure, I need you to test these builds so I know for sure that I didn't mess anything up. Because I've slimmed down the driver, that means we can't use the installer anymore. In the enclosed zip file, there will be two directories. One is labeled "New" the other is labeled "Old." Old is the one that should work after a fresh boot, but never work again. New should never work. It should be like the last driver I had you test. In order to install these, first uninstall the current version of the driver. Then, navigate to the kext you want to test in your terminal and run the bash script using `./install.sh`. This will place the kext in `/Library/Extensions` and prepare it for use. Make sure that the file is there and reboot your machine. Then, test the driver like usual, making sure to include any console output. I don't think there will be much since most of the usual culprits are chatpad related. You might get an "Unknown escape 128".

dezza69, 2016-08-22:
Hello,

You are awesome, thank you.

This is a cheap Xbox One Controller from EB Games which I've hacked apart.  There should be an image attached of the PCB which I used to help soldering the pads (old eyes!)... That part number is also screened onto the back.

If you need anymore info please let me know.

Also, to save your freeby email question which I'm sure you'll use at some other point, I have a paid up Apple Developer account - I'll never use an email question as I'm not very good at programming and am about 45 years behind the brilliance around me, if you're able (and I get them) you can use one of my free questions.

BTW - this computer is a pre-ordered 1st generation MacBook Pro Retina 15" .  Dunno if that will help Apple, but it may.

Again, thank you.

FranticRain, 2016-08-22:
Don't worry about my TSIs. I've been working on this driver for 3 years now, and this is the first time I've actually had to use one of these. Since they are limited, I just like to treat them with the respect they deserve. I'll take care of it.

RodrigoCard, 2016-08-22:
@FranticRain You can use one of my TSI if you need.
The only time I used one was to fix a problem with this driver

FranticRain, 2016-08-22:
@RodrigoCard Thanks for the offer, but using one of mine is perfectly fine. Don't worry about it.

FranticRain, 2016-08-23:
@dezza69 Don't forget to test the stuff from https://github.com/360Controller/360Controller/issues/333#issuecomment-241531854. I just realized that I forgot to attach the TSI kexts, so those are finally there, sorry about that.

dezza69, 2016-08-23:
Hello,

Sorry - I missed that one.

Process - 
1. REMOVED EXISTING DRIVER, restarted, installed from .dmg, restarted, tested -> "Console - driver from dmg", Preferences OK (except on replug)
2. REMOVED DRIVER, restarted
3. Installed "old" kext, restarted, inserted joystick -> "Console - old installed from script - dmg REMOVED", NO preferences to test.
4. REMOVED KEXT, restarted
5. Installed "new" kext, restarted, inserted joystick -> "Console - new installed from script - dmg REMOVED", NO preferences to test.
6. REMOVED KEXT, restarted
7. Installed driver from .dmg, restarted
8. Installed "old" kext (over that installed from .dmg) "Console - old installed from script - dmg INSTALLED", Preferences OK (except on replug)
9. RESTARTED
10. REMOVED "old" KEXT
11. Installed "new" kext "Console - new installed from script - dmg INSTALLED after loading System Preferences", Preferences NO response.  (see image for console response upon loading preference pane)

Hopefully, I did this correctly.  If not, please let me know and I'll do it again.

Thank you.

Dezza
[Testing.zip](https://github.com/360Controller/360Controller/files/433715/Testing.zip)

FranticRain, 2016-08-23:
Thanks for testing that. Its exactly what I needed and expected. I'll get that TSI in as fast as possible.

That side note is interesting. I wish I had someone else with third party controllers that would actually respond to me so we could double check all of this stuff just in case. Oh well. We'll see what Apple has to say about it.

dezza69, 2016-08-23:
I have 3 genuine Microsoft Xbox One controllers: STD, 3.5mm Headphone socket and Elite Controller and this PDC which you're very generously working on get functioning.

More than happy to test anything for you with any of those, or other things if I'm able (just ask).

Thanks very much, really appreciate your time.

:)

Dezza

FranticRain, 2016-08-27:
Okay, I got a response from the TSI and I need some more stuff from you. Is there any way for you to get access to a machine running macOS 10.10.x or earlier? They'd like to know if this is exclusive to 10.11 or not. Additionally, he wants a USB packet trace. I don't know how to get that, so I'll ask him in my response. If you do happen to know how to do that, then please do.

dezza69, 2016-08-27:
Yes, I've a few possibilities and no fear of running them!

Would they like / best to be from the same computer?

I'll get started tomorrow.

Dezza

FranticRain, 2016-08-27:
I don't think it particularly matters what machine it is on. He was not specific about it needing to be the same machine, but after mentioning that it issue was on a Macbook Pro, he responded: "...does the kext work with the problematic devices on a MacBook Pro with an earlier version of macOS installed 10.10.x and earlier?" I'll leave it up to you to interpret.

RodrigoCard, 2016-08-27:
Probably they are asking this because of the changes made in their usb
libs. To make sure it is not something they broke or their side.

-- 
Enviado do Gmail para celular

FranticRain, 2016-08-28:
I got a report from a separate, but related issue, that third party Xbox One controllers work on 10.10. I still definitely want you to try it, but if you can confirm his results then this is 100% Apple's fault. That would be great news for me, at least.

dezza69, 2016-08-28:
Hi,

Sorry for the delay - the Yosemite installer needed re-downloading...

I am very happy to confirm the number 2 driver you supplied as per the image below works after dis-connecting/re-connecting my Performance Designs Controller on the exact same MacBook Pro Retina, under these conditions:
a) the kext needs to be automatically removed by the kernel.

It will not work on an immediate disconnect/re-connect, but waiting a short period of time appears for everything to work fine - even after an immediate disconnect/re-connect not work then pausing and re-connecting.  I did this 4 times, each time worked flawlessly.

It does not work on El Capitan after disconnecting and re-connecting.

Thank you for all your efforts, happy to do more testing as needed (even for Apple).

[image]

[13 more replies]

closed 2018-04-03 by FranticRain