360Controller/360Controller #334: Model 1537 not found (WIRELESS CONTROLLER for XBOX ONE)

garethwelton, 2016-08-07:
Model 1537 Xbox One Genuine Controller, can't be found have tried the latest release and 16.1 but to no avail. Other 3rd party controllers working fine. GameStop PC1000GS

Serials:
1537 -IL
x899789-002

FranticRain, 2016-08-07:
I need your vendor and product IDs https://github.com/360Controller/360Controller#wired-xbox-one-controllers

xboxoneS, 2016-08-08:
Same issue the new bluetooth controller. Also connecting with USB cable

Product ID:	0x02ea
Vendor ID:	0x045e  (Microsoft Corporation)

FranticRain, 2016-08-08:
So this is about the Bluetooth Controller? There is an existing issue for that. #291 

More importantly, if you are trying to get it working via USB, you will want to look at this comment to get a build that will work with that controller: https://github.com/360Controller/360Controller/issues/291#issuecomment-237337851

closed 2016-08-08 by FranticRain

garethwelton, 2016-08-11:
Thanks was some confusion as I was trying to use the bluetooth controller but wired. Thanks again