360Controller/360Controller #335: PowerA Wired Controller not working being recognized. Product ID: 0x5397 Vendor ID:0x24c6

siracu55, 2016-08-08:


FranticRain, 2016-08-08:
Duplicate #312

closed 2016-08-08 by FranticRain

FranticRain, 2016-08-08:
Is this an Xbox One or Xbox 360 controller? Also, what is the official name of this controller?

siracu55, 2016-08-08:
This is an Xbox 360 wired controller. The official name is PowerA FUS1ON Tournament Controller.

FranticRain, 2016-08-08:
Okay sorry. I misinterpreted this based on the old name of this issue, and then realized it later. This should fix your problem: https://drive.google.com/file/d/0B5PLCKoMd4zGaGF1Y0NVSEtCMnc/view?usp=sharing

siracu55, 2016-08-08:
That worked! Thanks a ton! Now to just figure out how to get Dolphin to recognize it...

FranticRain, 2016-08-08:
This might be of help: http://wpguru.co.uk/2016/01/how-to-connect-your-xbox-360-controller-to-dolphin-for-mac/

siracu55, 2016-08-08:
I'm all set. Thanks again for your help.