360Controller/360Controller #336: Thrustmaster Ferrari 458 Racing Wheel

yency01, 2016-08-09:
Hello,
first of all, thanks for the great work on this controller.

The issue comes with my new controller, which is recognized by your driver, but not in the game, F1 2013 and DIRT3.
Controller Thrustmaster Ferrari 458 Racing Wheel for XBox360
ProductID 0x5600
VendorID 0x24c6
Version 1.01

All possible options already tried ... any ideas?

BTW, original XBox360 gamepad works

Thanks

FranticRain, 2016-08-09:
I just hacked this together and have ABSOLUTELY no way to test it whatsoever, so who knows what will happen. But I've modified the "Pretend to be an Xbox 360 Controller" checkbox in the "Advanced" tab to work with Xbox 360 controllers. Basically, your wheel should pretend to be Microsoft Xbox 360 controller when that checkbox is activated. Hopefully this will work and I didn't forget anything critical. https://drive.google.com/file/d/0B5PLCKoMd4zGQnB2V3M5MXN3N2M/view?usp=sharing

yency01, 2016-08-09:
> Antworten an: 360Controller/360Controller <reply@reply.github.com>

Hi Drew ... great job and so fast ... it works perfectly now, thanks a lot ...

 

Greetings, Jens

FranticRain, 2016-08-09:
Glad to hear it.

closed 2016-08-09 by FranticRain

jimbobgaming, 2021-07-02:
I need this file but it's now gone :(

Daitcher, 2022-04-23:
yep a reupload of the file would be great if possible @FranticRain