360Controller/360Controller #339: Driver for steering wheel

3tasker1, 2016-08-20:
Hiya,

I've been scouring the internet to find help to get my logitech G920 steering wheel setup on my mac with next to no avail. I've tried Free the Wheel. 

I saw this and (Stupidly) didn't realise it was for controllers rather than wheels. However, it did make the wheel initialise properly. Usually it stops mid-calibration.

Is there any way this could be modified to just initialise my wheel? Maybe then it will be detected by Free the Wheel.

I'm a Database Developer and have touched a bit of C++ but nothing on this scale of complexity.

Any help would be greatly appreciated!

Many Thanks

Chris

FranticRain, 2016-08-20:
We don't exactly make changes for specific controllers in the driver. Its not something we can really feasibly do. Your wheel should show up perfectly fine as a controller to video games, however. If it is some arbitrary packet we can throw at all Xbox One controllers, we should be able to solve your problem and completely void the need for Free the Wheel. I skimmed through the source code for Free the Wheel and it looks like its just a special packet that is sent to the controller. Unfortunately, it doesn't have a packet listed for the G920, so we will need to find it ourselves. Follow these sniffing instructions on a Windows machine that will set the controller up for proper 900 degree rotation: https://franticrain.github.io/sniffing/index.html

3tasker1, 2016-08-20:
Hi there.
Thank you so much for this. I've uploaded the file here...https://www.dropbox.com/s/838nbcqu3a9mgbm/Logitech-G920?dl=0

Many Thanks
Chris

FranticRain, 2016-08-20:
While I look over this, I made a branch of FreeTheWheel that should support the G920, if the Logitech documentation is right. It may not be. The documentation was released after the G920, but it doesn't mention the G920 even once. https://github.com/FranticRain/FreeTheWheel

3tasker1, 2016-08-20:
Thank you so much for that.
Unfortunately however it didn't work. It doesn't find the Steering wheel even though the MAC recognises it correctly.

> Chriss-MBP:~ chris$ /Users/chris/Downloads/FreeTheWheel-master/FreeTheWheel ; exit;
> ================================================================================
> =                               Free The Wheel (FTW)                           =
> =                             © Feral Interactive 2012                         =
> =    This program comes with ABSOLUTELY NO WARRANTY; This is free software,    =
> =            You are welcome to redistribute it under certain conditions,      =
> =                            Released under GNU License                        =
> ================================================================================
> =   --info       - display list of supported devices.                          =
> =   --restore    - Restore your wheel to restricted (default) mode.            =
> ================================================================================
> Looking for supported wheels to enable in NATIVE mode. . .
> 
> 
> Done.
> logout
> Saving session...
> ...copying shared history...
> ...saving history...truncating history files...
> ...completed.
> 
> 
> [Process completed]

3tasker1, 2016-08-20:
Also if this helps. Here are the details from System Report...
> G920 Driving Force Racing Wheel for Xbox One:
> Product ID:	0xc261
> Vendor ID:	0x046d  (Logitech Inc.)
> Version:	96.01
> Serial Number:	0000cf40161fb019
> Speed:	Up to 12 Mb/sec
> Manufacturer:	Logitech
> Location ID:	0x14100000 / 2
> Current Available (mA):	1000
> Extra Operating Current (mA):	0

FranticRain, 2016-08-20:
I added some debug text to that FreeTheWheel fork. Let me know what it says.

3tasker1, 2016-08-20:
There's no extra output.

3tasker1, 2016-08-20:
Here is the output

> Found deviceID: 41158060
> Didn't match with deviceID: 41158060
> Found deviceID: 41158060
> Didn't match with deviceID: 41158060
> Found deviceID: 1971506
> Didn't match with deviceID: 1971506
> Found deviceID: 41158060
> Didn't match with deviceID: 41158060
> Found deviceID: 41158060
> Didn't match with deviceID: 41158060
> Found deviceID: 41158060
> Didn't match with deviceID: 41158060
> Found deviceID: 1971506
> Didn't match with deviceID: 1971506

FranticRain, 2016-08-20:
If there is no extra output, then that thing is failing monumentally. It isn't finding a single device, much less checking them against its vendor and product ID list. When I run it on my Late 2013 rMBP, it spits out:
```
Found deviceID: 39978412
Didn't match with deviceID: 39978412
Found deviceID: 39978412
Didn't match with deviceID: 39978412
Found deviceID: 39978412
Didn't match with deviceID: 39978412
Found deviceID: 39978412
Didn't match with deviceID: 39978412
```
For some dumb reason, I didn't make them print in hex, but that first one is "0x26205AC", aka 0x0262 and 0x05AC. Thats the VID and PID of my trackpad. Make sure to re-download and re-build the fork and try again.

In addition, here is a version of the 360Controller driver that should work with your wheel a bit better. Check it in System Preferences -> Xbox 360 Controllers. You should be able to see button presses at least, and maybe wheel and pedal movements will show up as controller inputs. https://drive.google.com/file/d/0B5PLCKoMd4zGTkdvWmZFU2F0MXc/view?usp=sharing

3tasker1, 2016-08-20:
My apologies. There was extra output. I've updated the comment with output. I'll download that now

3tasker1, 2016-08-20:
Whilst it is indeed better. There are no pedals being registered and the steering axis is registered as Left Trigger where 15deg of wheel maxes out the trigger. All buttons work.

FranticRain, 2016-08-20:
Huh, so they did re-order the packet. That's a real pain in the butt. I'll put together a fix for that in a second.

I updated FreeTheWheel to print in hex so I can read the printouts better, but there is no listing of the G920 on that list. Are you sure it was plugged in? The driver isn't inhibiting FreeTheWheel from seeing the G920 because it sees my Xbox One controller when it is plugged in.

3tasker1, 2016-08-20:
I ran it again with the the last driver you sent and It's finding it.

> Found deviceID: -1033829267
> Logitech G920 Native mode enabled.
> WARNING: SendCommand failed with result: e00002c7
> Calibrated 900 degree wheel movement. (VendorID/DeviceID c261046d)

FranticRain, 2016-08-20:
Okay. Here is a version of the driver that should get those axes right. I'm sort of guessing what the order is here because its impossible to tell based on the sniff. There isn't much to go on there. https://drive.google.com/file/d/0B5PLCKoMd4zGSDB6T0JZRUlUYW8/view?usp=sharing

As for FreeTheWheel failing, that makes sense. I think there is a better way to get that packet that I need. In Windows, plug in the controller and everything, and then open up the Logitech Gaming Sofware. Start sniffing, and then change the wheel operating range to 900. After that, stop sniffing. The packet that activates the full range should be in that sniff. That will make it easier than having to sift through the hundred or so packets in this previous sniff and guess which one it is.

3tasker1, 2016-08-20:
Aaaah. With the new driver: 
Accelerator is RTrigger but 100% accelerator = 20% RTrigger.
Steering is still LTrigger.
Still no Clutch or Brake.
Also, LSB on wheel pushes RSB and vice versa.
I'll do the sniffing now.

3tasker1, 2016-08-20:
Sorry about the delay. My internet is awful and it's making me do a Firmware Update.

3tasker1, 2016-08-20:
Here are the packets https://www.dropbox.com/s/uv97hqny5u6vol4/G920-set900?dl=0
Many Thanks
Chris

FranticRain, 2016-08-20:
There wasn't anything useful in there. Too bad. The computer didn't send a single packet to the wheel.

I'm really concerned about this mixup between the LSB and RSB. That is a change in the button packet, which I can totally fix, but its really weird and disconcerting that they would change that. I don't like making individual hacks for controllers, especially when wheels aren't the focus of this driver. So I'm going to stop messing with the driver until we figure out this FreeTheWheel thing.

There should be some new patched to FreeTheWheel that offer a different packet based on what Linux uses. See if it works.

3tasker1, 2016-08-20:
Hmm. Now the wheel isn't being found.

> Found VID: 1532, PID: 1e
> Didn't match with deviceID: 1e1532
> Found VID: 1532, PID: 1e
> Didn't match with deviceID: 1e1532
> Found VID: 5ac, PID: 274
> Didn't match with deviceID: 27405ac
> Found VID: 5ac, PID: 274
> Didn't match with deviceID: 27405ac
> Found VID: 45e, PID: 28e
> Didn't match with deviceID: 28e045e
> Found VID: 5ac, PID: 274
> Didn't match with deviceID: 27405ac
> Found VID: 5ac, PID: 274
> Didn't match with deviceID: 27405ac
> Found VID: 5ac, PID: 274
> Didn't match with deviceID: 27405ac

FranticRain, 2016-08-20:
Go into the "Advanced" tab of the preference pane and deactivate the "Pretend to the a 360 Controller" option. Your wheel is pretending to be an Xbox 360 Controller. (28e045e)

3tasker1, 2016-08-20:
Aaaah thank you! 
Excellent! That looks good!

> Found VID: 46d, PID: c261
> Logitech G920 Native mode enabled.
> Calibrated 900 degree wheel movement. (VendorID/DeviceID c261046d)
> Found VID: 5ac, PID: 274

3tasker1, 2016-08-20:
This leaves the following Results:
Steering wheel is Left Stick with 900deg motion. However, rest is far left/right of the stick. IE, with the wheel centred the Left stick flickers between full left and full right.
No brake or acceleration pedal

FranticRain, 2016-08-20:
Oh so THATS what the trick is. The 900 degree motion makes that axis bigger than 16 bits. This is where things start to get really tricky. I can't change the report descriptor to have full access to the detail of this 900 degree motion. So while you will be able to move the wheel the full distance, it won't be anywhere near as precise as you might want, because I will have to scale that larger value back to 16 bits. If you want the actual precision from this controller, someone is going to have to make a real driver for it. Our focus here is controllers first and foremost. Everything that isn't a standard controller just get changed to look like a standard controller.

If you could do one last set of sniffs for me, I should have an easier time figuring out how this packet is set up. Have your controller plugged in to windows and calibrated and everything. Then start a sniff: Turn the wheel all the way to the left, hold it there for a second or so, and then all the way to the right, hold it, let it go back to center and stop. Then for each of the pedals, start a sniff, depress the pedal, hold it for a second, and let it go. I'd prefer if they were all separate sniffs, but you can do it all in one go if you really want to.

3tasker1, 2016-08-20:
Here are the sniffs
https://www.dropbox.com/s/y7dvnuifxja4emd/G920-packets.zip?dl=0
I included Clutch for completeness but am unsure how you could map that to a controller.

FranticRain, 2016-08-20:
Are you sure you were sniffing the correct USB port? All of those sniffs contain the same three packets.
```
00 00 00 00 00 00 00 00
01 00 00 00 00 00 00 00
01 00 06 00 00 00 00 00
```
None of those are valid packets.

3tasker1, 2016-08-20:
Nothing I seem to do changes the packets being captured. Definitely on the right filter.
Following filter control devices are available:

> 1 \\.\USBPcap1
>   \??\USB#ROOT_HUB30#4&1f16ebb7&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
>     [Port 3] USB Composite Device
>       Razer BlackWidow
>         HID Keyboard Device
>       USB Input Device
>         HID Keyboard Device
>         HID-compliant consumer control device
>         HID-compliant system controller
>         HID-compliant device
>         HID-compliant device
>       USB Input Device
>         HID-compliant mouse
>     [Port 4] Logitech G920 Driving Force Racing Wheel (XBox One)
> 2 \\.\USBPcap2
>   \??\USB#ROOT_HUB20#4&2b1006f0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
>     [Port 1] Generic USB Hub
>       [Port 1] USB Composite Device
>         USB Input Device
>           HID-compliant mouse
>         USB Input Device
>           HID Keyboard Device
>           HID-compliant consumer control device
>           HID-compliant system controller
>       [Port 4] Realtek Bluetooth 4.0 Adapter
>         Microsoft Bluetooth LE Enumerator
>         Bluetooth Device (RFCOMM Protocol TDI)
>         Microsoft Bluetooth Enumerator
>         Bluetooth Device (Personal Area Network)
> 3 \\.\USBPcap3
>   \??\USB#ROOT_HUB20#4&387a9e1&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
>     [Port 1] Generic USB Hub
>       [Port 1] USB Composite Device
>         HP Truevision HD
> Select filter to monitor (q to quit): 1
> Output file name (.pcap): G920-test-1

FranticRain, 2016-08-20:
Huh. That is definitely weird. Check the Control Panel -> Hardware and Sound -> Devices and Printers, then right click on your controller, open "Game Controller Settings". Then in the new window, select your wheel and click properties. Using that window, make sure everything still works. I want to make sure that its still functioning alright. You could of course test it with a game instead of doing that, but this is easier if you don't have any good to go.

3tasker1, 2016-08-20:
There you go. These seem better. After a restart the Wheel was showing up under a different filter/port?
Odd.
https://www.dropbox.com/s/mla7o158b66u4f8/G920-packets-working.zip?dl=0

FranticRain, 2016-08-20:
This is really weird. These packets should start with 0x20. Thats the button packet for Xbox One controllers. Instead we get these 0x01 packets. For example: ```01 0f 00 00 7c 7d ff ff ff 05```
So we have:
```
Byte 0 - 01 - Header type
Byte 1 - 0f - Packet length? Actual packet length is 8 bytes, but 0x0f is 16
Byte 2 - 00 - Button Packet
Byte 3 - 00 - Button Packet
Byte 4 - 7c - Wheel axis
Byte 5 - 7d - Wheel axis
Byte 6 - ff - Accelerator
Byte 7 - ff - Brake
Byte 8 - ff - Clutch
Byte 9 - 05 - Unknown
```
This is opposed to a typical packet from a controller that is formatted as:
```
Byte 00 - packet type
Byte 01 - reserved
Byte 02 - counter
Byte 03 - size
Byte 04 - buttons
Byte 05 - buttons
Byte 06 - left trigger
Byte 07 - left trigger
Byte 08 - right trigger
Byte 09 - right trigger
Byte 10 - left stick x
Byte 11 - left stick x
Byte 12 - left stick y
Byte 13 - left stick y
Byte 14 - right stick x
Byte 15 - right stick x
Byte 16 - right stick y
Byte 17 - right stick y
```
Or to the button packets I got from your init sniff:
```
Byte 00 - packet type
Byte 01 - reserved
Byte 02 - counter
Byte 03 - size
Byte 04 - buttons
Byte 05 - buttons
Byte 06 - ??
Byte 07 - ??
Byte 08 - ??
Byte 09 - ??
Byte 10 - ??
Byte 11 - ??
Byte 12 - ??
Byte 13 - ??
Byte 14 - ??
Byte 15 - ??
Byte 16 - ??
Byte 17 - ??
Byte 18 - ??
Byte 19 - ??
Byte 20 - ??
```
We are going to have to do this the messy, ugly, way, which is sniffing through this driver. Its going to print a bunch of garbage to your Console.app, which you can find in /Applications/Utilities. What I need you to do is look at the data that is being printed and tell me which sets of bytes are getting changed as you do specific tasks like rotating the wheel or pressing on the gas. Make sure to make your Console.app wide enough so that each printout only takes up one line. Otherwise it'll be much harder than it needs to be. Good luck. https://drive.google.com/file/d/0B5PLCKoMd4zGWHM3WkJaSTExUEE/view?usp=sharing

3tasker1, 2016-08-20:
Here are the packets at certain states
There were no packets for ANY button presses. A few times a new packet arrived with the same data but not for every button. Makes me think the packet is being trimmed before being logged?
First 2 - Steering axis. Left = 0 0, right = FF FF
3 - Accelerator. Rest = 0 Full = FF
4 - Not Used?
5 - Brake. Rest = 0 Full = FF
6 - Not Used?
7 - Not Used?
8 - Clutch Rest = 0 Full = FB(Highest I could get, probably FF)

And here is my reference data. Hope this helps! Thank you so much for all your time!
http://pastebin.com/vZ99tsaU

FranticRain, 2016-08-20:
Yeah, sorry. I cut off the initial stuff because I knew what it was and I didn't want to complicate things. Thanks for your help. I'll get this all slapped together in a bit and go ahead and bind clutch to the right stick x so you can at least bind it to something if a game lets you rebind controls.

3tasker1, 2016-08-20:
Aaaaaah okay. I see.

Thanks a lot, that'll be absolutely perfect!

Thanks

Chris

FranticRain, 2016-08-20:
Okay, give this one a shot and let me know if it gives you what you expect. https://drive.google.com/file/d/0B5PLCKoMd4zGcW9nZ0VRRzlCLWM/view?usp=sharing

3tasker1, 2016-08-21:
Thank you so much for that! Each axis maps correctly!
The only problems now are:
Steering starts from 0 0 or FF FF, not 8000. Assuming this relates to the problem with 900deg rotation you mentioned earlier. IE, when at rest, the Stick is either full left or full right.
Clutch is perfect
Brake and Accelerator need to be scaled. 100% of each is about 20-30% of each trigger.

FranticRain, 2016-08-21:
Give this one a shot. I moved clutch over the left y and fixed the other issues. It just makes more sense there. https://drive.google.com/file/d/0B5PLCKoMd4zGdDlUbG1KcWZwVEU/view?usp=sharing

3tasker1, 2016-08-21:
That is absolutely perfect!!!
Thank you so much for all of your hard work, it is truly appreciated!!

Thanks Again!

Chris

FranticRain, 2016-08-21:
Glad to hear it. This will be in the next official release, but you should just be able to continue using that build until then.

I also submitted a pull request to the upstream FreeTheWheel repository and it got merged, so use that from now on.

closed 2016-08-21 by FranticRain

3tasker1, 2016-08-21:
Excellent!

Thank you so much again!