360Controller/360Controller #340: Hori Real Arcade Pro V Kai for Xbox One / Xbox 360 does not connect.

xyywy, 2016-08-20:
Hello,

Could you please help me with my controller?

I am using the Real Arcade Pro V Kai for Xbox One / Xbox 360 which now cannot be deteced by the driver 0.16.3. 
http://stores.horiusa.com/real-arcade-pro-v-kai-for-xbox-one-xbox-360/
There is a switch on the stick. You can make it work on Xbox360 mode or Xbox One mode. 

On xbox360 mode: 
Product ID: 0x550e
Vendor ID: 0x24c6

And on Xbox One mode:
Product ID: 0x0078
Vendor ID: 0x0f0d  (HORI CO., LTD.)

Thanks a lot for helping me!!!

FranticRain, 2016-08-20:
This should work in both modes. You might have trouble with the Xbox One mode though. We are still fighting third party support for Xbox One controllers. It works for some, not for others. https://drive.google.com/file/d/0B5PLCKoMd4zGbVhOX091Nkttd1k/view?usp=sharing

closed 2016-08-20 by FranticRain

xyywy, 2016-08-20:
Thank you!!!
You are really GREAT!!! 
Thanks for helping so many people!!!