360Controller/360Controller #343: Using MadCatz controller

davidy93, 2016-08-22:
Hi, so when I go to System Preferences and test out the controller I am able to get it to respond. All the button inputs show up on the controller test panel and the light on the controller comes on. But when I actually play a game, the game does not acknowledge that I have a controller plugged in. I've tried 0.16.3 and 0.16.1.

FranticRain, 2016-08-22:
Unfortunately, that is a problem with the game. We can't provide fixes for individual games, so this is out of our control. You haven't said whether this is an Xbox One controller or an Xbox 360 controller. If it is an Xbox 360 Controller, install this version of the driver and check the "Pretend to be a 360 Controller" option in the "Advanced" tab of the preference pane. If its an Xbox One controller, that option should already be available in the preference pane in the same place. https://drive.google.com/file/d/0B5PLCKoMd4zGQnB2V3M5MXN3N2M/view?usp=sharing

Beyond that, there is nothing we can do to fix this. You'll have to ask the game developer for help.

closed 2016-08-22 by FranticRain

davidy93, 2016-08-23:
Now it works! Thanks!