360Controller/360Controller #344: Inconsistently "re-enabling" Wired 360 Controller

DrPuppet, 2016-08-23:
Hi folks! So, I really appreciated it when you added the "enable" feature to the system preferences panel. However, for the past couple of versions or so, I have had mixed luck re-enabling the driver after unchecking the box (for Aspyr games). I either restart or do the old-school Terminal command. So it is not impossible to work around, but I was wondering if this is a known issue?

Running a mid-2011 27" iMac, OS 10.11.6.

Thank you!

FranticRain, 2016-08-23:
I just did a little testing with an Xbox One controller, but I can't recreate this issue. I don't have any wired Xbox 360 controllers to test with. Its a bit weird to think that you are fixing this with a terminal command considering toggling the checkbox is calling a terminal command. Are there any cases where you can get it to consistently fail?

DrPuppet, 2016-08-23:
Well, the only Aspyr game I've been playing recently is Rage. Other than that, I cannot think of any consistent variables.

FranticRain, 2016-08-23:
And how does the driver fail, specifically? When it is re-enabled, does the controller light up properly? Does the preference pane see a controller, but not recognize any button presses? Does disconnecting and reconnecting the controller have any impact?

DrPuppet, 2016-08-23:
Oh, pardon. When I plug it in, the "enable" checkbox has been rechecked (without my input), but is greyed out. The controller is not listed in the dropdown menu. I believe once or twice I have been able to solve it by disconnecting and reconnecting, as you said, but this has not worked with the latest update, as far as I can recall. Rebooting the computer or (at least the last time) typing in the Terminal command seem to work. I should add that, as long as I keep the box checked, it works consistently, even after disconnecting and reconnecting the controller multiple times.

FranticRain, 2016-08-23:
Oh okay, I can definitely recreate this now. Let me see what I can do.

DrPuppet, 2016-08-23:
Awesome, thank you!

FranticRain, 2016-08-23:
Install this version of the driver and let me know if it fixes your issue. That option will no longer be greyed out. There is no reason for it to ever be in that state. And the status of the check should now be properly based on whether or not the kext is loaded. https://drive.google.com/file/d/0B5PLCKoMd4zGSXRuc0V5N2V0MVU/view?usp=sharing

DrPuppet, 2016-08-24:
This version is very promising. I installed it last night, rebooted; I believe "enable" was checked by default. I unchecked, dis- and reconnected the controller, and played Rage with no issues. I'll try to re-enable and play a non-Aspyr game tonight, just to make sure.

DrPuppet, 2016-08-26:
Sorry for the delay. I finally had a chance to play (and finish!) a non-Aspyr game (Postal 2), and the controller worked flawlessly. Indeed, I checked "enable" while it was connected, and the light came on and all was well (i.e. I did not even have to disconnect and reconnect, although that still seems to be necessary when disabling). Thanks again!

closed 2016-08-26 by DrPuppet