360Controller/360Controller #345: No connection

Chris01076, 2016-08-28:
The controller vibrates and is connected to the computer but doesn't show or anything

FranticRain, 2016-08-28:
Please follow the guidelines in the README for debugging issues. We cannot help you fix a problem like this without any information about the controller you are trying to use. Is it an Xbox One controller? Or is it a 360 controller? What are the vendor and product IDs? What version of OS X are you running? https://github.com/360Controller/360Controller#my-controller-doesnt-work

Chris01076, 2016-08-28:
I'm using a wireless xbox one controller, Microsoft it came with the xbox and i'm using OS X El Capitan, Version 10.11.4

FranticRain, 2016-08-28:
Follow these instructions to get your vendor and product IDs for your controller. I've linked you to the wired section since you are connecting the controller to the PC via USB. https://github.com/360Controller/360Controller#wired-xbox-one-controllers

boarderman8, 2016-09-10:
I followed the above link as i am also having this issue. here is what i have 
[image]

FranticRain, 2016-09-10:
That means that this is probably yet ANOTHER init packet issue. Do you have access to a Windows machine? In the meantime, could you test an experimental version of the driver for me that might solve your issue? https://drive.google.com/file/d/0B5PLCKoMd4zGeEZIYVpSMFdUY3c/view?usp=sharing

boarderman8, 2016-09-10:
I do have access to a Windows machine. I'll try this update for you as soon as I get a chance

FranticRain, 2016-09-10:
If that build doesn't work, then get a USB sniff by following these instructions: https://franticrain.github.io/sniffing/index.html

boarderman8, 2016-09-10:
it works on my pc already, will the sniff still be useful?

FranticRain, 2016-09-10:
If you are saying that it works on your Windows machine but not your Mac, then yes, this will be useful. The problem right now is that your controller is pairing with the driver, but the driver isn't sending it the right data to cause it to initialize properly. However, your Windows PC is initializing the controller properly. By seeing what data the Windows PC sends to the controller, we can try and figure out what data the Mac needs to send to the controller to initialize it properly.

boarderman8, 2016-09-10:
https://drive.google.com/file/d/0B-jwAb69GMQhY0dQYmtUUnZ6bGM/view?usp=sharing

There it is.

boarderman8, 2016-09-10:
let me know if that worked

FranticRain, 2016-09-10:
Yeah. That's what I needed. The main problem is that it looks nearly identical to other sniffs from the no-audio-jack version of the Xbox One controller. There are some things in here where the driver violates its own packet protocol. It uses the packet index 3 multiple times. Its supposed to start from 0 and increment on each packet. The controller enforces this pretty heavily. If you don't maintain that number, rumble packets won't work. I'll give you two more builds to test in a second, but I need you to test all of them.

There are three groups of people when it comes to Xbox One controllers. Ones who use the old init packets (yours), ones who use the new init packets, and ones whose controllers don't care which init packets they get. All of my controllers are in the third camp, which makes debugging this a real pain in the butt. Because there is no rhyme or reason why this stuff is messed up. It works differently for different people using the same controller. Its not even related to firmware, because I've always updated my controller firmware to match those of the people with these issues. 

For all I know, its another bug in Apple's USB system because they're so smart and went and screwed the whole system up in 10.11 like the bunch of geniuses they are.

FranticRain, 2016-09-10:
Here are the two builds:
Follows the protocol: https://drive.google.com/file/d/0B5PLCKoMd4zGMzhTSktGNFhqcjQ/view?usp=sharing
Violates the protocol: https://drive.google.com/file/d/0B5PLCKoMd4zGblpkb043RE1reFU/view?usp=sharing

boarderman8, 2016-09-10:
The first one worked when I spoofed it as a 360 controller

FranticRain, 2016-09-10:
Okay, what? What is actually your problem? Spoofing doesn't change anything regarding initialization of the controller. Was your controller lighting up this entire time and you were just having an issue with a game?

boarderman8, 2016-09-10:
Yea, I was having the same issue as the other guy, it would connect and light up and vibrate but wouldn't work for any game. i tried spoofing with the other two drivers but this is the only one that worked.

FranticRain, 2016-09-10:
Great. That's just great.

closed 2016-09-10 by FranticRain