360Controller/360Controller #347: Xbox ONE liquid metal controller:

macmanpapi, 2016-08-31:
Xbox ONE liquid metal controller:

  Product ID:	0x543a
  Vendor ID:	0x24c6
  Version:	1.01
  Serial Number:	000049289A35D6D9
  Speed:	Up to 12 Mb/sec
  Manufacturer:	PowerA
  Location ID:	0x1d100000 / 1
  Current Available (mA):	1000
  Current Required (mA):	500
  Extra Operating Current (mA):	0

i need some help getting this to work please

FranticRain, 2016-08-31:
Duplicate https://github.com/360Controller/360Controller/issues/312#issuecomment-232778183

This controller does not work with OS X 10.11 due to a bug in OS X itself. It will work if you downgrade to 10.10. Otherwise, you are out of luck.

closed 2016-08-31 by FranticRain

macmanpapi, 2016-08-31:
thank you i will do