360Controller/360Controller #350: Wired Controller not recognized

bhuff97, 2016-09-06:
Product ID:	0x0501
  Vendor ID:	0x0e6f
  Version:	1.00
  Serial Number:	06925989
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14200000 / 7
  Current Available (mA):	1000
  Extra Operating Current (mA):	0

FranticRain, 2016-09-06:
Do you have any sort of product name for this controller? We like to have names attached to the product and vendor ID combinations, so we can get an idea of what type of controller it is.

bhuff97, 2016-09-06:
Play- Gaming Accesories
Xbox 360 & Windows Wired Controller
It's from GameStop

FranticRain, 2016-09-06:
Here you go: https://drive.google.com/file/d/0B5PLCKoMd4zGZHN6MldYLWJ1cDg/view?usp=sharing

closed 2016-09-06 by FranticRain

bhuff97, 2016-09-06:
Awesome. Thank You!