360Controller/360Controller #351: Xbox One Elite

startswithJ, 2016-09-07:
I am getting a similar problem to a few others -- when I connect the controller it vibrates, but doesn't turn the xbox light on. Could it be that its an Xbox One elite controller and not the regular? 

Product ID:	0x02e3
  Vendor ID:	0x045e  (Microsoft Corporation)

OS: El Capitan; version 10.11.6

Thank you for your help!

FranticRain, 2016-09-07:
It is possible that this is the same bug as is plaguing the third party Xbox One controllers. I've had multiple reports that the Xbox One Elite controller works just fine with this driver. #315 

In that issue, you'll find an experimental build. Try it out. Other than that, just try making sure you are using a data cable and not a charge cable. The third party bug only got introduced in 10.11, so if you revert to 10.10, it might resolve this issue.

startswithJ, 2016-09-07:
Ok, update:

The original xbox one controller worked just fine with the original driver installed, but the xbox one elite did not. After installing the experimental build, the xbox one elite controller is recognized right away! 

Thank you so much for your help, friend! Now hopefully all this works well with Rocket League when it is released for Mac tomorrow...

FranticRain, 2016-09-07:
Does you normal controller work fine with the experimental version? It should, but I just want to make sure. If it does, then I'll push the experimental version changes to master and it will be included in the next release.

startswithJ, 2016-09-07:
Forgive me, but I am somewhat dev illiterate. How do I uninstall the original driver so that I can test the experimental with the regular xbox one controller? Currently, when I plugin the regular controller it is only recognized by the "Xone controller" driver, but not the "xbox 360 controllers" driver.

FranticRain, 2016-09-07:
Whoa whoa whoa, hold on here. Do you have Xone-OSX installed? You need to uninstall that immediately. This changes things, a lot. To uninstall Xone-OSX, download the latest release from here: https://github.com/FranticRain/Xone-OSX/releases. In the release, there will be a file called `uninstall.command`. If you double click that, it will uninstall Xone-OSX. Next, if you want to uninstall 360Controller, open up your System Preferences from the Apple menu in the top left. Then open the "Xbox 360 Controllers" preference pane. From within this preference pane, switch the the "Advanced" tab. This will allow you to uninstall 360Controller. However, 360Controller should install over itself just fine, so you don't need to uninstall it if you are just going to install over it again.

startswithJ, 2016-09-08:
Alright, so I uninstalled the Xone Driver that only worked with the regular xbox one controller. Now i have ONLY the experimental driver installed and it recognizes both the regular xbox one controller AND the Xbox one elite controller. I take it this is good news?

FranticRain, 2016-09-08:
Possibly. Some users have issues with different init packets. For example, how these experimental init packets work for you, but the old ones didn't. Its not a complete confirmation of success, but its not a failure either. I'm going to go ahead and close this issue since it will be resolved in the next release.

Just a reminder, if Rocket League doesn't work with your controllers, make sure to check the "Pretend to be a 360 controller" option in the "Advanced" tab. If the game still doesn't work, there isn't anything we can do to solve it on a game-by-game basis. Its just up to the devs to add proper controller support to their game.

closed 2016-09-08 by FranticRain