360Controller/360Controller #352: Xbox One Halo 5 LE Controller not found

TrustMeImHelping, 2016-09-09:
The controller appears under System Report as "Bluetooth USB Host Controller," but does not appear on the controller test. 

Vendor ID: 0x05ac (Apple Inc.)
Product ID: 0x8290
Version: 1.04
iOS Version 10.10.5 Yosemite

When I plug the controller in it vibrates, but doesn't light up, nor are my inputs displayed on the controller test.

FranticRain, 2016-09-09:
That is definitely not your controller. Apple doesn't develop Xbox One controllers. Try looking again. If you don't see it, try a different USB cable. You could be using a "charge" cable that doesn't transfer data, and will therefore not work with this driver.

d235j, 2016-09-09:
That vendor/product ID pair is for the computer's Bluetooth radio, and has nothing to do with the Xbox One controller. If you don't see anything that looks like the Xbox One controller, then it's probably the cable.

TrustMeImHelping, 2016-09-09:
Will any USB cable used to connect speakers/android products work? Or is there a specific type of USB cable? I have tried 2 different cables: one used to charge a speaker and another to charge a portable phone battery. Both just make the controller vibrate.

FranticRain, 2016-09-09:
The driver matches to devices based on their vendor and product ID combination. It will not match this controller until we put those IDs into the driver. We need you to use a micro-USB cable to connect to your Mac. This will cause your controller to vibrate, thats just how macOS treats these controllers when the driver doesn't match with them. Then, we need you to open up that USB System Report tool and find the controller's vendor and product IDs. These will not show up unless you are using a data cable. The device will likely be named just "Controller" and it will almost certainly have a vendor ID of "0x045e". Look for any device that looks like that. If you can't find it, try a different cable.

TrustMeImHelping, 2016-09-16:
Hi, thanks for the help. Got a new cable and learned in the process that there are two different types: mini and micro. Everything works fine, all is well. Thread can be closed now; thanks again!

closed 2016-09-16 by FranticRain