360Controller/360Controller #353: No Device Found

akosipau, 2016-09-09:
In system log under USB, it detects by name "controller" however it says No Device Found.

FranticRain, 2016-09-09:
Give me your vendor and product IDs for the "Controller" entry as is clearly stated in the README. https://github.com/360Controller/360Controller#my-controller-doesnt-work

akosipau, 2016-09-09:
Here is it.

Product ID:	0x028e
Vendor ID:	0x1bad
Version:	4.90
Serial Number:	364E4867
Speed:	Up to 12 Mb/sec
Manufacturer:	Microsoft Corporation.
Location ID:	0x14200000 / 9
Current Available (mA):	1000
Extra Operating Current (mA):	0

FranticRain, 2016-09-09:
Here is a build that contains support for this device. Support for this device will also be included in the next official release. https://drive.google.com/file/d/0B5PLCKoMd4zGRHplUWVBUVY0em8/view?usp=sharing

closed 2016-09-09 by FranticRain